mavlink/MAVSDK-Python #8: Examples need more instructions

hamishwillee, 2018-05-14:
I tried to build the examples in a virtual env named mytemp. Followed build instructions and end up with pip  list as below. 
```
(mytemp) ubuntu@ubuntu:~/github/dronecore/3test/DroneCore-Python/examples$ pip list
Package             Version    
------------------- -----------
dronecore-action    0.2        
dronecore-core      0.2        
dronecore-mission   0.2        
dronecore-telemetry 0.2        
enum34              1.1.6      
futures             3.2.0      
grpcio              1.11.0     
grpcio-tools        1.11.0     
pip                 10.0.1     
pkg-resources       0.0.0      
protobuf            3.5.2.post1
setuptools          39.1.0     
six                 1.11.0     
wheel               0.31.1     
```
1. When I try running any example script I get the error `No module named dronecore_core.core_pb2`
   ```
   Traceback (most recent call last):
     File "sync_client.py", line 7, in <module>
       import dronecore_core.core_pb2 as dc_core
   ImportError: No module named dronecore_core.core_pb2
   ```
2. How do I set up the vehicle? Ie can I use SITL? Do I need to set up anything to tell the client about the MAV system id I want to connect to? Can I use a real device over serial?

Upshot, how are these examples supposed to be run?

xvzf, 2018-05-15:
Regarding the 1st point:
The examples are currently only for testing the generated code. You have to call the `./tools/run_protoc.sh` then there will be no import error!

xvzf, 2018-05-15:
That doesn't sound right. Just checked it again.

Do you have `grpcio` and `grpcio-tools` intstalled?

hamishwillee, 2018-05-15:
Yes, you can see them in the pip list too :-)

hamishwillee, 2018-05-15:
I can see the files here:
[image]
Just not located when I try build the examples.

xvzf, 2018-05-15:
Can you please try the recent version and check if the issue persists? If so please let me know which python and OS version you are using!

hamishwillee, 2018-05-15:
Same error. This is Python 3.5.2, in a venv virtual environment, on Ubuntu LTS 16.04 (running in a VM on Windows). The full steps to running example below:
```
ubuntu@ubuntu:~/github/dronecore$ cd 4test/
ubuntu@ubuntu:~/github/dronecore/4test$ python3 -m venv mypy3env4
ubuntu@ubuntu:~/github/dronecore/4test$ source mypy3env4/bin/activate
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test$ git clone https://github.com/dronecore/DroneCore-Python.git --recursive
Cloning into 'DroneCore-Python'...
remote: Counting objects: 100, done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 100 (delta 20), reused 53 (delta 17), pack-reused 42
Receiving objects: 100% (100/100), 19.21 KiB | 0 bytes/s, done.
Resolving deltas: 100% (35/35), done.
Checking connectivity... done.
Submodule 'proto' (https://github.com/dronecore/DroneCore-Proto.git) registered for path 'proto'
Cloning into 'proto'...
remote: Counting objects: 131, done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 131 (delta 3), reused 7 (delta 2), pack-reused 117
Receiving objects: 100% (131/131), 19.01 KiB | 0 bytes/s, done.
Resolving deltas: 100% (38/38), done.
Checking connectivity... done.
Submodule path 'proto': checked out 'f2e8b28b231e5f318977e755f6e5acfd385ad447'
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test$ python -m pip install grpcio grpcio-tools
Collecting grpcio
  Downloading https://files.pythonhosted.org/packages/2c/ff/f118147fd7a8d2d441d15e1cb7fefb2c1981586e24ef3a7d8a742535b085/grpcio-1.12.0-cp35-cp35m-manylinux1_x86_64.whl (9.0MB)
    100% |████████████████████████████████| 9.0MB 151kB/s 
Collecting grpcio-tools
  Downloading https://files.pythonhosted.org/packages/fe/8c/32c000c5bcb86469b6a4cadd1ff587edda979d69ad2a05625eafab6e95ea/grpcio_tools-1.12.0-cp35-cp35m-manylinux1_x86_64.whl (22.2MB)
    100% |████████████████████████████████| 22.2MB 65kB/s 
Collecting six>=1.5.2 (from grpcio)
  Downloading https://files.pythonhosted.org/packages/67/4b/141a581104b1f6397bfa78ac9d43d8ad29a7ca43ea90a2d863fe3056e86a/six-1.11.0-py2.py3-none-any.whl
Collecting protobuf>=3.5.0.post1 (from grpcio-tools)
  Downloading https://files.pythonhosted.org/packages/5b/c3/9b947e301e19bea75dc8c1fd3710eed5d2b31aa13ae13d5e38e891f784cc/protobuf-3.5.2.post1-cp35-cp35m-manylinux1_x86_64.whl (6.4MB)
    100% |████████████████████████████████| 6.4MB 39kB/s 
Requirement already satisfied (use --upgrade to upgrade): setuptools in ./mypy3env4/lib/python3.5/site-packages (from protobuf>=3.5.0.post1->grpcio-tools)
Installing collected packages: six, grpcio, protobuf, grpcio-tools
Successfully installed grpcio-1.12.0 grpcio-tools-1.12.0 protobuf-3.5.2.post1 six-1.11.0
You are using pip version 8.1.1, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test$ cd DroneCore-Python/
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python$ ./other/tools/run_protoc.sh
[+] Generating protobuf and gRPC bindings
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python$ pip install -e .
Obtaining file:///home/ubuntu/github/dronecore/4test/DroneCore-Python
Collecting Rx>=1.6.1 (from dronecore===0.1.0a0)
  Downloading https://files.pythonhosted.org/packages/33/0f/5ef4ac78e2a538cc1b054eb86285fe0bf7a5dbaeaac2c584757c300515e2/Rx-1.6.1-py2.py3-none-any.whl (179kB)
    100% |████████████████████████████████| 184kB 948kB/s 
Requirement already satisfied (use --upgrade to upgrade): protobuf>=3.5.1 in /home/ubuntu/github/dronecore/4test/mypy3env4/lib/python3.5/site-packages (from dronecore===0.1.0a0)
Requirement already satisfied (use --upgrade to upgrade): setuptools in /home/ubuntu/github/dronecore/4test/mypy3env4/lib/python3.5/site-packages (from protobuf>=3.5.1->dronecore===0.1.0a0)
Requirement already satisfied (use --upgrade to upgrade): six>=1.9 in /home/ubuntu/github/dronecore/4test/mypy3env4/lib/python3.5/site-packages (from protobuf>=3.5.1->dronecore===0.1.0a0)
Installing collected packages: Rx, dronecore
  Running setup.py develop for dronecore
Successfully installed Rx-1.6.1 dronecore
You are using pip version 8.1.1, however version 10.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python$ dir
docs  dronecore  dronecore.egg-info  other  proto  README.rst  requirements-dev.txt  requirements.txt  setup.cfg  setup.py
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python$ cd docs
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python/docs$ cd examples/
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python/docs/examples$ dir
async_client.py  fly_mission.py  multiple_device_handling.py  sync_client.py
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python/docs/examples$ python async_client.py 
Traceback (most recent call last):
  File "async_client.py", line 10, in <module>
    import dronecore.generated.core_pb2_grpc as dc_core_grpc
ImportError: No module named 'dronecore.generated.core_pb2_grpc'
(mypy3env4) ubuntu@ubuntu:~/github/dronecore/4test/DroneCore-Python/docs/examples$ 
```
This is not a completely new Ubuntu installation, but that should not matter as I'm using the venv - so it should be a clean Python env.

hamishwillee, 2018-05-15:
Also, can you point me to instructions for setting up the backend? Or are they not needed to run the examples?

xvzf, 2018-05-16:
There seems to be a bug with Python 3.5. I was able to reproduce it - only with Python 3.5. The latest version is working fine here.

I'll have a look!

Backend should be as easy as calling `make ENABLE_BACKEND=YES` in the DroneCore repository

xvzf, 2018-05-16:
Quick update:
First of all, the files were not generated, that is fixed. However the internal import still fails.

hamishwillee, 2018-05-16:
Great! Ubuntu LTS 16.04 is pretty much the default dev platform for PX4, and it gets Python 3.5.2 by default, so we should make this work - and as a last resort specify a version and how to get it in our docs.

> Backend should be as easy as calling make ENABLE_BACKEND=YES in the DroneCore repository

I'm sure that builds the backend but that is only a small part of the answer someone needs in order to be able to use this system. Does making that setting also start the backend as part of building? If not how? What does that backend expect to be talking to - Wifi/Serial? What is the id of the connected vehicle? How is it configured. 

All of that needs to be answered before someone can use these examples.

JonasVautherin, 2018-05-16:
Regarding the backend, the best way IMO would be to have it as a package, so people can just install it. And then, those who want to compile it from sources can do it from the main dronecore repo.

Unfortunately, it is not yet released as a package, and it is still in active development. But that will come soon :-).

JonasVautherin, 2018-05-16:
(Note: I added the "Python" label for the kanban board [here](https://waffle.io/dronecore/dronecore)).

xvzf, 2018-05-16:
Just tested https://github.com/dronecore/DroneCore-Python/pull/11 in a 16.04 VM! Seems to be working!

closed 2018-05-16 by JonasVautherin

JonasVautherin, 2018-05-17:
Actually, the examples are outdated. "SubscribeDevices" is not a thing anymore, that was part of the proof of concept. The development of the Python bindings is really just starting. So we can keep this issue open as a reminder that we should actually start implementing them, but then it means that all the PRs will connect to this issue.

So I thought we could close it for now, and either let development go with PRs, or open issues like "implement the core frontend" if we want to have issues.

PR #11 (2018-05-16): Fixes import issue for the generated files.
Fixes #8.