mavlink/MAVSDK-Python #10: Move docs/examples to /examples

hamishwillee, 2018-05-15:
@xvzf The new structure good, but can we move the examples back into the root. Docs are going to be stored in a separate gitbook. If you want to have intermediate docs, put them in the wiki and let me know :-)

xvzf, 2018-05-17:
I planned on using Sphinx for the Code Documentation - can you convert that to gitbook?

hamishwillee, 2018-05-17:
I am a huge fan of Sphinx, but we have about 7 non-Python doc sets, hosted using gitbook on gitbook.com. It makes no sense for one of them to be documented using a different toolchain and  markup.

So lets write in markdown. In the first instance we can just use the wiki for docs. 

It is possible to convert between reStructuredText and markdown using tools like pandoc, but it is a pain to do and keep doing. So lets not.

xvzf, 2018-05-17:
Okay!
I'd like to have some in-code documentation. Makes it easier to work witth! Do you have any proposal for python that is compatible with gitbook?

hamishwillee, 2018-05-18:
> I'd like to have some in-code documentation. Makes it easier to work witth! Do you have any proposal for python that is compatible with gitbook?

I completely understand. Unfortunately we're too early down the client API and docs path for me to give final advice.

For now, just document it however you prefer, ignoring the ultimate toolchain. Note however that it might all get ripped out and modified at some point in future.  I'm not saying that will happen - it is possible that the model for docs will end up being "every client library for itself".

hamishwillee, 2018-05-18:
And in either case, please move examples folder into the root :-)

xvzf, 2018-06-04:
I think this can be closed now!

hamishwillee, 2018-06-05:
Yes it can. Thank you!

closed 2018-06-05 by hamishwillee