mavlink/MAVSDK-Python #29: Troubles installing

julianoes, 2018-10-24:
I'm trying to set this up on macOS but I'm getting:

```
[+] Installing the DronecodeSDK autogenerator
Processing /Users/julianoes/src/DronecodeSDK-Python/proto/pb_plugins
Requirement already satisfied: protobuf in /Users/julianoes/Library/Python/3.7/lib/python/site-packages (from dcsdkgen==0.1a0) (3.6.1)
Requirement already satisfied: jinja2 in /Users/julianoes/Library/Python/3.7/lib/python/site-packages (from dcsdkgen==0.1a0) (2.10)
Requirement already satisfied: setuptools in /usr/local/lib/python3.7/site-packages (from protobuf->dcsdkgen==0.1a0) (40.0.0)
Requirement already satisfied: six>=1.9 in /Users/julianoes/Library/Python/3.7/lib/python/site-packages (from protobuf->dcsdkgen==0.1a0) (1.11.0)
Requirement already satisfied: MarkupSafe>=0.23 in /Users/julianoes/Library/Python/3.7/lib/python/site-packages (from jinja2->dcsdkgen==0.1a0) (1.0)
Building wheels for collected packages: dcsdkgen
  Running setup.py bdist_wheel for dcsdkgen: started
  Running setup.py bdist_wheel for dcsdkgen: finished with status 'done'
  Stored in directory: /private/var/folders/6h/nc4rf_dj3bdd9_0yz7vgtwb00000gn/T/pip-ephem-wheel-cache-dr7a4pnl/wheels/76/58/d2/7a926efaca7b5a6468f13775931776959adf41f3b2ee2a4edc
Successfully built dcsdkgen
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
  The script dcsdkgen is installed in '/Users/julianoes/Library/Python/3.7/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
sed: -i: No such file or directory
# Generated by the gRPC Python protocol compiler plugin. DO NOT EDIT!
import grpc

from . import gimbal_pb2 as gimbal__pb2

class GimbalServiceStub(object):
  # missing associated documentation comment in .proto file
  pass

  def __init__(self, channel):
    """Constructor.

    Args:
      channel: A grpc.Channel.
    """
    self.SetPitchAndYaw = channel.unary_unary(
        '/dronecode_sdk.rpc.gimbal.GimbalService/SetPitchAndYaw',
        request_serializer=gimbal__pb2.SetPitchAndYawRequest.SerializeToString,
        response_deserializer=gimbal__pb2.SetPitchAndYawResponse.FromString,
        )

class GimbalServiceServicer(object):
  # missing associated documentation comment in .proto file
  pass

  def SetPitchAndYaw(self, request, context):
    # missing associated documentation comment in .proto file
    pass
    context.set_code(grpc.StatusCode.UNIMPLEMENTED)
    context.set_details('Method not implemented!')
    raise NotImplementedError('Method not implemented!')

def add_GimbalServiceServicer_to_server(servicer, server):
  rpc_method_handlers = {
      'SetPitchAndYaw': grpc.unary_unary_rpc_method_handler(
          servicer.SetPitchAndYaw,
          request_deserializer=gimbal__pb2.SetPitchAndYawRequest.FromString,
          response_serializer=gimbal__pb2.SetPitchAndYawResponse.SerializeToString,
      ),
  }
  generic_handler = grpc.method_handlers_generic_handler(
      'dronecode_sdk.rpc.gimbal.GimbalService', rpc_method_handlers)
  server.add_generic_rpc_handlers((generic_handler,))
 -> [+] Generated protobuf and gRPC bindings for gimbal
: program not found or is not executable
[... 1275 lines cut ...]
          servicer.SubscribeFlightMode,
          request_deserializer=telemetry__pb2.SubscribeFlightModeRequest.FromString,
          response_serializer=telemetry__pb2.FlightModeResponse.SerializeToString,
      ),
      'SubscribeHealth': grpc.unary_stream_rpc_method_handler(
          servicer.SubscribeHealth,
          request_deserializer=telemetry__pb2.SubscribeHealthRequest.FromString,
          response_serializer=telemetry__pb2.HealthResponse.SerializeToString,
      ),
      'SubscribeRCStatus': grpc.unary_stream_rpc_method_handler(
          servicer.SubscribeRCStatus,
          request_deserializer=telemetry__pb2.SubscribeRCStatusRequest.FromString,
          response_serializer=telemetry__pb2.RCStatusResponse.SerializeToString,
      ),
  }
  generic_handler = grpc.method_handlers_generic_handler(
      'dronecode_sdk.rpc.telemetry.TelemetryService', rpc_method_handlers)
  server.add_generic_rpc_handlers((generic_handler,))
 -> [+] Generated protobuf and gRPC bindings for telemetry
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/julianoes/src/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Telemetry.py to /Users/julianoes/src/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/telemetry.py: No such file or directory
 -> [+] Generated plugin for telemetry
[+] Done
```

JonasVautherin, 2018-10-24:
Which version of Python do you have?

julianoes, 2018-10-24:
it's Python 3.7.

xvzf, 2018-10-24:
Can you try it inside a virtual env?

xvzf, 2018-10-24:
Another quick note, you can go for 
```bash
# pip install --user -r requirements.txt -r requirements-dev.txt -r requirements-test.txt
```
instead of installing each dependency!

julianoes, 2018-10-25:
>     # pip install --user -r requirements.txt -r requirements-dev.txt -r requirements-test.txt

This needs to be in the README if it's the suggested way of doing it. Please please update the README.

xvzf, 2018-10-25:
I'll add that later!

Is this issue resolved? Then I'll mark it as closed with the PR updating the Readme!

julianoes, 2018-10-25:
Ok, adding the path and installing all the requirements (with pip3, not pip) made the `run_protoc.sh` run through.

julianoes, 2018-10-25:
Then I did:
```
pip3 install --user -e .
```

and then got as far as:
```
python3 examples/takeoff_and_land.py
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 3, in <module>
    from dronecode_sdk import connect as dronecode_sdk_connect
  File "/Users/julianoes/src/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 38, in <module>
    from .plugins import *
  File "/Users/julianoes/src/DronecodeSDK-Python/dronecode_sdk/plugins/__init__.py", line 3, in <module>
    from .gimbal import *
  File "/Users/julianoes/src/DronecodeSDK-Python/dronecode_sdk/plugins/gimbal.py", line 3, in <module>
    from ..generated import gimbal_pb2, gimbal_pb2_grpc
  File "/Users/julianoes/src/DronecodeSDK-Python/dronecode_sdk/generated/gimbal_pb2_grpc.py", line 4, in <module>
    import gimbal_pb2 as gimbal__pb2
ModuleNotFoundError: No module named 'gimbal_pb2'
```

xvzf, 2018-10-25:
Can you post a dump of `tree`? This seems pretty strange to me and I cannot reproduce it

julianoes, 2018-10-25:
```
tree
.
├── README.rst
├── docs
│   └── TODO.md
├── dronecode_sdk
│   ├── __init__.py
│   ├── __pycache__
│   │   ├── __init__.cpython-37.pyc
│   │   ├── _base.cpython-37.pyc
│   │   └── async_plugin_manager.cpython-37.pyc
│   ├── _base.py
│   ├── async_plugin_manager.py
│   ├── generated
│   │   ├── __init__.py
│   │   ├── __pycache__
│   │   │   ├── __init__.cpython-37.pyc
│   │   │   ├── gimbal_pb2.cpython-37.pyc
│   │   │   └── gimbal_pb2_grpc.cpython-37.pyc
│   │   ├── action_pb2.py
│   │   ├── action_pb2_grpc.py
│   │   ├── calibration_pb2.py
│   │   ├── calibration_pb2_grpc.py
│   │   ├── camera_pb2.py
│   │   ├── camera_pb2_grpc.py
│   │   ├── core_pb2.py
│   │   ├── core_pb2_grpc.py
│   │   ├── discovery_pb2.py
│   │   ├── discovery_pb2_grpc.py
│   │   ├── gimbal_pb2.py
│   │   ├── gimbal_pb2_grpc.py
│   │   ├── mission_pb2.py
│   │   ├── mission_pb2_grpc.py
│   │   ├── telemetry_pb2.py
│   │   └── telemetry_pb2_grpc.py
│   └── plugins
│       ├── __init__.py
│       ├── __pycache__
│       │   ├── __init__.cpython-37.pyc
│       │   └── gimbal.cpython-37.pyc
│       ├── action.py
│       ├── calibration.py
│       ├── camera.py
│       ├── core.py
│       ├── discovery.py
│       ├── gimbal.py
│       ├── mission.py
│       └── telemetry.py
├── dronecode_sdk.egg-info
│   ├── PKG-INFO
│   ├── SOURCES.txt
│   ├── dependency_links.txt
│   ├── requires.txt
│   └── top_level.txt
├── examples
│   ├── takeoff_and_land.py
│   ├── telemetry_flight_mode.py
│   ├── telemetry_is_armed_is_in_air.py
│   └── telemetry_takeoff_and_land.py
├── other
│   ├── docker
│   │   └── testing
│   │       ├── Dockerfile
│   │       ├── build_python.sh
│   │       └── test.sh
│   ├── templates
│   │   ├── call.j2
│   │   ├── enum.j2
│   │   ├── file.j2
│   │   ├── request.j2
│   │   ├── stream.j2
│   │   └── struct.j2
│   └── tools
│       ├── run_protoc.sh
│       └── run_tests.sh
├── output.txt
├── proto
│   ├── README.md
│   ├── pb_plugins
│   │   ├── README.rst
│   │   ├── dcsdkgen
│   │   │   ├── __init__.py
│   │   │   ├── __main__.py
│   │   │   ├── autogen.py
│   │   │   ├── autogen_file.py
│   │   │   ├── enum.py
│   │   │   ├── methods.py
│   │   │   ├── name_parser.py
│   │   │   ├── struct.py
│   │   │   ├── type_info.py
│   │   │   └── utils.py
│   │   ├── requirements.txt
│   │   └── setup.py
│   └── protos
│       ├── action
│       │   └── action.proto
│       ├── calibration
│       │   └── calibration.proto
│       ├── camera
│       │   └── camera.proto
│       ├── core
│       │   └── core.proto
│       ├── discovery
│       │   └── discovery.proto
│       ├── gimbal
│       │   └── gimbal.proto
│       ├── mission
│       │   └── mission.proto
│       └── telemetry
│           └── telemetry.proto
├── requirements-dev.txt
├── requirements-test.txt
├── requirements.txt
├── setup.cfg
├── setup.py
├── tests
└── tox.ini
```

julianoes, 2018-10-27:
Hm, yes works fine on Arch Linux. The `run_protoc.sh` output is different on macOS, I'll check that on Monday.

xvzf, 2018-10-27:
Sure!

Make sure to include a `pip freeze` aswell!

xvzf, 2018-10-27:
@mrpollo got it working a couple of weeks ago, maybe he also has an idea!

closed 2018-10-29 by xvzf

Huibean, 2018-12-28:
I still got empty plugins dir in my mac, Any Idea?
```shell
➜  DronecodeSDK-Python git:(master) ✗ ./other/tools/run_protoc.sh
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from
 -> [+] Generated protobuf and gRPC bindings for gimbal
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/huibean/Work/DronecodeSDK-Python/other/tools/../../dronecode_sdk/plugins/Gimbal.py to /Users/huibean/Work/DronecodeSDK-Python/other/tools/../../dronecode_sdk/plugins/gimbal.py: No such file or directory
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for core
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1
```

```shell
➜  DronecodeSDK-Python git:(master) ipython3 examples/takeoff_and_land.py
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
~/Work/DronecodeSDK-Python/examples/takeoff_and_land.py in <module>()
      3 import asyncio
      4
----> 5 from dronecode_sdk import connect as dronecode_sdk_connect
      6
      7 drone = dronecode_sdk_connect(host="127.0.0.1")

~/Work/DronecodeSDK-Python/dronecode_sdk/__init__.py in <module>()
     36 # Plugins rely on the eventloop
     37 from .async_plugin_manager import AsyncPluginManager
---> 38 from .plugins import *
     39
     40

~/Work/DronecodeSDK-Python/dronecode_sdk/plugins/__init__.py in <module>()
      1 # -*- coding: utf-8 -*-
      2
----> 3 from .gimbal import *
      4 from .core import *
      5 from .calibration import *

ModuleNotFoundError: No module named 'dronecode_sdk.plugins.gimbal'```

JonasVautherin, 2018-12-28:
@Huibean: are you sure you cloned the git submodules?

And would you mind opening a new issue for your problem, since this one was closed already =)

Huibean, 2018-12-28:
@JonasVautherin yes, I have the submodule installed

julianoes, 2018-12-29:
@Huibean please open a new issue.

Huibean, 2018-12-29:
@julianoes yep,it is [here ](https://github.com/Dronecode/DronecodeSDK-Python/issues/38)

PR #30 (2018-10-29): Fix macOS install
This fixes #29. The problem was that sed on BSD is a bit different, see https://stackoverflow.com/questions/4247068/.

Needs cross-testing on Linux.