mavlink/MAVSDK-Python #38: ./other/tools/run_protoc.sh got empty plugins direction

Huibean, 2018-12-28:
```shell
➜  DronecodeSDK-Python git:(master) ✗ ./other/tools/run_protoc.sh
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from
 -> [+] Generated protobuf and gRPC bindings for gimbal
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/huibean/Work/DronecodeSDK-Python/other/tools/../../dronecode_sdk/plugins/Gimbal.py to /Users/huibean/Work/DronecodeSDK-Python/other/tools/../../dronecode_sdk/plugins/gimbal.py: No such file or directory
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for core
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1
```

```shell
➜  DronecodeSDK-Python git:(master) ipython3 examples/takeoff_and_land.py
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
~/Work/DronecodeSDK-Python/examples/takeoff_and_land.py in <module>()
      3 import asyncio
      4
----> 5 from dronecode_sdk import connect as dronecode_sdk_connect
      6
      7 drone = dronecode_sdk_connect(host="127.0.0.1")

~/Work/DronecodeSDK-Python/dronecode_sdk/__init__.py in <module>()
     36 # Plugins rely on the eventloop
     37 from .async_plugin_manager import AsyncPluginManager
---> 38 from .plugins import *
     39
     40

~/Work/DronecodeSDK-Python/dronecode_sdk/plugins/__init__.py in <module>()
      1 # -*- coding: utf-8 -*-
      2
----> 3 from .gimbal import *
      4 from .core import *
      5 from .calibration import *

ModuleNotFoundError: No module named 'dronecode_sdk.plugins.gimbal'```

xvzf, 2018-12-28:
Hi!

You have to generate the plugins and then install the SDK, e.g. by executing `pip install -e .` inside the project root

Huibean, 2018-12-28:
@xvzf I just did that, but  problem still, here is the project tree and full log 
```shell
➜  DronecodeSDK-Python git:(master) ✗ tree
.
├── README.md
├── docs
│   └── TODO.md
├── dronecode_sdk
│   ├── __init__.py
│   ├── __pycache__
│   │   ├── __init__.cpython-36.pyc
│   │   └── async_plugin_manager.cpython-36.pyc
│   ├── _base.py
│   ├── async_plugin_manager.py
│   ├── generated
│   │   ├── __init__.py
│   │   ├── action_pb2.py
│   │   ├── action_pb2_grpc.py
│   │   ├── calibration_pb2.py
│   │   ├── calibration_pb2_grpc.py
│   │   ├── camera_pb2.py
│   │   ├── camera_pb2_grpc.py
│   │   ├── core_pb2.py
│   │   ├── core_pb2_grpc.py
│   │   ├── discovery_pb2.py
│   │   ├── discovery_pb2_grpc.py
│   │   ├── gimbal_pb2.py
│   │   ├── gimbal_pb2_grpc.py
│   │   ├── info_pb2.py
│   │   ├── info_pb2_grpc.py
│   │   ├── mission_pb2.py
│   │   ├── mission_pb2_grpc.py
│   │   ├── telemetry_pb2.py
│   │   └── telemetry_pb2_grpc.py
│   └── plugins
│       ├── __init__.py
│       └── __pycache__
│           └── __init__.cpython-36.pyc
├── dronecode_sdk.egg-info
│   ├── PKG-INFO
│   ├── SOURCES.txt
│   ├── dependency_links.txt
│   ├── requires.txt
│   └── top_level.txt
├── examples
│   ├── mission.py
│   ├── takeoff_and_land.py
│   ├── telemetry_flight_mode.py
│   ├── telemetry_is_armed_is_in_air.py
│   └── telemetry_takeoff_and_land.py
├── other
│   ├── docker
│   │   └── testing
│   │       ├── Dockerfile
│   │       ├── build_python.sh
│   │       └── test.sh
│   ├── templates
│   │   ├── call.j2
│   │   ├── enum.j2
│   │   ├── file.j2
│   │   ├── request.j2
│   │   ├── stream.j2
│   │   ├── struct.j2
│   │   └── type_conversions
│   └── tools
│       ├── run_protoc.sh
│       └── run_tests.sh
├── proto
│   ├── README.md
│   ├── pb_plugins
[... 113 lines cut ...]
 -> [+] Generated plugin for telemetry
[+] Done
➜  DronecodeSDK-Python git:(master) ✗ pip3 install -e .
Obtaining file:///Users/huibean/Work/DronecodeSDK-Python
Requirement already satisfied: protobuf>=3.5.1 in /Users/huibean/anaconda3/lib/python3.6/site-packages (from dronecode-sdk==0.1.0a0) (3.6.1)
Requirement already satisfied: aiogrpc>=1.5 in /Users/huibean/anaconda3/lib/python3.6/site-packages (from dronecode-sdk==0.1.0a0) (1.5)
Requirement already satisfied: grpcio>=1.11.0 in /Users/huibean/anaconda3/lib/python3.6/site-packages (from dronecode-sdk==0.1.0a0) (1.17.1)
Requirement already satisfied: setuptools in /Users/huibean/anaconda3/lib/python3.6/site-packages (from protobuf>=3.5.1->dronecode-sdk==0.1.0a0) (36.5.0.post20170921)
Requirement already satisfied: six>=1.9 in /Users/huibean/anaconda3/lib/python3.6/site-packages (from protobuf>=3.5.1->dronecode-sdk==0.1.0a0) (1.11.0)
Installing collected packages: dronecode-sdk
  Found existing installation: dronecode-sdk 0.1.0a0
    Uninstalling dronecode-sdk-0.1.0a0:
      Successfully uninstalled dronecode-sdk-0.1.0a0
  Running setup.py develop for dronecode-sdk
Successfully installed dronecode-sdk
➜  DronecodeSDK-Python git:(master) ✗ examples/takeoff_and_land.py
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 5, in <module>
    from dronecode_sdk import connect as dronecode_sdk_connect
  File "/Users/huibean/Work/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 38, in <module>
    from .plugins import *
  File "/Users/huibean/Work/DronecodeSDK-Python/dronecode_sdk/plugins/__init__.py", line 3, in <module>
    from .gimbal import *
ModuleNotFoundError: No module named 'dronecode_sdk.plugins.gimbal'
```

julianoes, 2018-12-29:
Please use `pip3 install -r requirements.txt -r requirements-dev.txt`, unless your distribution brings Python 3 by default.

xvzf, 2018-12-29:
@julianoes Agreed. I'll do that right away!

Also I'd strongly advice we opt in for venv because it setups all the path needed to execute the autogenerator! (we just cannot do `python3 -m dcsdkgen`, protobuf doesn't like it at all )

xvzf, 2018-12-29:
@Huibean  Please come back with the result! :-)

Huibean, 2018-12-29:
@xvzf @julianoes In fact, I am using pip3 all the time, but it not seems to be the problem here, by the way I more likely to figure out how message is generated

Huibean, 2018-12-29:
I just solve this, ./other/tools/run_protoc.sh could not handle the dcsdkgen path well, so I give a fix path to it and python3 -m grpc_tools.protoc run through
```shell
➜  DronecodeSDK-Python git:(master) ✗ which dcsdkgen
dcsdkgen not found
```
```shell
        python3 -m grpc_tools.protoc -I$(dirname ${PROTO_FILE}) \
                                     --plugin=protoc-gen-custom=$('/Users/huibean/.local/bin/dcsdkgen') \ #replace which dcsdkgen
                                     --custom_out=${PLUGIN_DIR} \
                                     --custom_opt=py \
                                     ${PROTO_FILE}
```

xvzf, 2018-12-29:
> ➜  DronecodeSDK-Python git:(master) ✗ which dcsdkgen
dcsdkgen not found

This is likely because your `PATH` is not set correctly! Are you using a venv?

Can you post the output of `echo $PATH` ?

Huibean, 2018-12-30:
@xvzf when use --user in pip install, it would install to ~/.local/bin,  I don't have ~/.local/bin in my $PATH, I haven't remove this manually before, not sure install anaconda would affected this

julianoes, 2018-12-30:
@Huibean what about adding `~/.local/bin` to your `$PATH`? Isn't that common practice?

xvzf, 2018-12-30:
@Huibean You have to add it to your path or use a virtual environment as described above which setups the path for you!