mavlink/MAVSDK-Python #46: Crash uploading mission items in gRPC

julianoes, 2019-02-12:
A crash report by that needs to be investigated:

> Fatal error during mission upload. Probably after several missions were uploaded. Many commands (>50). Output: https://paste.ofcode.org/qmzXxgYH2LkVpYiyXBGJ5u
Version: Not sure yet, but <0.13.0 and possibly newer versions.

https://paste.ofcode.org/qmzXxgYH2LkVpYiyXBGJ5u

julianoes, 2019-02-13:
According to https://groups.google.com/forum/#!topic/grpc-io/M84p6FD-BcE

> FTR, it looks like "grpc_call_start_batch returned 8" corresponds to:
>
> /** there is already an outstanding read/write operation on the call */
>
> GRPC_CALL_ERROR_TOO_MANY_OPERATIONS,
> That may mean that for a given call, you are not waiting for outstanding read/write operations to finish (the tag completion returned by grpc_completion_queue) before starting a new read/write operation on that call
(which is a violation of how gRPC's low level API should be used).

JonasVautherin, 2019-02-13:
I am not sure if it is a crash in the upload. From the logs, it seems like it may come from the mission progress stream (which is, IMO, more likely to create a `grpc_call_start_batch returned 8`).

JonasVautherin, 2019-02-13:
@julianoes: Did you manage to reproduce it? Would be nice to understand where that happens (e.g. in upload, start, or mission progress).

julianoes, 2019-02-13:
Yes, trying to reproduce now.

julianoes, 2019-02-13:
Ok, it happened:
```
[11:40:09|Debug] Send mission item 197 (mission_impl.cpp:1096)
[11:40:09|Debug] Send mission item 198 (mission_impl.cpp:1096)
[11:40:09|Debug] Send mission item 199 (mission_impl.cpp:1096)
[11:40:09|Info ] Mission accepted (mission_impl.cpp:165)
[11:40:09|Debug] MAVLink: info: Executing Mission (system_impl.cpp:291)
[11:40:09|Debug] current: -1, total: 100 (mission_impl.cpp:1171)
[11:40:09|Debug] current: 0, total: 100 (mission_impl.cpp:1171)
E0213 11:40:09.341707831   26159 server_cc.cc:619]           Fatal: grpc_call_start_batch returned 8
E0213 11:40:09.341729832   26159 server_cc.cc:620]           ops[0]: SEND_MESSAGE ptr=0x7fffc8000d60

Thread 3 "backend_bin" received signal SIGABRT, Aborted.
[Switching to Thread 0x7ffff64da700 (LWP 26159)]
0x00007ffff763153f in raise () from /lib64/libc.so.6
(gdb) 
(gdb) 
(gdb) 
(gdb) bt
#0  0x00007ffff763153f in raise () from /lib64/libc.so.6
#1  0x00007ffff761b895 in abort () from /lib64/libc.so.6
#2  0x0000000000a7e3a3 in grpc::Server::PerformOpsOnCall (this=0x1076b80, ops=0x7fffbeffca98, call=0x7fffbeffc970)
    at /home/julianoes/src/DronecodeSDK/build/default/third_party/grpc/build/grpc/src/grpc/src/cpp/server/server_cc.cc:622
#3  0x000000000092231f in grpc::internal::Call::PerformOps (this=0x7fffbeffc970, ops=0x7fffbeffca98)
    at third_party/grpc/include/grpc++/impl/codegen/call.h:692
#4  0x00000000009aa96d in grpc::ServerWriter<dronecode_sdk::rpc::mission::MissionProgressResponse>::Write (this=0x7fffbeffc710, msg=..., options=...)
    at third_party/grpc/include/grpc++/impl/codegen/sync_stream.h:670
#5  0x0000000000912ed1 in grpc::internal::WriterInterface<dronecode_sdk::rpc::mission::MissionProgressResponse>::Write (this=0x7fffbeffc718, msg=...)
    at third_party/grpc/include/grpc++/impl/codegen/sync_stream.h:124
#6  0x000000000090e259 in dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::SubscribeMissionProgress(grpc::ServerContext*, dronecode_sdk::rpc::mission::SubscribeMissionProgressRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::mission::MissionProgressResponse>*)::{lambda(int, int)#1}::operator()(int, int) const (this=0x106a500, current=0, total=100) at ../../backend/src/plugins/mission/mission_service_impl.h:164
#7  0x0000000000916c60 in std::_Function_handler<void (int, int), dronecode_sdk::backend::MissionServiceImpl<dronecode_sdk::Mission>::SubscribeMissionProgress(grpc::ServerContext*, dronecode_sdk::rpc::mission::SubscribeMissionProgressRequest const*, grpc::ServerWriter<dronecode_sdk::rpc::mission::MissionProgressResponse>*)::{lambda(int, int)#1}>::_M_invoke(std::_Any_data const&, int&&, std::_Any_data const&) (__functor=..., 
    __args#0=@0x7ffff64d9944: 0, __args#1=@0x7ffff64d9940: 100) at /usr/include/c++/8/bits/std_function.h:297
#8  0x00007ffff7e4826b in std::function<void (int, int)>::operator()(int, int) const (this=0x106a0d0, __args#0=0, __args#1=100)
    at /usr/include/c++/8/bits/std_function.h:687
#9  0x00007ffff7e43b0d in dronecode_sdk::MissionImpl::<lambda()>::operator()(void) const (__closure=0x7ffff64d9b90)
    at ../../plugins/mission/mission_impl.cpp:1172
#10 0x00007ffff7e45240 in std::_Function_handler<void(), dronecode_sdk::MissionImpl::report_progress()::<lambda()> >::_M_invoke(const std::_Any_data &)
    (__functor=...) at /usr/include/c++/8/bits/std_function.h:297
#11 0x00007ffff7cf5d36 in std::function<void ()>::operator()() const (this=0x7ffff64d9b90) at /usr/include/c++/8/bits/std_function.h:687
#12 0x00007ffff7d3f744 in dronecode_sdk::ThreadPool::worker (this=0x1070a98) at ../../core/thread_pool.cpp:45
#13 0x00007ffff7d4286e in std::__invoke_impl<void, void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> (
    __f=@0x1067f60: (void (dronecode_sdk::ThreadPool::*)(dronecode_sdk::ThreadPool * const)) 0x7ffff7d3f6f2 <dronecode_sdk::ThreadPool::worker()>, 
    __t=@0x1067f58: 0x1070a98) at /usr/include/c++/8/bits/invoke.h:73
#14 0x00007ffff7d424c1 in std::__invoke<void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> (
    __fn=@0x1067f60: (void (dronecode_sdk::ThreadPool::*)(dronecode_sdk::ThreadPool * const)) 0x7ffff7d3f6f2 <dronecode_sdk::ThreadPool::worker()>, 
    __args#0=@0x1067f58: 0x1070a98) at /usr/include/c++/8/bits/invoke.h:95
#15 0x00007ffff7d42ddf in std::thread::_Invoker<std::tuple<void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> >::_M_invoke<0ul, 1ul> (
    this=0x1067f58) at /usr/include/c++/8/thread:244
#16 0x00007ffff7d42d5a in std::thread::_Invoker<std::tuple<void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> >::operator() (
    this=0x1067f58) at /usr/include/c++/8/thread:253
#17 0x00007ffff7d42bd0 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (dronecode_sdk::ThreadPool::*)(), dronecode_sdk::ThreadPool*> > >::_M_run (this=0x1067f50) at /usr/include/c++/8/thread:196
#18 0x00007ffff7a21943 in ?? () from /lib64/libstdc++.so.6
#19 0x00007ffff7b3158e in start_thread () from /lib64/libpthread.so.0
#20 0x00007ffff76f66a3 in clone () from /lib64/libc.so.6
```

And now all threads:
```
Thread 19 (Thread 0x7fffbd7fa700 (LWP 27300)):
#0  0x00007ffff76f0efd in syscall () from /lib64/libc.so.6
#1  0x00007ffff7a1f5e1 in std::__atomic_futex_unsigned_base::_M_futex_wait_until(unsigned int*, unsigned int, bool, std::chrono::duration<long, std::ratio<1l, 1l> >, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /lib64/libstdc++.so.6
[... cut ...]
#4  0x0000000000905cd8 in dronecode_sdk::backend::DronecodeSDKBackend::Impl::wait (this=0x10706c0) at ../../backend/src/backend.cpp:29
#5  0x0000000000900d58 in dronecode_sdk::backend::DronecodeSDKBackend::wait (this=0x7fffffffdad8) at ../../backend/src/backend.cpp:50
#6  0x0000000000900c2c in runBackend (mavlink_listen_port=14540, onServerStarted=0x0, context=0x0) at ../../backend/src/backend_api.cpp:15
#7  0x0000000000900bc9 in main (argc=1, argv=0x7fffffffdbe8) at ../../backend/src/dronecode_sdk_server.cpp:5

```

julianoes, 2019-02-13:
@JonasVautherin could this be a leak given we have multiple threads subscribed to mission progress and also multiple for the in-air status?

JonasVautherin, 2019-02-13:
So it happens in `SubscribeMissionProgress`, right?

> could this be a leak given we have multiple threads subscribed to mission progress and also multiple for the in-air status?

Looking at the code, I don't think we leak. If we register multiple times, the old callbacks will never be called so the stream will be [hanging there](https://github.com/Dronecode/DronecodeSDK/blob/develop/backend/src/plugins/mission/mission_service_impl.h#L167) until the caller cancels the call, but that's not exactly a leak. gRPC supports having multiple calls on the same function.

What could happen, maybe, is that two threads could be trying to [`writer->Write`](https://github.com/Dronecode/DronecodeSDK/blob/develop/backend/src/plugins/mission/mission_service_impl.h#L164) at the same time, because two events were emitted almost at the same time:

```
[11:40:09|Debug] current: -1, total: 100 (mission_impl.cpp:1171)
[11:40:09|Debug] current: 0, total: 100 (mission_impl.cpp:1171)
```

Could that be that they aren't called on the same thread on our C++ side?

julianoes, 2019-02-13:
> What could happen, maybe, is that two threads could be trying to [`writer->Write`](https://github.com/Dronecode/DronecodeSDK/blob/develop/backend/src/plugins/mission/mission_service_impl.h#L164) at the same time, because two events were emitted almost at the same time:

Right I noticed that it only happens in the case where updates are fired at (almost) the same time.

> Could that be that they aren't called on the same thread on our C++ side?

Do they need to be called on the same thread? They might be on the thread pool in which case it might be different threads from the pool.

JonasVautherin, 2019-02-13:
> Do they need to be called on the same thread? They might be on the thread pool in which case it might be different threads from the pool.

In which case we need a mutex on the shared `writer`, which [we do for the camera](https://github.com/Dronecode/DronecodeSDK/blob/develop/backend/src/plugins/camera/camera_service_impl.h#L179). That was not propagated everywhere, and I am hoping that the auto-generation will prevent those kinds of issues in the future.

 My question would then be: does it make sense to call them on different threads (as in: "couldn't it happen that the order is not preserved?")? And of course, maybe it is not so easy to make sure they are called on the same thread, which is possibly why we added the mutex in the camera.

zulufoxtrot, 2019-02-13:
Some more info:
- The only occurence of crashes is right after mission upload.
- The mission is uploaded _and_ executed despite the crash.
- This appears to happen after a few uploads (my application uploads new missions frequently) and never at first upload.

julianoes, 2019-02-13:
https://github.com/Dronecode/DronecodeSDK/blob/develop/plugins/mission/mission_impl.cpp#L1125

The question is why we have two reports in short succession, that's also odd.

JonasVautherin, 2019-02-13:
> https://github.com/Dronecode/DronecodeSDK/blob/91ed9c2478adf6b18704d2cd127623bc30d1eb46/plugins/mission/mission_impl.cpp#L1125

What's your point there? :sweat_smile: 

> The question is why we have two reports in short succession, that's also odd.

Yes. And what does -1 mean? Is it related to @zulufoxtrot's second point: "The mission is uploaded _and_ executed despite the crash."? Like... I don't know, when a new mission overwrites an older one, then somehow -1 is emitted and directly 0 after that?

When playing with the python examples, I was also feeling like sometimes I receive the 0 event, and sometimes the first one is 1.

@zulufoxtrot: Thanks for those points, that's super constructive :+1:! Very good bug report :-).

zulufoxtrot, 2019-02-13:
New full output, crash at second upload:

```
./build/default/backend/src/backend_bin
[03:12:12|Info ] DronecodeSDK version: 0.14.1 (dronecode_sdk_impl.cpp:25)
[03:12:12|Debug] New: System ID: 0 Comp ID: 0 (dronecode_sdk_impl.cpp:285)
[03:12:12|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:41)
[03:12:12|Info ] Server started (grpc_server.cpp:25)
[03:12:12|Info ] Waiting to discover system... (connection_initiator.h:57)
[03:12:12|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:200)
[03:12:12|Debug] Component Autopilot (1) added. (system_impl.cpp:381)
[03:12:12|Debug] Discovered 1 component(s) (UUID: 4294967298) (system_impl.cpp:540)
[03:12:12|Info ] System discovered [UUID: 4294967298] (connection_initiator.h:61)
[03:12:13|Debug] MAVLink: info: data link #1 lost (system_impl.cpp:292)
[03:12:13|Debug] MAVLink: info: Mission finished, loitering. (system_impl.cpp:292)
[03:12:13|Debug] MAVLink: info: data link #1 regained (system_impl.cpp:292)
[03:12:14|Debug] Send mission item 0 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 1 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 2 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 3 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 4 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 5 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 6 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 7 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 8 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 9 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 10 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 11 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 12 (mission_impl.cpp:1098)
[03:12:14|Debug] Send mission item 13 (mission_impl.cpp:1098)
[03:12:14|Info ] Mission accepted (mission_impl.cpp:167)
[03:12:14|Debug] MAVLink: info: Executing Mission (system_impl.cpp:292)
[03:13:23|Debug] MAVLink: info: Mission finished, loitering. (system_impl.cpp:292)
[03:14:37|Debug] current: 14, total: 14 (mission_impl.cpp:1173)
[03:18:46|Debug] Send mission item 0 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 1 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 2 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 3 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 4 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 5 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 6 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 7 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 8 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 9 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 10 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 11 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 12 (mission_impl.cpp:1098)
[03:18:46|Debug] Send mission item 13 (mission_impl.cpp:1098)
[03:18:46|Info ] Mission accepted (mission_impl.cpp:167)
[03:18:46|Debug] MAVLink: info: Executing Mission (system_impl.cpp:292)
[03:18:46|Debug] current: 1, total: 14 (mission_impl.cpp:1173)
[03:18:46|Debug] current: -1, total: 14 (mission_impl.cpp:1173)
E0213 15:18:46.158610307   12604 server_cc.cc:619]           Fatal: grpc_call_start_batch returned 8
E0213 15:18:46.158645495   12604 server_cc.cc:620]           ops[0]: SEND_INITIAL_METADATA(nil)
E0213 15:18:46.158664340   12604 server_cc.cc:620]           ops[1]: SEND_MESSAGE ptr=0x7faed4000d60
Aborted (core dumped)
```

julianoes, 2019-02-13:
@JonasVautherin right, so I guess we should fix both, the double output with `-1` but also the crash due to the thread-unsafety (clearly that's a word!).
I'm already trying to reproduce with a log around the `Writer()`.

zulufoxtrot, 2019-02-13:
@julianoes thanks, you guys are doing all the hard work :)

Now I just got a crash at first upload. However the output is different:

```
./build/default/backend/src/backend_bin
[03:21:29|Info ] DronecodeSDK version: 0.14.1 (dronecode_sdk_impl.cpp:25)
[03:21:29|Debug] New: System ID: 0 Comp ID: 0 (dronecode_sdk_impl.cpp:285)
[03:21:29|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:41)
[03:21:29|Info ] Server started (grpc_server.cpp:25)
[03:21:29|Info ] Waiting to discover system... (connection_initiator.h:57)
[03:21:29|Info ] New device on: 127.0.0.1:14557 (udp_connection.cpp:200)
[03:21:29|Debug] Component Autopilot (1) added. (system_impl.cpp:381)
[03:21:29|Debug] Discovered 1 component(s) (UUID: 4294967298) (system_impl.cpp:540)
[03:21:29|Info ] System discovered [UUID: 4294967298] (connection_initiator.h:61)
[03:21:30|Debug] MAVLink: info: data link #1 lost (system_impl.cpp:292)
[03:21:30|Debug] MAVLink: info: Mission finished, loitering. (system_impl.cpp:292)
[03:21:30|Debug] MAVLink: info: data link #1 regained (system_impl.cpp:292)
[03:21:32|Debug] current: -1, total: 0 (mission_impl.cpp:1173)
[03:21:47|Debug] Send mission item 0 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 1 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 2 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 3 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 4 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 5 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 6 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 7 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 8 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 9 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 10 (mission_impl.cpp:1098)
[03:21:47|Debug] current: 13, total: 21 (mission_impl.cpp:1173)
[03:21:47|Debug] Send mission item 11 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 12 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 13 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 14 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 15 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 16 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 17 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 18 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 19 (mission_impl.cpp:1098)
[03:21:47|Debug] Send mission item 20 (mission_impl.cpp:1098)
[03:21:47|Info ] Mission accepted (mission_impl.cpp:167)
[03:21:47|Debug] MAVLink: info: Executing Mission (system_impl.cpp:292)
[03:21:47|Debug] current: -1, total: 21 (mission_impl.cpp:1173)
[03:21:47|Debug] current: 1, total: 21 (mission_impl.cpp:1173)
Segmentation fault (core dumped)
```

julianoes, 2019-02-13:
Interesting, this time it's not the `Fatal: grpc_call_start_batch returned 8`. This would mean it's probably not actually that gRPC error but rather undefined behaviour that we're seeing.

JonasVautherin, 2019-02-13:
```
[03:18:46|Debug] current: 1, total: 14 (mission_impl.cpp:1173)
[03:18:46|Debug] current: -1, total: 14 (mission_impl.cpp:1173)
```

Wrong order here :sweat_smile:. That's generally not an issue, but we should keep that in mind.

> Interesting, this time it's not the `Fatal: grpc_call_start_batch returned 8`.

Or sometimes it ends up with a segv. I could imagine that, too.

> so I guess we should fix both, the double output with `-1` but also the crash due to the thread-unsafety (clearly that's a word!).

What about you take the `-1`, and I take the mutex? Or @zulufoxtrot do you want to contribute to that fix in C++? :blush:

julianoes, 2019-02-13:
I'm already trying the mutex but now only one mission update arrives on the Python side which is odd.

zulufoxtrot, 2019-02-13:
@JonasVautherin I wish I could, but I'm non proficient in C++.

JonasVautherin, 2019-02-13:
> @JonasVautherin I wish I could, but I'm non proficient in C++.

No worries, reporting and testing is already super helpful! Thanks a lot for that :+1:.

julianoes, 2019-02-13:
Yet another one:
```
[03:55:45|Info ] Mission accepted (mission_impl.cpp:165)
[03:55:45|Debug] MAVLink: info: Executing Mission (system_impl.cpp:292)
[03:55:45|Debug] current: -1, total: 100 (mission_impl.cpp:1173)
[03:55:45|Debug] current: 0, total: 100 (mission_impl.cpp:1173)
E0213 15:55:45.823407871   19347 byte_stream.cc:53]          assertion failed: stream->cursor < stream->backing_buffer->count

Thread 3 "backend_bin" received signal SIGABRT, Aborted.
```

julianoes, 2019-02-13:
Fix coming up in: https://github.com/Dronecode/DronecodeSDK/pull/668

julianoes, 2019-02-14:
@JonasVautherin should re release `v0.14.2` for to include this fix?

closed 2019-02-14 by julianoes

JonasVautherin, 2019-02-14:
@zulufoxtrot: would you mind testing your code against the latest `develop` branch before we release the patch? Otherwise we can directly release, but it would be nicer if you confirm the fix first :-)

zulufoxtrot, 2019-02-14:
Sure, I'll try and report back soon.

zulufoxtrot, 2019-02-14:
Been running for 30 minutes, about 30 uploads, no crash so far.