mavlink/MAVSDK-Python #47: AttributeError: type object 'GpsInfo' has no attribute 'FixType'

zulufoxtrot, 2019-02-13:
Getting this error on the latest commit.

GpsInfo appears to be the only thing that's affected. All my other telemetry callbacks work as expected.

```
Traceback (most recent call last):
  File "Autopilot.py", line 196, in get_gps_info
    async for gps_info in Autopilot.vehicle.telemetry.gps_info():
  File "/Users/youri/docs/boulot/TBC/controleur/controleur/Libs/DronecodeSDK-Python/dronecode_sdk/plugins/telemetry.py", line 903, in gps_info
    yield GpsInfo.translate_from_rpc(response.gps_info)
  File "/Users/youri/docs/boulot/TBC/controleur/controleur/Libs/DronecodeSDK-Python/dronecode_sdk/plugins/telemetry.py", line 465, in translate_from_rpc
    GpsInfo.FixType.translate_from_rpc(rpcGpsInfo.fix_type)
AttributeError: type object 'GpsInfo' has no attribute 'FixType'
```

Source:

```
        async for gps_info in Autopilot.vehicle.telemetry.gps_info():

            logger.info(f"GPS info: {gps_info}")
```

julianoes, 2019-02-13:
@JonasVautherin that one is for you :smile:

zulufoxtrot, 2019-02-14:
In plugins/telemetry.py, line 465:

Changing 

`  GpsInfo.FixType.translate_from_rpc(rpcGpsInfo.fix_type)
`
to
`  FixType.translate_from_rpc(rpcGpsInfo.fix_type)
`
fixes it.

The FixType class is outside of GpsInfo.

JonasVautherin, 2019-02-14:
Right, thanks for that! :blush: 

Looking into the generation script to understand what happened there. Sorry for the delay, I'll fix it in the afternoon :slightly_smiling_face:.

JonasVautherin, 2019-02-14:
So it's a bit more complicated than I thought: it has to do with the hierarchy tree of a type. `GpsInfo.FixType` is a bug, it should generate only `FixType`. But then, `Result`, which is nested inside `<plugin>Result`, must become `<plugin>Result.Result` in Python.

Which means that a `type_info` should contain the hierarchy of that type. I'll try to add this tonight!

zulufoxtrot, 2019-02-20:
Shouldn't there also be a commit to DronecodeSDK-Python to reference the latest commit to DronecodeSDK-proto?

JonasVautherin, 2019-02-20:
Yes, sorry I was not clear in the other PR. Opening the new one now.

PR #51 (2019-02-21): fix: use parent_type from type_info when translating struct
* Fix generation for nested structs.
* Use `file_ext` option when running protoc, following latest changes in dcsdkgen

@zulufoxtrot: FYI

Resolves #47.
closed 2019-02-21 by JonasVautherin