mavlink/MAVSDK-Python #48: Python script fails

joe-marrone, 2019-02-13:
I just completed the build of DronecodeSDK-Python as per directions I got from the dev team.   I did all the updates and installed Python 3.6, and everything appeared to install OK.

The GUI runs and  I can get the drone to takeoff and land with manual commands.  The backend is installed and works fine.  But I just can't get the script to run. I tried to run the sample script but it threw an error.  Can someone look at this and see what the problem might be?   

Thanks,
Joe

```
joe-pl@joe-PL:/src/DronecodeSDK/DronecodeSDK-Python$  python3.6 examples/takeoff_and_land.py
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 7, in <module>
    drone = dronecode_sdk_connect(host="127.0.0.1")
  File "/src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 49, in connect
    globals()[plugin](plugin_manager)
KeyError: 'Action'
```

JonasVautherin, 2019-02-13:
Can you show me your `./other/tools/run_protoc.sh` and the output when you run it?

joe-marrone, 2019-02-14:
Hi @julianoes : here are the files:
1) ./other/tools/run_protoc.sh
```
joe-pl@joe-PL:/src/DronecodeSDK/DronecodeSDK-Python$ ./other/tools/run_protoc.sh[+] Installing the DronecodeSDK autogenerator
Processing /src/DronecodeSDK/DronecodeSDK-Python/proto/pb_plugins
Requirement already satisfied: protobuf in /usr/local/lib/python3.6/dist-packages (from dcsdkgen==0.1a0) (3.6.1)
Requirement already satisfied: jinja2 in /usr/lib/python3/dist-packages (from dcsdkgen==0.1a0) (2.8)
Requirement already satisfied: setuptools in /home/joe-pl/.local/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf->dcsdkgen==0.1a0) (1.10.0)
Requirement already satisfied: MarkupSafe in /usr/lib/python3/dist-packages (from jinja2->dcsdkgen==0.1a0) (0.23)
Building wheels for collected packages: dcsdkgen
  Building wheel for dcsdkgen (setup.py) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-1mg27exa/wheels/56/6a/f9/442f17b77f0159caa7b73f548be53ff2c07476f9d8b7d9736b
Successfully built dcsdkgen
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
./other/tools/run_protoc.sh: line 11: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
./other/tools/run_protoc.sh: line 16: python3.6.6: command not found
sed: can't read /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/generated/gimbal_pb2_grpc.py: No such file or directory
 -> [+] Generated protobuf and gRPC bindings for gimbal
./other/tools/run_protoc.sh: line 36: python3.6.6: command not found
./other/tools/run_protoc.sh: line 48: python3.6.6: command not found
mv: cannot move '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/' to a subdirectory of itself, '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/gimbal.py'
./other/tools/run_protoc.sh: line 52: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
 -> [+] Generated plugin for gimbal
./other/tools/run_protoc.sh: line 16: python3.6.6: command not found
sed: can't read /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/generated/info_pb2_grpc.py: No such file or directory
 -> [+] Generated protobuf and gRPC bindings for info
./other/tools/run_protoc.sh: line 36: python3.6.6: command not found
./other/tools/run_protoc.sh: line 48: python3.6.6: command not found
mv: cannot move '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/' to a subdirectory of itself, '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/info.py'
./other/tools/run_protoc.sh: line 52: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
 -> [+] Generated plugin for info
./other/tools/run_protoc.sh: line 16: python3.6.6: command not found
sed: can't read /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/generated/mission_pb2_grpc.py: No such file or directory
 -> [+] Generated protobuf and gRPC bindings for mission
./other/tools/run_protoc.sh: line 36: python3.6.6: command not found
./other/tools/run_protoc.sh: line 48: python3.6.6: command not found
mv: cannot move '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/' to a subdirectory of itself, '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/mission.py'
./other/tools/run_protoc.sh: line 52: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
 -> [+] Generated plugin for mission
./other/tools/run_protoc.sh: line 16: python3.6.6: command not found
sed: can't read /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/generated/camera_pb2_grpc.py: No such file or directory
 -> [+] Generated protobuf and gRPC bindings for camera
./other/tools/run_protoc.sh: line 36: python3.6.6: command not found
./other/tools/run_protoc.sh: line 48: python3.6.6: command not found
mv: cannot move '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/' to a subdirectory of itself, '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/camera.py'
./other/tools/run_protoc.sh: line 52: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
 -> [+] Generated plugin for camera
./other/tools/run_protoc.sh: line 16: python3.6.6: command not found
sed: can't read /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/generated/core_pb2_grpc.py: No such file or directory
 -> [+] Generated protobuf and gRPC bindings for core
./other/tools/run_protoc.sh: line 36: python3.6.6: command not found
./other/tools/run_protoc.sh: line 48: python3.6.6: command not found
mv: cannot move '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/' to a subdirectory of itself, '/src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/core.py'
./other/tools/run_protoc.sh: line 52: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
[... cut ...]
        echo " -> [+] Generated plugin for ${PROTO_IMPORT_NAME%_*}"

    done
}

function install_dcsdkgen {
    cd ${PROTO_DIR}/pb_plugins
    if [[ "$VIRTUAL_ENV" != "" ]]
    then
        pip3.6 install .
    else
        pip3.6  install --user .
    fi
}

echo "[+] Installing the DronecodeSDK autogenerator"
install_dcsdkgen
echo "[+] Done"
echo "[+] Generating plugins from "
generate
echo "[+] Done"
```
3) `joe-pl@joe-PL:/src/DronecodeSDK/DronecodeSDK-Python$  python3.6 --version
Python 3.6.8`

joe-marrone, 2019-02-14:
Jonas:  Julian asked for this on the forum and I already sent it.
Thanks,
Joe

JonasVautherin, 2019-02-14:
As you can see in the output you paster, you have errors when running the script:

```
./other/tools/run_protoc.sh: line 16: python3.6.6: command not found
```

Interestingly, I see `python3.6` on line 16 in the script you pasted. Are you 100% sure that the script you pasted is the one you ran above?

Also, I see that your `protoc.sh` script still has a `python3` that was not replaced by `python3.6`.

Finally, you have permission issues, as said in the output:

```
./other/tools/run_protoc.sh: line 11: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
```

What are the permissions on that file (can you `$ ls -l path/to/__init__.py`)? On what system are you running? Are you in some docker container by any chance?

joe-marrone, 2019-02-14:
Yes, the version of `./other/tools/run_protoc.sh` is the one I am using and it does say python3.6 on line 16, etc.

Permissions
`joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk$ ls -l __init__.py`
`-rw-rw-r-- 1 joe-pl joe-pl 1227 Dec 13 16:06 __init__.py`

Running ubuntu 16.04 LTS on PC.  This is not a virtual machine.
I am not using docker

Text of __init__.py
```
# -*- coding: utf-8 -*-

# List of the core plugins
CORE_PLUGINS = [
    "Action",
    "Calibration",
    "Camera",
    "Gimbal",
    "Info",
    "Mission",
    "Telemetry"
    ]

# Check for compatibility
import platform
if float(".".join(platform.python_version_tuple()[0:-1])) < 3.6:
    print("[!] DronecodeSDK-Python is only available on Python >= 3.6")
    import sys
    sys.exit(1)

# Do asyncio specific initialization
import asyncio
try:
    # Try to import uvloop, provides _MUCH_ better performance compared to the
    # standart unix selector event loop
    import uvloop
    asyncio.set_event_loop_policy(uvloop.EventLoopPolicy())
except ImportError:
    # No uvloop installed on the system; the default eventloop works as well!
    pass

def get_event_loop():
    """ Asyncio eventloop """
    return asyncio.get_event_loop()

# Plugins rely on the eventloop
from .async_plugin_manager import AsyncPluginManager
from .plugins import *

def connect(*args, **kwargs):
    """
    Generates a dronecore instance with all available Core plugins registered
    and ready to use
    """
    plugin_manager = AsyncPluginManager(*args, **kwargs)

    for plugin in CORE_PLUGINS:
        globals()[plugin](plugin_manager)

    return plugin_manager
```

JonasVautherin, 2019-02-15:
> Permissions
`joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk$ ls -l __init__.py`
`-rw-rw-r-- 1 joe-pl joe-pl 1227 Dec 13 16:06 __init__.py`

That's not the right file. The error says:

```
./other/tools/run_protoc.sh: line 52: /src/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/__init__.py: Permission denied
```

Which means that the file you should check for permissions is `src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/plugins/__init__.py`. You checked `src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/__init__.py`.

---------------------------

> Yes, the version of `./other/tools/run_protoc.sh` is the one I am using and it does say python3.6 on line 16, etc.

Well, I cannot help more on this one, you have to figure out what's happening here. With the information I have, it seems that `$ python3.6 --version` works. So if the script calls `python3.6` on line 16, it should not complain about `python3.6.6`. For instance you could try to change `python3.6` for `pythonasfd` on line 16, and check that when you run the script, it complains about `pythonasfd: command not found` and not `python3.6.6`. If it still complains about `python3.6.6`, then it may mean that you are not running the script you think you are running.

joe-marrone, 2019-02-15:
Update:
Re: the `run_protoc.sh` file, ie, it refers to `python` instead of `python3`. I discovered there are THREE instances of this file and they are different. Two use python3. The correct one is: `src/DronecodeSDK/DronecodeSDK-Python/other/tools/run_protoc.sh` 
It ran successfully.
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/dronecodesdk-python/other/tools$ ./run_protoc.sh
[+] Installing the DronecodeSDK autogenerator
Processing /home/joe-pl/src/DronecodeSDK/DronecodeSDK-Python/dronecodesdk-python/proto/pb_plugins
Requirement already satisfied: protobuf in /usr/local/lib/python3.6/dist-packages (from dcsdkgen==0.1a0) (3.6.1)
Requirement already satisfied: jinja2 in /usr/lib/python3/dist-packages (from dcsdkgen==0.1a0) (2.8)
Requirement already satisfied: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf->dcsdkgen==0.1a0) (1.10.0)
Requirement already satisfied: setuptools in /home/joe-pl/.local/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: MarkupSafe in /usr/lib/python3/dist-packages (from jinja2->dcsdkgen==0.1a0) (0.23)
Building wheels for collected packages: dcsdkgen
  Building wheel for dcsdkgen (setup.py) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-36alh1gt/wheels/6a/6d/3f/4f229c63cf9563f68cf6c5b41ebafcd2fd646445c3feaf7301
Successfully built dcsdkgen
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for gimbal
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for info
 -> [+] Generated plugin for info
 -> [+] Generated protobuf and gRPC bindings for mission
 -> [+] Generated plugin for mission
 -> [+] Generated protobuf and gRPC bindings for camera
 -> [+] Generated plugin for camera
 -> [+] Generated protobuf and gRPC bindings for core
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for calibration
 -> [+] Generated plugin for calibration
 -> [+] Generated protobuf and gRPC bindings for discovery
 -> [+] Generated plugin for discovery
 -> [+] Generated protobuf and gRPC bindings for action
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for telemetry
 -> [+] Generated plugin for telemetry
[+] Done
```
However, the result of running the `takeoff_and_land.py` script is the same .. it throws the same error.
```
joe-pl@joe-PL:~/DronecodeSDK-Python/examples$ python3.6 takeoff_and_land.py
Traceback (most recent call last):
  File "takeoff_and_land.py", line 7, in <module>
    drone = dronecode_sdk_connect(host="127.0.0.1")
  File "/src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 49, in connect
    globals()[plugin](plugin_manager)
KeyError: 'Action'
```
So I'm back to square one.  The culprit appears to be `globals()[plugin](plugin_manager)` in th e`__init__.py` file.  What should I look for next?

Thanks, Joe

JonasVautherin, 2019-02-16:
> It ran successfully.

Yes, this time the generation worked!

> I discovered there are THREE instances of this file (why is that?) and they are different. Two use python3. The correct one is: `src/DronecodeSDK/DronecodeSDK-Python/other/tools/run_protoc.sh`

Can you show me the path to the other two? Are you sure you don't have multiple clones of the repo (since you started over a few times, I could imagine that)?

You say:

> The correct one is: `src/DronecodeSDK/DronecodeSDK-Python/other/tools/run_protoc.sh`

But then you run `joe-pl@joe-PL:~/DronecodeSDK-Python/examples$ python3.6 takeoff_and_land.py`. Shouldn't you run something like `src/DronecodeSDK/DronecodeSDK-Python/examples/takeoff_and_land.py` instead?

joe-marrone, 2019-02-19:
Jonas, this is the information from my files:
```
~src/DronecodeSDK/DronecodeSDK-Python/examples/takeoff_and_land.py
~src/DronecodeSDK/DronecodeSDK-Python/other/tools/run_protoc.sh (calls for python3.6)
~src/DronecodeSDK/DronecodeSDK-Python/dronecodesdk-python/other/tools/run_protoc.sh (calls for python3)
~src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/__init__.py
```

There is also another similar directory:
```
~/DronecodeSDK-Python/other/tools/run_protoc.sh  (uses python 3)
~/DronecodeSDK-Python/examples/takeoff_and_land
```

I think there are too many duplicates of these files, that it gets confusing as to which one to use. Is there some way to simplify things so I know what path to use?

No matter which directory I use the result is the same.

1)
 `~src/DronecodeSDK/DronecodeSDK-Python/`
`/examples/`      when I run the script from this directory it fails:

```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 takeoff_and_land.py
Traceback (most recent call last):
  File "takeoff_and_land.py", line 7, in <module>
    drone = dronecode_sdk_connect(host="127.0.0.1")
  File "/src/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 49, in connect
    globals()[plugin](plugin_manager)
KeyError: 'Action'
```

`/other/tools/run_protoc.sh` (this calls for python3.6)

```
#!/bin/bash
SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
WORK_DIR="${SCRIPT_DIR}/../../"
PROTO_DIR="${WORK_DIR}/proto"
GENERATED_DIR="${WORK_DIR}/dronecode_sdk/generated"
PLUGIN_DIR="${WORK_DIR}/dronecode_sdk/plugins"
PLUGIN_INIT="${PLUGIN_DIR}/__init__.py"
export TEMPLATE_PATH="${WORK_DIR}/other/templates/"

function generate {
    echo -e "# -*- coding: utf-8 -*-\n" > $PLUGIN_INIT

    for PROTO_FILE in `find ${PROTO_DIR} -name "*.proto" -type f`; do

        # Generate bindings for each file individually
        python3.6 -m grpc_tools.protoc -I${GENERATED_DIR} \
                                     --proto_path=$(dirname ${PROTO_FILE}) \
                                     --python_out=${GENERATED_DIR} \
                                     --grpc_python_out=${GENERATED_DIR} \
                                     ${PROTO_FILE}

        # For some reason the import is broken with Python3.5.x and works fine
        # with Python3.6.x, set an absolute path and everything is fine
        PROTO_IMPORT_NAME="$(basename -- ${PROTO_FILE%.*})_pb2"

        # We need to create the .original backup files, otherwise we're not compatible with
        # BSD sed.
        sed -i'.sedoriginal' -e "s/import ${PROTO_IMPORT_NAME}/from . import ${PROTO_IMPOR$
            "${GENERATED_DIR}/${PROTO_IMPORT_NAME}_grpc.py"
        # Clean up the backup files.
        find ${GENERATED_DIR} -name '*.sedoriginal' -delete

        echo " -> [+] Generated protobuf and gRPC bindings for ${PROTO_IMPORT_NAME%_*}"

        # Generate plugin
[... 90 lines cut ...]

        # Add to imports
        echo "from .${WANTED_PLUGIN_NAME%.py} import *" >> $PLUGIN_INIT
        echo " -> [+] Generated plugin for ${PROTO_IMPORT_NAME%_*}"

    done
}

function install_dcsdkgen {
    cd ${PROTO_DIR}/pb_plugins
    if [[ "$VIRTUAL_ENV" != "" ]]
    then
        pip3 install .
    else
        pip3  install --user .
    fi
}

echo "[+] Installing the DronecodeSDK autogenerator"
install_dcsdkgen
echo "[+] Done"
echo "[+] Generating plugins from "
generate
echo "[+] Done"
```

JonasVautherin, 2019-02-19:
You apparently have two clones of the repo. Just keep the one that works (`~/src/DronecodeSDK/DronecodeSDK-Python`) and remove the one that doesn't (`~/DronecodeSDK-Python`).

After you run `run_protoc.sh`, you probably need to [install the SDK](https://github.com/dronecode/dronecodesdk-python#install-the-package-locally) again. Let's try to run this:

```sh
cd ~/src/DronecodeSDK/DronecodeSDK-Python
./other/tools/run_protoc.sh
pip3.6 install -e .
python3.6 ./examples/takeoff_and_land.py
```

joe-marrone, 2019-02-19:
I think  messed up.
Instead of removing  ~/DronecodeSDK-Python I removed ~/DronecodeSDK.
Then it failed at pip3.r install -e .
So now it looks like I'll have to go further back to repair the damage.  Will try tomorrow.
```
joe-pl@joe-PL:~$ rm -r DronecodeSDK
joe-pl@joe-PL:~$ ls
catkin_ws                                        Public
Desktop                                          S
Documents                                        SDK_Extensions
Downloads                                        shares
DronecodeSDK-Python                              src
env                                              Templates
eProsima_FastCDR-1.0.7-Linux                     test_page.md
eProsima_FastRTPS-1.5.0-Linux                    ubuntu_sim_common_deps.sh
examples.desktop                                 ubuntu_sim_nuttx.sh
gcc-arm-none-eabi-7-2017-q4-major                ubuntu_sim_ros_gazebo.sh
gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2  ubuntu_sim.sh
Music                                            venv
Navy STTR N19A-T019                              Videos
Pictures                                         VirtualBox VMs
```
```
joe-pl@joe-PL:~$ cd ~/src/DronecodeSDK/DronecodeSDK-Python
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ ./other/tools/run_protoc.sh
[+] Installing the DronecodeSDK autogenerator
Processing /home/joe-pl/src/DronecodeSDK/DronecodeSDK-Python/proto/pb_plugins
Requirement already satisfied: protobuf in /usr/local/lib/python3.6/dist-packages (from dcsdkgen==0.1a0) (3.6.1)
Requirement already satisfied: jinja2 in /usr/lib/python3/dist-packages (from dcsdkgen==0.1a0) (2.8)
Requirement already satisfied: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf->dcsdkgen==0.1a0) (1.10.0)
Requirement already satisfied: setuptools in /home/joe-pl/.local/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: MarkupSafe in /usr/lib/python3/dist-packages (from jinja2->dcsdkgen==0.1a0) (0.23)
Building wheels for collected packages: dcsdkgen
  Building wheel for dcsdkgen (setup.py) ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-lrumjcyy/wheels/01/25/39/425b70549d8a866d68261b6bb1fb9d84d57423152fc1f22e9b
Successfully built dcsdkgen
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for gimbal
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for info
 -> [+] Generated plugin for info
 -> [+] Generated protobuf and gRPC bindings for mission
 -> [+] Generated plugin for mission
 -> [+] Generated protobuf and gRPC bindings for camera
 -> [+] Generated plugin for camera
 -> [+] Generated protobuf and gRPC bindings for core
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for calibration
 -> [+] Generated plugin for calibration
 -> [+] Generated protobuf and gRPC bindings for discovery
 -> [+] Generated plugin for discovery
 -> [+] Generated protobuf and gRPC bindings for action
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for telemetry
 -> [+] Generated plugin for telemetry
[+] Done
```
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ pip3.6 install -e .
Obtaining file:///home/joe-pl/src/DronecodeSDK/DronecodeSDK-Python
Requirement already satisfied: protobuf>=3.5.1 in /usr/local/lib/python3.6/dist-packages (from dronecode-sdk==0.1.0a0) (3.6.1)
Requirement already satisfied: aiogrpc>=1.5 in /usr/local/lib/python3.6/dist-packages (from dronecode-sdk==0.1.0a0) (1.5)
Requirement already satisfied: grpcio>=1.11.0 in /usr/local/lib/python3.6/dist-packages (from dronecode-sdk==0.1.0a0) (1.17.1)
Requirement already satisfied: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf>=3.5.1->dronecode-sdk==0.1.0a0) (1.10.0)
[... 16 lines cut ...]
    The installation directory you specified (via --install-dir, --prefix, or
    the distutils default setting) was:
    
        /usr/local/lib/python3.6/dist-packages/
    
    Perhaps your account does not have write access to this directory?  If the
    installation directory is a system-owned directory, you may need to sign in
    as the administrator or "root" account.  If you do not have administrative
    access to this machine, you may wish to choose a different installation
    directory, preferably one that is listed in your PYTHONPATH environment
    variable.
    
    For information on other options, you may wish to consult the
    documentation at:
    
      https://setuptools.readthedocs.io/en/latest/easy_install.html
    
    Please make the appropriate changes for your system and try again.
    
    
    ----------------------------------------
  Can't roll back dronecode-sdk; was not uninstalled
Command "/usr/bin/python3.6 -c "import setuptools, tokenize;__file__='/home/joe-pl/src/DronecodeSDK/DronecodeSDK-Python/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" develop --no-deps" failed with error code 1 in /home/joe-pl/src/DronecodeSDK/DronecodeSDK-Python/
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$
```

JonasVautherin, 2019-02-20:
Seems like you removed the wrong one, indeed :sweat_smile:.

Still, `run_protoc.sh` worked, so I assumed it is referring to python3.6 and pip3.6. The one that fails is:

```
pip3.6 install -e .
```

Because of a `Permission denied` error. I am not sure, but it looks like you may have to run it as `sudo`:

```
sudo pip3.6 install -e .
```

joe-marrone, 2019-02-20:
EUREKA!  It finally worked, despite the fact that I deleted the wrong file. I was able to use `sudo pip3.6 install -e .` and it installed correctly. After that running the script worked for the first time.  Thank you so much for your patience and assistance.  

My only question is what will the result be of deleting `~/DronecodeSDK` instead of `~/DronecodeSDK-Python`?  So far I haven't seen any issues, but I wonder.

Thanks again,
Joe

IeiuniumLux, 2019-03-08:
My apologies for using a close issue. However, it's the only post I found about issues running run_protoc.sh

In my case, it cannot generate the plugins.  The server builds and works fine, but I'm not sure why run_protoc.sh fails.

`[+] Installing the DronecodeSDK autogenerator
Processing /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/proto/pb_plugins
Requirement already satisfied: protobuf in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (3.7.0)
Requirement already satisfied: jinja2 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (2.10)
Requirement already satisfied: six>=1.9 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (1.12.0)
Requirement already satisfied: setuptools in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: MarkupSafe>=0.23 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jinja2->dcsdkgen==0.1a0) (1.1.0)
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
  Running setup.py install for dcsdkgen ... done
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for action
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Action.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/action.py: No such file or directory
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for calibration
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Calibration.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/calibration.py: No such file or directory
 -> [+] Generated plugin for calibration
 -> [+] Generated protobuf and gRPC bindings for camera
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Camera.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/camera.py: No such file or directory
 -> [+] Generated plugin for camera
 -> [+] Generated protobuf and gRPC bindings for core
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Core.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/core.py: No such file or directory
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for discovery
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Discovery.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/discovery.py: No such file or directory
 -> [+] Generated plugin for discovery
 -> [+] Generated protobuf and gRPC bindings for gimbal
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Gimbal.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/gimbal.py: No such file or directory
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for info
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Info.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/info.py: No such file or directory
 -> [+] Generated plugin for info
 -> [+] Generated protobuf and gRPC bindings for mission
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Mission.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/mission.py: No such file or directory
 -> [+] Generated plugin for mission
 -> [+] Generated protobuf and gRPC bindings for telemetry
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Telemetry.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/telemetry.py: No such file or directory
 -> [+] Generated plugin for telemetry
[+] Done
`

JonasVautherin, 2019-03-08:
@IeiuniumLux: I started a new issue with your question in #55.