mavlink/MAVSDK-Python #50: Python Script Errors

joe-marrone, 2019-02-20:
Now that I have the DronecodeSDK-Python simulation environment running, I have been trying some of the example scripts.  `takeoff_and_land.py` works.  However, I tried several of the others and they fail.  Here's two examples.
1)
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 mission.py
Task exception was never retrieved
future: <Task finished coro=<run() done, defined at mission.py:11> exception=TypeError('__init__() takes 10 positional arguments but 11 were given',)>
Traceback (most recent call last):
  File "mission.py", line 22, in run
    float('nan')))
TypeError: __init__() takes 10 positional arguments but 11 were given

^CTraceback (most recent call last):
  File "mission.py", line 83, in <module>
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt

^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 40, in _python_exit
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```
2)
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 calibration.py
-- Starting gyro calibration
Traceback (most recent call last):
  File "calibration.py", line 21, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "calibration.py", line 13, in run
    async for progress_data in drone.calibration.calibrate_gyro():
AttributeError: 'NoneType' object has no attribute 'calibrate_gyro'
```
Do they fail because they have to be used in conjunction with another script?  Can I run several scripts at the same time?  Looking for an explanation so I can be prepared to start writing some mission scripts.

Thanks,
Joe

JonasVautherin, 2019-02-20:
Seems to me like you are not up-to-date with the `proto` submodule.

If you run `git status`, you should see that `proto` appears in red as "modified". This means that you should update it:

```
git submodule update --recursive
```

Then, because the proto files have changed, you need to generate the code again, and install the SDK again.

```
./other/tools/run_protoc.sh
sudo pip3.6 install -e .
```

Notes:
* `./other/tools/run_protoc.sh` will also appear as "modified", because you changed "python3" and "pip3" to "python3.6" and "pip3.6". That's normal on Ubuntu 16.04.
* The whole process is still a bit annoying for you to use (i.e. you have to generate and install after each update), and we are working on making this better :-).

joe-marrone, 2019-02-20:
Jonas, I tried this:
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/proto$ git status
HEAD detached at 5053b6c
nothing to commit, working directory clean
```
So then  did this and nothing happened:
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/proto$ git submodule update --recursive
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/proto$ 
```
Ran the `mission.py` script and this time it worked!
Is HEAD detached anything to worry about?

I didn't rerun the following, but that didn't seem to matter.
```
/other/tools/run_protoc.sh
sudo pip3.6 install -e .
```
I'm really excited to finally see a mission running on the sim, and will be getting deeper into this aspect now.

Thanks,
Joe

JonasVautherin, 2019-02-20:
You would have to run `git submodule update --recursive` from the root of the repo, so from `~/src/DronecodeSDK/DronecodeSDK-Python`.

> Finally, I ran the `mission.py` script and this time it worked!

Hmm, I don't get how that could have happened without regenerating the correct files :thinking:. But well, why not.

Feel free to reopen if the error comes back!

closed 2019-02-20 by JonasVautherin

joe-marrone, 2019-02-21:
Jonas, you tipped me off to something I was unaware of.  It seems that  have duplicate `src/DronecodeSDK/DronecodeSDK-Python` directories in both the **root** and **home** directories.  The mission.py runs from the root, but NOT from home!  This means I still have more than one clone on the system, and they have different behaviors.  So now I run jMAVSim and backend from the home directory, but then have to shift to the root directory to run the Python wrapper. Don't know how I got into this predicament, but is there any way to resolve this short of a total rebuild?
Thanks,
Joe

joe-marrone, 2019-02-21:
Quick update. I copied `mission.py` from the root directory into the home directory, and it worked from the home directory, so this fixed the problem.  This leads me to believe that the scripts are different.
-Joe

JonasVautherin, 2019-02-21:
You should be able to remove the one in the root directory, and just update the one in the user directory. It is failing because the "proto" submodule there is not up-to-date.

You have to update your git clone in the user directory (see the instructions [above](https://github.com/Dronecode/DronecodeSDK-Python/issues/50#issuecomment-465724638)), run `run_protoc.sh` and `sudo pip3.6 install -e .` again.

joe-marrone, 2019-02-21:
Update: The` mission.py` scripts in the $/ and $~ directories are different.  One runs and the other doesn't. I copied the "good" one into the $~ directory, renamed it `mission-good.py`, and it ran OK. Here it is:
```
#from $~src/DronecodeSDK/DronecodeSDK-Python/examples/mission-good.py

#!/usr/bin/env python3

import asyncio

from dronecode_sdk import connect as dronecode_sdk_connect
from dronecode_sdk import (MissionItem, MissionItems)

drone = dronecode_sdk_connect(host="127.0.0.1")

async def run():
    my_items = []
    my_items.append(MissionItem(47.398039859999997, 8.5455725400000002, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))
    my_items.append(MissionItem(47.398036222362471, 8.5450146439425509, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))
    my_items.append(MissionItem(47.397825620791885, 8.5450092830163271, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))

    await drone.mission.setReturnToLaunchAfterMission(True)

    mission_items = MissionItems(my_items)
    upload_mission_status, _ = await drone.mission.uploadMission(mission_items)
    print(f"-- Upload mission: {upload_mission_status}")

    arm_status, _ = await drone.action.arm()
    print(f"-- Arm result: {arm_status}")

    start_mission_status, _ = await drone.mission.startMission()
    print(f"-- Start mission: {start_mission_status}")

async def print_mission_progress():
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: {mission_progress.current_item_index}/{mission_progress.mission_count}")

async def observe_is_in_air():
    """ Monitors whether the drone is flying or not and
    returns after landing """

    was_in_air = False

    async for is_in_air in drone.telemetry.in_air():
        if is_in_air:
            was_in_air = is_in_air

        if was_in_air and not is_in_air:
            await asyncio.get_event_loop().shutdown_asyncgens()
            return

def setup_tasks():
    asyncio.ensure_future(run())
    asyncio.ensure_future(print_mission_progress())

if __name__ == "__main__":
    setup_tasks()
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
```
Then I edited the GPS coordinates to be my location, but it failed.  Here is the modified file and the error message.
```# mission-nh.py

#!/usr/bin/env python3

import asyncio

from dronecode_sdk import connect as dronecode_sdk_connect
from dronecode_sdk import (MissionItem, MissionItems)

drone = dronecode_sdk_connect(host="127.0.0.1")

async def run():
    my_items = []
    my_items.append(MissionItem(42.7122310, -71.4605572, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))
    my_items.append(MissionItem(42.7122400, -71.4605580, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))
#    my_items.append(MissionItem(42.7122500, -71.4605590, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))
    my_items.append(MissionItem(42.7122310, -71.4605572, 25, 10, True, 0, 0, MissionItem.CameraAction.NONE, 0))

    await drone.mission.setReturnToLaunchAfterMission(True)

    mission_items = MissionItems(my_items)
    upload_mission_status, _ = await drone.mission.uploadMission(mission_items)
    print(f"-- Upload mission: {upload_mission_status}")

    arm_status, _ = await drone.action.arm()
    print(f"-- Arm result: {arm_status}")

    start_mission_status, _ = await drone.mission.startMission()
    print(f"-- Start mission: {start_mission_status}")

async def print_mission_progress():
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: {mission_progress.current_item_index}/{mission_progress.mission_count}")

async def observe_is_in_air():
    """ Monitors whether the drone is flying or not and
    returns after landing """

    was_in_air = False

    async for is_in_air in drone.telemetry.in_air():
        if is_in_air:
            was_in_air = is_in_air

        if was_in_air and not is_in_air:
            await asyncio.get_event_loop().shutdown_asyncgens()
            return

def setup_tasks():
    asyncio.ensure_future(run())
    asyncio.ensure_future(print_mission_progress())

if __name__ == "__main__":
    setup_tasks()
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
```
This is the error message:
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 mission1-nh.py
Mission progress: 0/3
-- Upload mission: SUCCESS
-- Arm result: SUCCESS
-- Start mission: ERROR
^CTraceback (most recent call last):
  File "mission1-nh.py", line 58, in <module>
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 40, in _python_exit
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$
```

JonasVautherin, 2019-02-22:
> Update: The` mission.py` scripts in the $/ and $~ directories are different. One runs and the other doesn't. I copied the "good" one into the $~ directory, renamed it `mission-good.py`, and it ran OK.

Your problem is not that the `mission.py` in `$~` is bad, your problem is that it is outdated. The way it works is that "DronecodeSDK-Python" is a git repository, which contains a git submodule called "proto". When you update the DronecodeSDK-Python repository by running e.g. `git pull`, then you also need to update the submodule by running `git submodule update`. This is what I am explaining [above](https://github.com/Dronecode/DronecodeSDK-Python/issues/50#issuecomment-465724638).

> This is the error message

I guess your problem is that your simulated drone is too far away from the waypoints you set, so the drone is refusing to fly the mission. If you had the latest version of the SDK, you would get an exception giving more details (see instructions above).

joe-marrone, 2019-02-22:
Jonas, here is the result of `git status`.  I don't see any reference to `proto` submodule.  
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   examples/mission.py
	modified:   examples/takeoff_and_land.py
	modified:   examples/telemetry_takeoff_and_land.py
	modified:   other/tools/run_protoc.sh

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	dronecodesdk-python/
	examples/calibration.py
	examples/camera.py
	examples/firmware_version.py
	examples/gps_demo.py
	examples/mission-2.py
	examples/mission-good.py
	examples/mission1-nh.py
	examples/telemetry.py

no changes added to commit (use "git add" and/or "git commit -a")
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ 
```
When I execute `git pull` this is the result.  Now I see reference to the `proto` submodule.  I am told to  move or remove these files. Some of them were copied from the /$ directory.  Not sure how to handle this.
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git pull
remote: Enumerating objects: 75, done.
remote: Counting objects: 100% (75/75), done.
remote: Compressing objects: 100% (38/38), done.
remote: Total 75 (delta 41), reused 65 (delta 37), pack-reused 0
Unpacking objects: 100% (75/75), done.
From https://github.com/Dronecode/DronecodeSDK-Python
   ece5a41..62a5041  master     -> origin/master
Fetching submodule proto
remote: Enumerating objects: 87, done.
remote: Counting objects: 100% (87/87), done.
remote: Compressing objects: 100% (39/39), done.
remote: Total 92 (delta 56), reused 65 (delta 44), pack-reused 5
Unpacking objects: 100% (92/92), done.
From https://github.com/Dronecode/DronecodeSDK-Proto
   8a683a9..7cf75fe  master     -> origin/master
 * [new branch]      various-fixes -> origin/various-fixes
Updating ece5a41..62a5041
error: Your local changes to the following files would be overwritten by merge:
	examples/mission.py
	examples/takeoff_and_land.py
	examples/telemetry_takeoff_and_land.py
	other/tools/run_protoc.sh
Please, commit your changes or stash them before you can merge.
error: The following untracked working tree files would be overwritten by merge:
	examples/calibration.py
	examples/camera.py
	examples/firmware_version.py
	examples/telemetry.py
Please move or remove them before you can merge.
Aborting
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ 
```
Without doing anything about moving or removing files, I ran this with no specific message.
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ 
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git submodule update
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ 
```
You'll have to forgive me but I'm not a programmer and have been following the build as if it were a cookbook procedure, and right now I know enough to be dangerous. Sorry to use up so much of your time. I will look for a tutorial on using git repositories to get familiar with how to use them.
Thanks,
Joe

JonasVautherin, 2019-02-22:
No worries. We will improve the SDK installation procedure ASAP.

First, make a copy the example files you modified:

* examples/mission.py
* examples/takeoff_and_land.py
* examples/telemetry_takeoff_and_land.py

Since you don't know git (yet), I would advise you not to edit those files but to create new ones (which you did for others like `examples/camera.py` and `examples/mission-2.py`). You could for instance copy all _your_ examples into `myexamples` and work from there.

Once you have saved your work (the 3 files listed above), you can run, from `~/src/DronecodeSDK/DronecodeSDK-Python`:

```
git add other/tools/run_protoc.sh
git commit -m "Make run_protoc.sh run with python3.6 instead of python3"
git stash
git pull --rebase
git submodule update
cd proto/pb_plugins
sudo pip3.6 install -e .
cd ../..
./other/tools/run_protoc.sh
sudo pip3.6 install -e .
python3.6 examples/mission.py
```

This should update the repo, install the new version of the generator plugin, generate the new SDK, install it, and then run the mission example.

Look at the output of every single one of those commands. If it fails, don't go further and paste the error message here.

Good luck ;-)

joe-marrone, 2019-02-22:
Feedback on the procedure above:  the second step failed.
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git add other/tools/run_protoc.sh
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git commit -m "Make run_protoc.sh run with python3.6 instead of python3"

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'joe-pl@joe-PL.(none)')
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ 
```

Joe

joe-marrone, 2019-02-22:
I decided to put my email and name in and that worked.  I kept going but encountered another error.
-Joe
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git add other/tools/run_protoc.sh
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git commit -m "Make run_protoc.sh run with python3.6 instead of python3"

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: unable to auto-detect email address (got 'joe-pl@joe-PL.(none)')
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git config --global user.email "jmarrone@phasorlab.com"
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git config --global user.name "Joe_Marrone"
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git commit -m "Make run_protoc.sh run with python3.6 instead of python3"
[master 58dce81] Make run_protoc.sh run with python3.6 instead of python3
 1 file changed, 4 insertions(+), 4 deletions(-)
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git stash
Saved working directory and index state WIP on master: 58dce81 Make run_protoc.sh run with python3.6 instead of python3
HEAD is now at 58dce81 Make run_protoc.sh run with python3.6 instead of python3
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ git pull --rebase
First, rewinding head to replay your work on top of it...
error: The following untracked working tree files would be overwritten by checkout:
	examples/calibration.py
	examples/camera.py
	examples/firmware_version.py
	examples/telemetry.py
Please move or remove them before you can switch branches.
Aborting
could not detach HEAD
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python$ 
```

JonasVautherin, 2019-02-22:
```
error: The following untracked working tree files would be overwritten by checkout:
	examples/calibration.py
	examples/camera.py
	examples/firmware_version.py
	examples/telemetry.py
Please move or remove them before you can switch branches.
```

So you should move those files somewhere out of the repository. Like in `~/myexamples`, or whatever. By mixing your code with the SDK repository, you just create conflicts for git.

joe-marrone, 2019-02-25:
Update: I went thru the steps above and everything appeared to install OK.  When I ran the `mission.py` example it failed.
This is the jMAVim output.
```
pxh> WARN  [mavlink] ERROR: wp index out of bounds
ERROR [mavlink] ERROR: wp index out of bounds
WARN  [mavlink] ERROR: wp index out of bounds
ERROR [mavlink] ERROR: wp index out of bounds
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-02-25/14_53_33.ulg
INFO  [logger] Start file log (type: mission)
INFO  [logger] Opened mission log file: ./mission_log/2019-02-25/14_53_33.ulg
INFO  [commander] Returning to launchhis is the 
INFO  [navigator] RTL HOME activated
INFO  [logger] closed logfile, bytes written: 2948
INFO  [logger] closed logfile, bytes written: 554033
```
This is the `mission.py` output.
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 mission.py
-- Uploading mission
-- Arming
-- Starting mission
^CTraceback (most recent call last):
  File "mission.py", line 83, in <module>
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 40, in _python_exit
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```
This is the `mission.py` script.  This is the original script ... I didn't make any changes.
```
#!/usr/bin/env python3

import asyncio

from dronecode_sdk import connect as dronecode_sdk_connect
from dronecode_sdk import (MissionItem)

drone = dronecode_sdk_connect(host="127.0.0.1")

async def run():
    mission_items = []
    mission_items.append(MissionItem(47.398039859999997,
                                     8.5455725400000002,
                                     25,
                                     10,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.398036222362471,
                                     8.5450146439425509,
                                     25,
                                     10,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan')))
    mission_items.append(MissionItem(47.397825620791885,
                                     8.5450092830163271,
                                     25,
                                     10,
                                     True,
                                     float('nan'),
                                     float('nan'),
                                     MissionItem.CameraAction.NONE,
                                     float('nan'),
                                     float('nan')))

    await drone.mission.set_return_to_launch_after_mission(True)

    print("-- Uploading mission")
    await drone.mission.upload_mission(mission_items)

    print("-- Arming")
    await drone.action.arm()

    print("-- Starting mission")
    await drone.mission.start_mission()

async def print_mission_progress():
    async for mission_progress in drone.mission.mission_progress():
        print(f"Mission progress: {mission_progress.current_item_index}/{mission_progress.mission_count}")

async def observe_is_in_air():
    """ Monitors whether the drone is flying or not and
    returns after landing """

    was_in_air = False

    async for is_in_air in drone.telemetry.in_air():
        if is_in_air:
            was_in_air = is_in_air

        if was_in_air and not is_in_air:
            await asyncio.get_event_loop().shutdown_asyncgens()
            return

def setup_tasks():
    asyncio.ensure_future(run())
    asyncio.ensure_future(print_mission_progress())

if __name__ == "__main__":
    setup_tasks()
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
```

JonasVautherin, 2019-02-25:
It looks like it was working, and you interrupted it with `Ctrl + C`, doesn't it?

```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 mission.py
-- Uploading mission
-- Arming
-- Starting mission
^C
```

joe-marrone, 2019-02-25:
Yes I did interrupt it because it stopped working at that point and the system hung up. The only way I could do anything was to ^C.  Here is what happens:
You can see the mission starting in jMAVSIM, but it almost immediately does RTL.
```
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-02-25/22_03_21.ulg
INFO  [logger] Start file log (type: mission)
INFO  [logger] Opened mission log file: ./mission_log/2019-02-25/22_03_21.ulg
INFO  [commander] Returning to launch
INFO  [navigator] RTL HOME activated
INFO  [logger] closed logfile, bytes written: 2985
INFO  [logger] closed logfile, bytes written: 549657
```
Same thing happens in the backend:
```
[05:03:21|Debug] Send mission item 0 (mission_impl.cpp:1003)
[05:03:21|Info ] Mission accepted (mission_impl.cpp:163)
[05:03:21|Debug] MAVLink: info: ARMED by arm/disarm component command (system_impl.cpp:280)
[05:03:21|Debug] MAVLink: info: [logger] file: ./log/2019-02-25/22_03_21.ulg (system_impl.cpp:280)
[05:03:21|Debug] MAVLink: info: Returning to launch (system_impl.cpp:280)
[05:03:21|Debug] MAVLink: info: RTL HOME activated (system_impl.cpp:280)
[05:03:31|Debug] MAVLink: info: DISARMED by auto disarm on land (system_impl.cpp:280)
```
So the Python script only goes to that point and then stops:
```
joe-pl@joe-PL:~$ cd src/DronecodeSDK/DronecodeSDK-Python/examples/
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 mission.py
-- Uploading mission
-- Arming
-- Starting mission
```
As I say, the only way I can get it out of this condition is to ^C.  What could be happening?

Joe

JonasVautherin, 2019-02-25:
> You can see the mission starting in jMAVSIM, but it almost immediately does RTL.

So it takes off?

joe-marrone, 2019-02-26:
No, it does not take off.

JonasVautherin, 2019-02-26:
Do the other examples work?

joe-marrone, 2019-02-26:
The following scripts run and appear to work OK.
```
takeoff_and_land.py
calibration.py (but it failed magnetometer and stopped)
camera.py (but it couldn't take photo - DENIED)
firmware_version.py
telemetry_flight_mode.py
telemetry_is_armed_is_in_air.py
telemetry.py
telemetry_takeoff_and_land.py
```
All of the missions scripts, both the one that came with the distro and the ones I created from the "old" versions, do NOT run.  Evidantly, one of the "old" versions that once ran no linger does.

Hope this helps.
Joe

JonasVautherin, 2019-02-26:
What if you only upload the mission from the SDK, open QGC and start it? Does that work?

joe-marrone, 2019-02-26:
I open QGC and it can run a manually inputted waypoint mission.  But when I try to launch `mission.py` the same thing happens as before and it doesn't execute. And the QGC screen shows no movement of the drone.  
On the latest attempt, jMAVSim said "waypoint index out of bounds".

JonasVautherin, 2019-02-26:
Which version of the backend are you running? When you run the C++ backend, it should show you the version in the log output. Maybe it is outdated.

joe-marrone, 2019-02-26:
How do I find out the version of the backend? Here is some of the backend startup. Does this have the information?
```
joe-pl@joe-PL:~/src/DronecodeSDK$ ./build/default/backend/src/backend_bin
[09:49:15|Info ] DronecodeSDK version: 0.10.2 (dronecode_sdk_impl.cpp:24)
[09:49:15|Debug] New: System ID: 0 Comp ID: 0 (dronecode_sdk_impl.cpp:284)
[09:49:15|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:41)
[09:49:15|Info ] Server started (grpc_server.cpp:25)
[09:49:15|Info ] Waiting to discover system... (connection_initiator.h:57)
[09:49:15|Info ] New device on: 127.0.0.1:14580 (udp_connection.cpp:200)
[09:49:15|Debug] Component Autopilot (1) added. (system_impl.cpp:369)
[09:49:16|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:515)
[09:49:16|Info ] System discovered [UUID: 5283920058631409231] (connection_initiator.h:61)
[09:49:59|Debug] current: -1, total: 0 (mission_impl.cpp:1078)
[09:49:59|Debug] Send mission item 0 (mission_impl.cpp:1003)
[09:49:59|Info ] Mission accepted (mission_impl.cpp:163)
[09:49:59|Debug] MAVLink: critical: ERROR: wp index out of bounds (system_impl.cpp:280)
[09:49:59|Debug] MAVLink: info: ARMED by arm/disarm component command (system_impl.cpp:280)
[09:49:59|Debug] MAVLink: critical: ERROR: wp index out of bounds (system_impl.cpp:280)
[09:49:59|Debug] MAVLink: info: Returning to launch (system_impl.cpp:280)
[09:49:59|Debug] MAVLink: info: RTL HOME activated (system_impl.cpp:280)
[09:50:09|Debug] MAVLink: info: DISARMED by auto disarm on land (system_impl.cpp:280)
[09:52:22|Debug] Send mission item 0 (mission_impl.cpp:1003)
[09:52:22|Info ] Mission accepted (mission_impl.cpp:163)
[09:52:22|Debug] MAVLink: critical: ERROR: wp index out of bounds (system_impl.cpp:280)
[09:52:22|Debug] MAVLink: critical: ERROR: wp index out of bounds (system_impl.cpp:280)
[09:52:22|Debug] MAVLink: info: ARMED by arm/disarm component command (system_impl.cpp:280)
[09:52:22|Debug] MAVLink: info: [logger] file: ./log/2019-02-25/14_52_22.ulg (system_impl.cpp:280)
[09:52:23|Debug] MAVLink: info: Returning to launch (system_impl.cpp:280)
[09:52:23|Debug] MAVLink: info: RTL HOME activated (system_impl.cpp:280)
[09:52:32|Debug] MAVLink: info: DISARMED by auto disarm on land (system_impl.cpp:280)
```

joe-marrone, 2019-02-26:
Another issue: I keep getting this error:
`MAVLink: critical: ERROR: wp index out of bounds (system_impl.cpp:280)`
Is this related to the fact that my location is different than the waypoints in the script?  Sounds like it is related to the home coordinates.

Is this a parameter that can be changed?

Joe

JonasVautherin, 2019-02-26:
> How do I find out the version of the backend?

The first line of the output says: 

> [09:49:15|Info ] DronecodeSDK version: 0.10.2 (dronecode_sdk_impl.cpp:24)

Please update to [0.14.2](https://github.com/Dronecode/DronecodeSDK/releases), either by building it or by installing the `.deb` available there.

joe-marrone, 2019-02-26:
I tried to rebuild DronecodeSDK as per the instructions.  When I got to the backend install it told me there is no `backend_bin` file.
```
joe-pl@joe-PL:~/src/DronecodeSDK$ sudo ldconfig
joe-pl@joe-PL:~/src/DronecodeSDK$ sudo apt-get install golang
Reading package lists... Done
Building dependency tree       
Reading state information... Done
golang is already the newest version (2:1.6-1ubuntu4).
0 upgraded, 0 newly installed, 0 to remove and 52 not upgraded.
joe-pl@joe-PL:~/src/DronecodeSDK$ make BUILD_BACKEND=1
Build dir: build/default
[1/1] cd /home/joe-pl/src/DronecodeSDK/build/default && :
joe-pl@joe-PL:~/src/DronecodeSDK$ ./build/default/backend/src/backend_bin
bash: ./build/default/backend/src/backend_bin: No such file or directory
joe-pl@joe-PL:~/src/DronecodeSDK$ 
```

I searched the entire directory tree and `backend_bin` does not exist.  In fact, there is no `backend/bin` folder in the directory path any more.  Not sure what just happened.  
I continue to be confused about which directory to use when installing this build.  Some of the instructions say to work from the `~/DronecodeSDK` directory, and sometimes it says to install a module in the `~/src/DronecodeSDK` directory.  This is easy to mess up.  So I am not sure everything is in the right place. Right now all that runs is the GUI.

joe-marrone, 2019-02-26:
I tried installing the .deb file, but there is still no `backend_bin` file.  
```
joe-pl@joe-PL:~$ sudo dpkg -i ./Downloads/libdronecode-sdk-dev_0.14.2_ubuntu16.04_amd64.deb
[sudo] password for joe-pl: 
Sorry, try again.
[sudo] password for joe-pl: 
Selecting previously unselected package libdronecode-sdk-dev.
(Reading database ... 269009 files and directories currently installed.)
Preparing to unpack .../libdronecode-sdk-dev_0.14.2_ubuntu16.04_amd64.deb ...
Unpacking libdronecode-sdk-dev (0.14.2) ...
Setting up libdronecode-sdk-dev (0.14.2) ...
Processing triggers for libc-bin (2.23-0ubuntu10) ...
joe-pl@joe-PL:~$ ls
catkin_ws                                        Public
Desktop                                          S
Documents                                        SDK_Extensions
Downloads                                        shares
DronecodeSDK                                     src
env                                              Templates
eProsima_FastCDR-1.0.7-Linux                     test_page.md
eProsima_FastRTPS-1.5.0-Linux                    ubuntu_sim_common_deps.sh
examples.desktop                                 ubuntu_sim_nuttx.sh
fib.py                                           ubuntu_sim_ros_gazebo.sh
gcc-arm-none-eabi-7-2017-q4-major                ubuntu_sim.sh
gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2  venv
Music                                            Videos
Navy STTR N19A-T019                              VirtualBox VMs
Pictures
```
This is the path for installing the backend, and there is no folder called `backend`
```
joe-pl@joe-PL:~/src$ cd DronecodeSDK/build/default/
joe-pl@joe-PL:~/src/DronecodeSDK/build/default$ ls
build.ninja          dronecode_sdkConfig.cmake   install_manifest.txt
CMakeCache.txt       dronecode_sdkTargets.cmake  integration_tests
CMakeFiles           gmock_main.pc               plugins
cmake_install.cmake  gmock.pc                    rules.ninja
core                 gtest_main.pc               third_party
CTestTestfile.cmake  gtest.pc                    unit_tests_runner
joe-pl@joe-PL:~/src/DronecodeSDK/build/default$ 
```
What is happening?

JonasVautherin, 2019-02-26:
Oh, right, maybe it's not in the `.deb`. So you have to build from source:

```sh
make BUILD_BACKEND=YES
```

Then, if the build is successful, it will be in `./build/default/backend/src/backend_bin`.

joe-marrone, 2019-02-26:
Before the .deb install, I did the standard install.  Just now I tried `make BUILD_BACKEND=YES`.  Still no ./build/default/backend/src/backend_bin`

Joe

JonasVautherin, 2019-02-26:
```
make clean
make BUILD_BACKEND=YES
```

joe-marrone, 2019-02-26:
That worked.  Now  have to resolve the WP out of range issue.

JonasVautherin, 2019-02-26:
Can you run the newly-build backend and show me the version number?

And can you try to run `examples/mission.py` again?

joe-marrone, 2019-02-26:
The `mission.py` script still does not run.  As soon as the mission is starts, it gives a RTL command.

joe-marrone, 2019-02-26:
Tthis is the latest backend build. Strangely it doesn't have the latest revision.
```
joe-pl@joe-PL:~$ cd src/DronecodeSDK/
joe-pl@joe-PL:~/src/DronecodeSDK$ ./build/default/backend/src/backend_bin
[05:04:28|Info ] DronecodeSDK version: 0.10.2 (dronecode_sdk_impl.cpp:24)
[05:04:28|Debug] New: System ID: 0 Comp ID: 0 (dronecode_sdk_impl.cpp:284)
[05:04:28|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:41)
[05:04:28|Info ] Server started (grpc_server.cpp:25)
[05:04:28|Info ] Waiting to discover system... (connection_initiator.h:57)
[05:04:28|Info ] New device on: 127.0.0.1:14580 (udp_connection.cpp:200)
[05:04:28|Debug] Component Autopilot (1) added. (system_impl.cpp:369)
[05:04:29|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:515)
[05:04:29|Info ] System discovered [UUID: 5283920058631409231] (connection_initiator.h:61)
```

JonasVautherin, 2019-02-26:
Did you run something like `git pull` before `make BUILD_BACKEND=YES`?

joe-marrone, 2019-02-27:
I did not do `git pull` os I did that, then ran `make BUILD_BACKEND=YES` again.  During this, there were many errors.  The log file is _very_ long so I am inlcuding only th eend part where most of the errors occurred.
```
                          ^
../../backend/src/core/core_service_impl.h:22:28: error: ‘ConnectionStateResponse’ is not a member of ‘dronecode_sdk::rpc::core’
../../backend/src/core/core_service_impl.h:22:62: error: template argument 1 is invalid
         grpc::ServerWriter<rpc::core::ConnectionStateResponse> *writer) override
                                                              ^
../../backend/src/core/core_service_impl.h:67:38: error: ‘ConnectionStateResponse’ in namespace ‘dronecode_sdk::rpc::core’ does not name a type
     static dronecode_sdk::rpc::core::ConnectionStateResponse
                                      ^
../../backend/src/core/core_service_impl.h: In lambda function:
../../backend/src/core/core_service_impl.h:25:99: error: there are no arguments to ‘createRpcConnectionStateResponse’ that depend on a template parameter, so a declaration of ‘createRpcConnectionStateResponse’ must be available [-fpermissive]
             const auto rpc_connection_state_response = createRpcConnectionStateResponse(uuid, true);
                                                                                                   ^
../../backend/src/core/core_service_impl.h:25:99: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
../../backend/src/core/core_service_impl.h:26:21: error: request for member ‘Write’ in ‘writer->’, which is of non-class type ‘int’
             writer->Write(rpc_connection_state_response);
                     ^
../../backend/src/core/core_service_impl.h: In lambda function:
../../backend/src/core/core_service_impl.h:31:61: error: there are no arguments to ‘createRpcConnectionStateResponse’ that depend on a template parameter, so a declaration of ‘createRpcConnectionStateResponse’ must be available [-fpermissive]
                 createRpcConnectionStateResponse(uuid, false);
                                                             ^
../../backend/src/core/core_service_impl.h:32:21: error: request for member ‘Write’ in ‘writer->’, which is of non-class type ‘int’
             writer->Write(rpc_connection_state_response);
                     ^
In file included from backend/src/mission/mission.grpc.pb.h:7:0,
                 from ../../backend/src/plugins/mission/mission_service_impl.h:7,
                 from ../../backend/src/grpc_server.h:15,
                 from ../../backend/src/backend.cpp:7:
backend/src/mission/mission.pb.h: At global scope:
backend/src/mission/mission.pb.h:337:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:443:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:549:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:645:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:761:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:857:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:963:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1059:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1165:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1268:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1374:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1470:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1573:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
   static PROTOBUF_CONSTEXPR int const kIndexInFileMessages =
          ^
backend/src/mission/mission.pb.h:1669:10: error: ‘PROTOBUF_CONSTEXPR’ does not name a type
[... 259 lines cut ...]
[09:56:32|Debug] MAVLink: emergency: Accel #0 fail:  TIMEOUT! (system_impl.cpp:292)
[09:56:32|Debug] MAVLink: emergency: Gyro #0 fail:  TIMEOUT! (system_impl.cpp:292)
[09:56:32|Debug] MAVLink: emergency: Mag #0 fail:  TIMEOUT! (system_impl.cpp:292)
[09:56:32|Debug] MAVLink: emergency: Baro #0 fail:  STALE! (system_impl.cpp:292)
[09:56:33|Debug] MAVLink: info: data link #0 regained (system_impl.cpp:292)
[09:57:08|Debug] MAVLink: info: ARMED by command line (system_impl.cpp:292)
[... cut ...]
[09:59:02|Debug] MAVLink: info: [logger] file: ./log/2019-02-27/14_59_02.ulg (system_impl.cpp:292)
[09:59:02|Debug] MAVLink: info: Returning to launch (system_impl.cpp:292)
[09:59:02|Debug] MAVLink: info: RTL HOME activated (system_impl.cpp:292)
[09:59:12|Debug] MAVLink: info: DISARMED by auto disarm on land (system_impl.cpp:292)

```
The backend does work and I can do a manual takeoff and land, but the `mission.py` still stops before the drone leave the ground, gives a command to RTL, and hangs up at that point.

Joe

joe-marrone, 2019-02-27:
Incidentally, the `takeoff_and_land.py` script worked perfectly.
```
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 takeoff_and_land.py
-- Arming
-- Taking off
-- Landing
joe-pl@joe-PL:~/src/DronecodeSDK/DronecodeSDK-Python/examples$ 
```
here is the backend log:
```
[10:06:30|Debug] MAVLink: info: ARMED by arm/disarm component command (system_impl.cpp:292)
[10:06:30|Debug] MAVLink: info: [logger] file: ./log/2019-02-27/15_06_30.ulg (system_impl.cpp:292)
[10:06:30|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m (system_impl.cpp:292)
[10:06:31|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m (system_impl.cpp:292)
[10:06:31|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:292)
[10:06:35|Debug] MAVLink: info: Landing at current position (system_impl.cpp:292)
[10:06:41|Debug] MAVLink: info: Landing detected (system_impl.cpp:292)
[10:06:41|Debug] MAVLink: info: DISARMED by auto disarm on land (system_impl.cpp:292)
```

Joe

JonasVautherin, 2019-02-27:
I don't get how you got to backend 0.14.2 while having compilation errors.

Anyway, make sure you are on tag 0.14.2 of DronecodeSDK, that the submodules are up-to-date, and that this builds _without errors_. Maybe you need to read a bit about git in order to achieve that.

I believe `examples/mission.py` doesn't work properly because you don't have the latest version of the backend (or maybe even frontend?). So that needs to be fixed.

I'm thinking that because you are not sending the full mission, and that may be due to an outdated backend:

```
[09:59:02|Debug] Send mission item 0 (mission_impl.cpp:1109)
[09:59:02|Info ] Mission accepted (mission_impl.cpp:164)
```

joe-marrone, 2019-02-27:
Jonas, at this point I'm totally lost.  it sounds like I need to do a total rebuild for the third time.
Initially I am not going to do any development work on the inner code of the sim.  What  I am really interested in is developing some Python scripts to test drone missions before flying them.  All I really need now is a functioning sim that allows me to test specific missions.  Is there any simpler way to achieve this?

Thanks for all the help.
Joe

joe-marrone, 2019-02-27:
Jonas: I did some reading about git and discovered something interesting.  There are two repos in my system.  Clearly there are differences between them.  I know this can't be good, and am wondering if this is the cause of the problems I'm running into??  IF  I have to rebuild the entire system, I want to be sure there is only ONE clone, so I have to know where to put it.  Does it go into `/src/DronecodeSDK` or `/DronecodeSDK`? 
First one: 
```joe-pl@joe-PL:**~/src/DronecodeSDK**$ git log --oneline -5
e5d4e0e cmake: fix android build and update grpc dependencies
4e8882d multiple_drones: some minor improvements
777084d udp_connection: check if entry is valid first
16e5f25 udp_connection: keep track of remotes and sysids
a16cf51 example: added example for multiple drones
joe-pl@joe-PL:**~/src/DronecodeSDK**$ git log --oneline --decorate -5
e5d4e0e (HEAD -> develop, origin/develop, origin/HEAD) cmake: fix android build and update grpc dependencies
4e8882d multiple_drones: some minor improvements
777084d udp_connection: check if entry is valid first
16e5f25 udp_connection: keep track of remotes and sysids
a16cf51 example: added example for multiple drones
joe-pl@joe-PL:~/src/DronecodeSDK$ git branch -v
* develop e5d4e0e cmake: fix android build and update grpc dependencies
  master  4a106be [ahead 2, behind 678] Remove == true
joe-pl@joe-PL:~/src/DronecodeSDK$ 
```
Second one:
```
joe-pl@joe-PL:**~/src/DronecodeSDK**$ git branch -v
* develop e5d4e0e cmake: fix android build and update grpc dependencies
  master  4a106be [ahead 2, behind 678] Remove == true
joe-pl@joe-PL:~/src/DronecodeSDK$ cd ~
joe-pl@joe-PL:~$ cd DronecodeSDK/
joe-pl@joe-PL:~/DronecodeSDK$ git log --oneline --decorate -5
a8ba9c1 (HEAD -> master, tag: v0.14.2, origin/master) docker: copy repo instead of cloning it from git
b51d707 docker: multi-stage backend
053a35f Docker backend service
c5ca314 cmake: fix bug in STRING REPLACE
3ca0ec9 camera: updated e90 camera_definition
joe-pl@joe-PL:**~/DronecodeSDK**$ git branch -vv
  develop e5d4e0e [origin/develop] cmake: fix android build and update grpc dependencies
* master  a8ba9c1 [origin/master] docker: copy repo instead of cloning it from git
joe-pl@joe-PL:~/DronecodeSDK$ 
```

JonasVautherin, 2019-02-27:
A git repo can go wherever you want. It's just like copying source files, except that there are versioned and you can get the history of changes etc.

Having two different clones is the same as, say, writing your phd thesis in two copies of the same Word document. Say you write 100 pages in `myThesis.doc`, then you copy it into `myThesis-final.doc`, and correct things here and there in `myThesis-final.doc`. Then you forget that you were working on `myThesis-final.doc`, and work on `myThesis.doc` for a while. And so on and so forth. What will happen? You will end up with two _similar_ documents, but each of them will have new stuff that you added and each of them will have old things that you corrected, but not the same ones!

That's basically what you had with Python: you showed me logs coming from the first clone, I told you how to fix them, and then you tried to run the second clone, and of course my fixes were not there (because you changed them in the first clone).

I know it is not ideal to have to build everything from source, especially if you don't know git, and we will improve that ASAP. Also building the python SDK was a bit more difficult for you because of Ubuntu 16.04 (from Ubuntu 16.10 on, you don't need to install python3.6).

Anyway, please keep only one copy of the `DronecodeSDK` repo, make sure it is up-to-date with something like:

```
git checkout v0.14.2
git submodule update
```

And then build it with:

```
make BUILD_BACKEND=YES
```

joe-marrone, 2019-02-28:
Jonas, what if I go to ubuntu 18.04? Would that help?

Joe

JonasVautherin, 2019-02-28:
> Jonas, what if I go to ubuntu 18.04? Would that help?

On the python side, it would mean that you would not have to install python3.6 and then replace all the occurences in the `run_protoc.sh` script.

On the c++ backend side, it would not change much. For that I am currently working on embedding the backend inside the Python SDK, but that will probably take some time.

joe-marrone, 2019-02-28:
I started a new build today, with the intent that `~/DronecodeSDK` is the repository.  Everything in the process went smoothly until building the backend.  Here is the log from that sequence:
```
joe-pl@joe-PL:~/DronecodeSDK$ make BUILD_BACKEND=1
Build dir: build/default
ninja: no work to do.
joe-pl@joe-PL:~/DronecodeSDK$ ls
appveyor.yml       cmake               debug_helpers  external_example                       install            Makefile       start_px4_sitl.sh
backend            CMakeLists.txt      docker         fix_style.sh                           integration_tests  plugins        stop_px4_sitl.sh
build              core                DronecodeSDK   generate_docs.sh                       iOS.cmake          README.md      third_party
ci-git-release.sh  create_packages.sh  example        generate_markdown_from_doxygen_xml.py  LICENSE.md         run-docker.sh  travis-docker-build.sh

joe-pl@joe-PL:~/DronecodeSDK$ ./build/default/backend/src/backend_bin
bash: ./build/default/backend/src/backend_bin: No such file or directory
joe-pl@joe-PL:~/DronecodeSDK$ git checkout v0.14.2
Note: checking out 'v0.14.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at a8ba9c1... docker: copy repo instead of cloning it from git
joe-pl@joe-PL:~/DronecodeSDK$ git submodule update
joe-pl@joe-PL:~/DronecodeSDK$ make BUILD_BACKEND=YES
Build dir: build/default
[1/1] Re-running CMake...
-- Version: 0.14.2
-- Include MAVLink passthrough plugin
-- BUILD_BACKEND not set: not building grpc backend
-- Configuring done
-- Generating done
-- Build files have been written to: /home/joe-pl/DronecodeSDK/build/default
[128/128] Linking CXX executable integration_tests/integration_tests_runner
joe-pl@joe-PL:~/DronecodeSDK$ ./build/default/backend/src/backend_bin
bash: ./build/default/backend/src/backend_bin: No such file or directory
joe-pl@joe-PL:~/DronecodeSDK$ 
```
There is no `build/default/backend/` directory.  This never happened before, so not sure why.

JonasVautherin, 2019-02-28:
> -- BUILD_BACKEND not set: not building grpc backend

This means that `make` had been called before, so `BUILD_BACKEND` is set to "NO". Can you try to `make clean` before running `make BUILD_BACKEND=YES`?

joe-marrone, 2019-03-01:
I did `make clean` and `make BUILD_BACKEND=YES`. But it did not complete.  At step 212 it started to throw errors.
```
[212/213] Building CXX object backend/test/CMakeFi...nit_tests_backend.dir/info_service_impl_test.cpp.o
FAILED: /usr/bin/c++   -DCARES_STATICLIB -DDRONECODE_SDK_VERSION=\"0.14.2-11-ge5d4e0e8\" -DLINUX -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -I../../core -I../../third_party/mavlink/include -I../../backend/test/../src -I../../backend/test/../src/plugins -I../../plugins -Ibackend/src -I../../ -I../../core/include -I../../core/include/dronecode_sdk -I../../plugins/action/include -I../../plugins/calibration/include -I../../plugins/camera/include -I../../plugins/mission/include -I../../plugins/mission/third_party/json11 -I../../plugins/telemetry/include -I../../plugins/info/include -isystem third_party/grpc/include -isystem third_party/boringssl/install -isystem third_party/protobuf/include -isystem third_party/cares/include -isystem ../../third_party/gtest/googletest/include -isystem ../../third_party/gtest/googletest -isystem ../../third_party/gtest/googlemock/include -isystem ../../third_party/gtest/googlemock -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wlogical-op -Wno-old-style-cast -pthread -std=gnu++11 -MMD -MT backend/test/CMakeFiles/unit_tests_backend.dir/info_service_impl_test.cpp.o -MF backend/test/CMakeFiles/unit_tests_backend.dir/info_service_impl_test.cpp.o.d -o backend/test/CMakeFiles/unit_tests_backend.dir/info_service_impl_test.cpp.o -c ../../backend/test/info_service_impl_test.cpp
In file included from ../../backend/test/../src/plugins/info/info_service_impl.h:2:0,
                 from ../../backend/test/info_service_impl_test.cpp:6:
backend/src/info/info.grpc.pb.h:69:11: error: ‘class dronecode_sdk::rpc::info::InfoService::Stub::experimental_async’ has pointer data members [-Werror=effc++]
     class experimental_async final :
           ^
backend/src/info/info.grpc.pb.h:69:11: error:   but does not override ‘dronecode_sdk::rpc::info::InfoService::Stub::experimental_async(const dronecode_sdk::rpc::info::InfoService::Stub::experimental_async&)’ [-Werror=effc++]
backend/src/info/info.grpc.pb.h:69:11: error:   or ‘operator=(const dronecode_sdk::rpc::info::InfoService::Stub::experimental_async&)’ [-Werror=effc++]
cc1plus: all warnings being treated as errors
[212/213] Building CXX object backend/test/CMakeFi...nit_tests_backend.dir/core_service_impl_test.cpp.o
FAILED: /usr/bin/c++   -DCARES_STATICLIB -DDRONECODE_SDK_VERSION=\"0.14.2-11-ge5d4e0e8\" -DLINUX -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -I../../core -I../../third_party/mavlink/include -I../../backend/test/../src -I../../backend/test/../src/plugins -I../../plugins -Ibackend/src -I../../ -I../../core/include -I../../core/include/dronecode_sdk -I../../plugins/action/include -I../../plugins/calibration/include -I../../plugins/camera/include -I../../plugins/mission/include -I../../plugins/mission/third_party/json11 -I../../plugins/telemetry/include -I../../plugins/info/include -isystem third_party/grpc/include -isystem third_party/boringssl/install -isystem third_party/protobuf/include -isystem third_party/cares/include -isystem ../../third_party/gtest/googletest/include -isystem ../../third_party/gtest/googletest -isystem ../../third_party/gtest/googlemock/include -isystem ../../third_party/gtest/googlemock -g -DDEBUG   -fno-exceptions -Wall -Wextra -Werror -Wshadow -Wno-strict-aliasing -Wold-style-cast -Wdouble-promotion -Wformat=2 -Weffc++ -Wlogical-op -Wno-old-style-cast -pthread -std=gnu++11 -MMD -MT backend/test/CMakeFiles/unit_tests_backend.dir/core_service_impl_test.cpp.o -MF backend/test/CMakeFiles/unit_tests_backend.dir/core_service_impl_test.cpp.o.d -o backend/test/CMakeFiles/unit_tests_backend.dir/core_service_impl_test.cpp.o -c ../../backend/test/core_service_impl_test.cpp
In file included from ../../backend/test/../src/core/core_service_impl.h:4:0,
                 from ../../backend/test/core_service_impl_test.cpp:9:
backend/src/core/core.grpc.pb.h:91:11: error: ‘class dronecode_sdk::rpc::core::CoreService::Stub::experimental_async’ has pointer data members [-Werror=effc++]
     class experimental_async final :
           ^
backend/src/core/core.grpc.pb.h:91:11: error:   but does not override ‘dronecode_sdk::rpc::core::CoreService::Stub::experimental_async(const dronecode_sdk::rpc::core::CoreService::Stub::experimental_async&)’ [-Werror=effc++]
backend/src/core/core.grpc.pb.h:91:11: error:   or ‘operator=(const dronecode_sdk::rpc::core::CoreService::Stub::experimental_async&)’ [-Werror=effc++]
cc1plus: all warnings being treated as errors
[212/213] Building CXX object backend/test/CMakeFi...t_tests_backend.dir/action_service_impl_test.cpp.o
FAILED: /usr/bin/c++   -DCARES_STATICLIB -DDRONECODE_SDK_VERSION=\"0.14.2-11-ge5d4e0e8\" -DLINUX -D_CRT_SECURE_NO_WARNINGS -isystem ../../third_party/tinyxml2 -I../../third_party/zlib -I../../core -I../../third_party/mavlink/include -I../../backend/test/../src -I../../backend/test/../src/plugins -I../../plugins -Ibackend/src -I../../ -I../../core/include -I../../core/include/dronecode_sdk -I../../plugins/action/include -I../../plugins/calibration/include -I../../plugins/camera/include -I../../plugins/mission/include -I../../plugins/mission/third_party/json11 -I../../plugins/telemetr [... cut ...]
joe-pl@joe-PL:~/DronecodeSDK/DronecodeSDK-Python$ cd examples/
joe-pl@joe-PL:~/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 takeoff_and_land.py
Traceback (most recent call last):
  File "takeoff_and_land.py", line 7, in <module>
    drone = dronecode_sdk_connect(host="127.0.0.1")
  File "/home/joe-pl/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 51, in connect
    globals()[plugin](plugin_manager)
KeyError: 'Action'
joe-pl@joe-PL:~/DronecodeSDK/DronecodeSDK-Python/examples$ 
```
What now??

joe-marrone, 2019-03-01:
UPDATE:  Sorry, I jumped the gun and did not complete the Python install, hence the errors.  I realized that 
`pip3 install -r requirements -r requirements-dev.txt` and `./other/tools/run_protoc.sh` had not been installed, and when they were, viola, the sim ran and `mission.py` ran!

But I'm still wondering about the errors in the BUILD_BACKEND=YES install.

JonasVautherin, 2019-03-02:
So now it works, even though there was a compilation error while building the backend. Is that correct?

joe-marrone, 2019-03-04:
Yes.  The mission.py script did run successfully despite the error messages from the last backend build.

Joe