mavlink/MAVSDK-Python #54: Issue Modifying Python Script

joe-marrone, 2019-03-01:
I am starting to use the DronecodeSDK-Python system.  It runs the canned `mission.py` script so I know it works.  But when I change the GPS parameters for my location, the script throws an error message. I did not make any other changes to the script, just the location parameters.  Note that the script freezes up and needs ^C to exit.  I also notice that the error message references another `mission.py` script in the 
`/home/joe-pl/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/plugins/mission.py` directory.  
This is a completely different script, so why the same name? Will this other `mission.py` work with other Python scripts I write?

It would be useful if there was a tutorial to assist in getting started using the Python wrapper to start experimenting.
Thanks for any help.
Joe 
```
joe-pl@joe-PL:~/DronecodeSDK/DronecodeSDK-Python/my_examples$ python3.6 mission-nh.py
-- Uploading mission
-- Arming
-- Starting mission
Task exception was never retrieved
future: <Task finished coro=<run() done, defined at mission-nh.py:11> exception=MissionError(<dronecode_sdk.plugins.mission.MissionResult object at 0x7fc3f6a6df98>, 'start_mission()')>
Traceback (most recent call last):
  File "mission-nh.py", line 53, in run
    await drone.mission.start_mission()
  File "/home/joe-pl/DronecodeSDK/DronecodeSDK-Python/dronecode_sdk/plugins/mission.py", line 478, in start_mission
    raise MissionError(result, "start_mission()")
dronecode_sdk.plugins.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()

^CTraceback (most recent call last):
  File "mission-nh.py", line 83, in <module>
    asyncio.get_event_loop().run_until_complete(observe_is_in_air())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 471, in run_until_complete
    self.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)

KeyboardInterrupt
^CError in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 40, in _python_exit
    t.join()
  File "/usr/lib/python3.6/threading.py", line 1056, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.6/threading.py", line 1072, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```

JonasVautherin, 2019-03-04:
>>> dronecode_sdk.plugins.mission.MissionError: ERROR: 'Error'; origin: start_mission(); params: ()

`start_mission()` raised a `MissionError` with code `ERROR` (which you could obviously catch with the usual `try... except...` python construct).

My guess is that your simulator refused to take off because the waypoints you uploaded are too far away from the drone.

Did you try to download the mission from QGC and run it from there?

joe-marrone, 2019-03-04:
After some experminting, I changed the location parameters in MAVLink to my home location. Then I ran the modified `mission.py` script. The drone took off and rose to the specified altitude.  It paused there and then landed.  Just prior to the mission starting it got a RTL command. The mission readouts are below.  
1) jMAVSim output:
```pxh> INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-03-04/19_10_19.ulg
INFO  [logger] Start file log (type: mission)
INFO  [logger] Opened mission log file: ./mission_log/2019-03-04/19_10_19.ulg
INFO  [commander] Takeoff detected
INFO  [commander] Returning to launch
INFO  [navigator] RTL HOME activated
INFO  [navigator] RTL: climb to 100 m (10 m above home)
INFO  [navigator] RTL: return at 100 m (10 m above home)
INFO  [navigator] RTL: descend to 100 m (10 m above home)
INFO  [navigator] RTL: land at home
INFO  [commander] Landing detected
INFO  [logger] closed logfile, bytes written: 6715
INFO  [logger] closed logfile, bytes written: 2483958
```
2) Backend output:
```
[02:10:19|Debug] Send mission item 0 (mission_impl.cpp:1109)
[02:10:19|Debug] Send mission item 1 (mission_impl.cpp:1109)
[02:10:19|Debug] Send mission item 2 (mission_impl.cpp:1109)
[02:10:19|Debug] Send mission item 3 (mission_impl.cpp:1109)
[02:10:19|Debug] Send mission item 4 (mission_impl.cpp:1109)
[02:10:19|Debug] Send mission item 5 (mission_impl.cpp:1109)
[02:10:19|Debug] Send mission item 6 (mission_impl.cpp:1109)
[02:10:19|Info ] Mission accepted (mission_impl.cpp:164)
[02:10:19|Debug] MAVLink: info: ARMED by arm/disarm component command (system_impl.cpp:292)
[02:10:19|Debug] MAVLink: info: [logger] file: ./log/2019-03-04/19_10_19.ulg (system_impl.cpp:292)
[02:10:19|Debug] MAVLink: info: Takeoff to 25.0 meters above home. (system_impl.cpp:292)
[02:10:19|Debug] MAVLink: info: Takeoff detected (system_impl.cpp:292)
[02:10:36|Debug] current: 0, total: 3 (mission_impl.cpp:1188)
[02:10:36|Debug] current: 1, total: 3 (mission_impl.cpp:1188)
[02:10:37|Debug] current: 2, total: 3 (mission_impl.cpp:1188)
[02:10:37|Debug] MAVLink: info: Returning to launch (system_impl.cpp:292)
[02:10:37|Debug] current: 3, total: 3 (mission_impl.cpp:1188)
[02:10:37|Debug] MAVLink: info: RTL HOME activated (system_impl.cpp:292)
[02:10:37|Debug] MAVLink: info: RTL: climb to 100 m (10 m above home) (system_impl.cpp:292)
[02:10:53|Debug] MAVLink: info: RTL: return at 100 m (10 m above home) (system_impl.cpp:292)
[02:10:53|Debug] MAVLink: info: RTL: descend to 100 m (10 m above home) (system_impl.cpp:292)
[02:10:53|Debug] MAVLink: info: RTL: land at home (system_impl.cpp:292)
[02:11:10|Debug] MAVLink: info: Landing detected (system_impl.cpp:292)
[02:11:11|Debug] MAVLink: info: DISARMED by auto disarm on land (system_impl.cpp:292)
```
3) Python script output:
```
joe-pl@joe-PL:~/DronecodeSDK/DronecodeSDK-Python/examples$ python3.6 mission.py
-- Uploading mission
-- Arming
-- Starting mission
Mission progress: 0/3
Mission progress: 1/3
Mission proPX4_HOME_LATgress: 2/3
Mission progress: 3/3
```

The script I used is the canned `mission.py` that is supplied in the `DronecodeSDK/DronecodeSDK-Python/examples/` file.  All I did was change the three location parameters to my location, along with changing
``` 
PX4_HOME_LAT
PX4_HOME_LON
PX4_HOME_ALT
```
in MAVLink.

Any clues as to why the mission did not run?

Joe

JonasVautherin, 2019-03-04:
So you say that the mission works fine if you don't change anything from the example, right?

Can you try to _not_ start the mission (comment out the `start_mission()` line), upload the mission, and then download and run it from QGC?

joe-marrone, 2019-03-04:
I commented out `start_mission()`.  How do I upload this `.py` file to QGC?  It wants a `.plan` file.

joe-marrone, 2019-03-04:
I played around with waypoint locations and did get the `mission.py` to run using the modified parameters. Th only thing I can think of is that I had the waypoints too close together and by putting them further apart I could see the mission running. So now I have a working system.

JonasVautherin, 2019-03-04:
Right, so it's not an SDK issue.

> I commented out start_mission(). How do I upload this .py file to QGC? It wants a .plan file.

That's not what I meant. You can connect both the SDK and QGC to the drone (in this case SITL) at the same time. If you comment out `start_mission()` and run `mission.py`, it will upload a mission to the drone. From QGC, you can download this mission and see how it looks on the map. You can also try to run it.

Closing now, as you figured out why your mission was not valid :+1:.

closed 2019-03-04 by JonasVautherin