mavlink/MAVSDK-Python #55: Program not found or is not executable --custom_out: protoc-gen-custom

JonasVautherin, 2019-03-08:
My apologies for using a close issue. However, it's the only post I found about issues running run_protoc.sh

In my case, it cannot generate the plugins.  The server builds and works fine, but I'm not sure why run_protoc.sh fails.

`[+] Installing the DronecodeSDK autogenerator
Processing /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/proto/pb_plugins
Requirement already satisfied: protobuf in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (3.7.0)
Requirement already satisfied: jinja2 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (2.10)
Requirement already satisfied: six>=1.9 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (1.12.0)
Requirement already satisfied: setuptools in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: MarkupSafe>=0.23 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jinja2->dcsdkgen==0.1a0) (1.1.0)
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
  Running setup.py install for dcsdkgen ... done
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for action
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Action.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/action.py: No such file or directory
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for calibration
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Calibration.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/calibration.py: No such file or directory
 -> [+] Generated plugin for calibration
 -> [+] Generated protobuf and gRPC bindings for camera
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Camera.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/camera.py: No such file or directory
 -> [+] Generated plugin for camera
 -> [+] Generated protobuf and gRPC bindings for core
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Core.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/core.py: No such file or directory
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for discovery
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Discovery.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/discovery.py: No such file or directory
 -> [+] Generated plugin for discovery
 -> [+] Generated protobuf and gRPC bindings for gimbal
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Gimbal.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/gimbal.py: No such file or directory
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for info
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Info.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/info.py: No such file or directory
 -> [+] Generated plugin for info
 -> [+] Generated protobuf and gRPC bindings for mission
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Mission.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/mission.py: No such file or directory
 -> [+] Generated plugin for mission
 -> [+] Generated protobuf and gRPC bindings for telemetry
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: rename /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Telemetry.py to /Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/telemetry.py: No such file or directory
 -> [+] Generated plugin for telemetry
[+] Done
`

_Originally posted by @IeiuniumLux in https://github.com/Dronecode/DronecodeSDK-Python/issues/48#issuecomment-470845052_

JonasVautherin, 2019-03-08:
Did you [install the prerequisites](https://github.com/dronecode/dronecodesdk-python#install-prerequisites) before running `./other/tools/run_protoc.sh`?

I just ran the following successfully on my computer:

```
cd /tmp
git clone https://github.com/dronecode/dronecodesdk-python --recursive
cd dronecodesdk-python
pip3 install -r requirements.txt -r requirements-dev.txt
./other/tools/run_protoc.sh
```

IeiuniumLux, 2019-03-08:
@JonasVautherin, thank you for creating this a new issue.   Yes, I ran successfully all those steps, except for the last one (./other/tools/run_protoc.sh). I'm using Python 3.6.8 and pip 18.1. In the instructions, it's not clear if DronecodeSDK-Python needs to be cloned underneath the DronecodeSDK repo or parallel to it. I did both, and run_protoc.sh still fails.

> git clone https://github.com/dronecode/dronecodesdk-python --recursive
> cd dronecodesdk-python
> pip3 install -r requirements.txt -r requirements-dev.txt

[image]

JonasVautherin, 2019-03-08:
> In the instructions, it's not clear if DronecodeSDK-Python needs to be cloned underneath the DronecodeSDK repo or parallel to it.

It is a separate repo, and should be treated as such. Otherwise I guess it would either be part of DronecodeSDK, or at least a submodule :-). Just clone them separately, that's better :+1:.

From your screenshot, I see that jmavsim is running, the backend succeeded in discovering it, and the python module is not there (obviously, because `run_protoc.sh` failed).

> I did both, and run_protoc.sh still fails.

Do you always get exactly the same message as you pasted above? Did `pip install` run successfully?

IeiuniumLux, 2019-03-08:
> Do you always get exactly the same message as you pasted above? Did `pip install` run successfully?

Yes, I do always get exactly the same message and 'pip install'  ran successfully without reporting any issues. I tried to run 'python -m grpc_tools.protoc' from the prompt, but it fails since files were not generated by run_protoc.sh

[image]

JonasVautherin, 2019-03-08:
It's a good idea to checkout the latest release `v0.14.2` instead of `master` :+1:.

> I tried to run 'python -m grpc_tools.protoc' from the prompt, but it fails since files were not generated by run_protoc.sh

So this runs `protoc`, which requires some arguments (including input files). So this error message is normal (you could run e.g. `python3 -m grpc_tools.protoc -h`).

The error message says:

> : program not found or is not executable

I am wondering if $(which dcsdkgen) just does not find our plugin (`dcsdkgen`, in `proto/pb_plugins`), and results in `""`, which then gets executed by `protoc`. So the error message says that `""` is not executable, which makes sense. And `dcsdkgen` is supposed to be installed by `run_protoc.sh` ([here](https://github.com/Dronecode/DronecodeSDK-Python/blob/master/other/tools/run_protoc.sh#L58)).

What do you think?

IeiuniumLux, 2019-03-08:
dcsdkgen was indeed installed:
[image]

JonasVautherin, 2019-03-09:
> dcsdkgen was indeed installed:

This doesn't mean it was installed. It means that the submodule is there. I guess `which dcsdkgen` says something like `dcsdkgen not found`, doesn't it?

Are you installing stuff on the system or in a venv?

IeiuniumLux, 2019-03-09:
> you could try the following:
> ```
> cd proto/pb_plugins
> pip install -r requirements.txt
> pip install -e .
> which dcsdkgen
> ```

Outputs below:

[abencomo: pb_plugins]$ pip install -r requirements.txt
```
Requirement already satisfied: protobuf in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (3.7.0)
Requirement already satisfied: jinja2 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from -r requirements.txt (line 2)) (2.10)
Requirement already satisfied: setuptools in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->-r requirements.txt (line 1)) (40.8.0)
Requirement already satisfied: six>=1.9 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->-r requirements.txt (line 1)) (1.12.0)
Requirement already satisfied: MarkupSafe>=0.23 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jinja2->-r requirements.txt (line 2)) (1.1.0)
```

pip install -e .
```
Obtaining file:///Users/abencomo/git/DronecodeSDK/DronecodeSDK-Python/proto/pb_plugins
Requirement already satisfied: protobuf in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (3.7.0)
Requirement already satisfied: jinja2 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (2.10)
Requirement already satisfied: six>=1.9 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (1.12.0)
Requirement already satisfied: setuptools in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: MarkupSafe>=0.23 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jinja2->dcsdkgen==0.1a0) (1.1.0)
Installing collected packages: dcsdkgen
  Found existing installation: dcsdkgen 0.1a0
    Uninstalling dcsdkgen-0.1a0:
      Successfully uninstalled dcsdkgen-0.1a0
  Running setup.py develop for dcsdkgen
Successfully installed dcsdkgen
```
[abencomo: pb_plugins]$ which dcsdkgen
```

```
nothing when enter which dcsdkgen

JonasVautherin, 2019-03-09:
Hmm. And can you find it on your system somewhere? You are on the master branch of DronecodeSDK-Python, right?

Maybe you could also try to install it in a venv instead of the system?

julianoes, 2019-03-12:
By default I don't have `dcsdkgen` in my path and I always need to do:
```
export PATH=$PATH:$HOME/.local/bin
```

```
which dcsdkgen
~/.local/bin/dcsdkgen
```

IeiuniumLux, 2019-03-12:
> By default I don't have `dcsdkgen` in my path and I always need to do:
> ```
> export PATH=$PATH:$HOME/.local/bin
> ```

That did it.  Thanks @julianoes 
@JonasVautherin, feel free to close this issue and thank you.
```
Requirement already satisfied: protobuf in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (3.7.0)
Requirement already satisfied: jinja2 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from dcsdkgen==0.1a0) (2.10)
Requirement already satisfied: setuptools in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (40.8.0)
Requirement already satisfied: six>=1.9 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from protobuf->dcsdkgen==0.1a0) (1.12.0)
Requirement already satisfied: MarkupSafe>=0.23 in /opt/local/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from jinja2->dcsdkgen==0.1a0) (1.1.0)
Installing collected packages: dcsdkgen
  Running setup.py install for dcsdkgen ... done
Successfully installed dcsdkgen-0.1a0
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for action
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for calibration
 -> [+] Generated plugin for calibration
 -> [+] Generated protobuf and gRPC bindings for camera
 -> [+] Generated plugin for camera
 -> [+] Generated protobuf and gRPC bindings for core
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for discovery
 -> [+] Generated plugin for discovery
 -> [+] Generated protobuf and gRPC bindings for gimbal
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for info
 -> [+] Generated plugin for info
 -> [+] Generated protobuf and gRPC bindings for mission
 -> [+] Generated plugin for mission
 -> [+] Generated protobuf and gRPC bindings for telemetry
 -> [+] Generated plugin for telemetry
[+] Done
```

julianoes, 2019-03-13:
Thanks for the feedback @IeiuniumLux.

@JonasVautherin I think we should add that to the docs.

JonasVautherin, 2019-03-13:
@julianoes: Thanks a lot!

So installing a python package system-wide doesn't put it in the PATH? What's the point of installing it on the system then?

closed 2019-03-13 by JonasVautherin

julianoes, 2019-03-14:
I don't install it system wide, I install it as a user, and that path is not added by default.