mavlink/MAVSDK-Python #57: SyntaxError: protoc-gen-cu

efischer91, 2019-04-02:
Hello,

I've been trying to install the python wrapper as stated here: [https://github.com/Dronecode/DronecodeSDK-Python/blob/master/README.md](DronecodeSDK-Python README.md)

Everything works fine until using the helper script to generate the wrapper. As soon as I start it, I get the following error log:

```
eve@eve-ThinkPad-X1-Carbon-3rd:~/Downloads/DronecodeSDK-Python$ ./other/tools/run_protoc.sh 
[+] Installing the DronecodeSDK autogenerator
Processing /home/eve/Downloads/DronecodeSDK-Python/proto/pb_plugins
Collecting protobuf (from dcsdkgen===0.1a0)
  Using cached https://files.pythonhosted.org/packages/81/59/c7b0815a78fd641141f24a6ece878293eae6bf1fce40632a6ab9672346aa/protobuf-3.7.1-cp35-cp35m-manylinux1_x86_64.whl
Collecting jinja2 (from dcsdkgen===0.1a0)
  Using cached https://files.pythonhosted.org/packages/7f/ff/ae64bacdfc95f27a016a7bed8e8686763ba4d277a78ca76f32659220a731/Jinja2-2.10-py2.py3-none-any.whl
Collecting six>=1.9 (from protobuf->dcsdkgen===0.1a0)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting setuptools (from protobuf->dcsdkgen===0.1a0)
  Using cached https://files.pythonhosted.org/packages/d1/6a/4b2fcefd2ea0868810e92d519dacac1ddc64a2e53ba9e3422c3b62b378a6/setuptools-40.8.0-py2.py3-none-any.whl
Collecting MarkupSafe>=0.23 (from jinja2->dcsdkgen===0.1a0)
  Using cached https://files.pythonhosted.org/packages/6e/57/d40124076756c19ff2269678de7ae25a14ebbb3f6314eb5ce9477f191350/MarkupSafe-1.1.1-cp35-cp35m-manylinux1_x86_64.whl
Installing collected packages: six, setuptools, protobuf, MarkupSafe, jinja2, dcsdkgen
  Running setup.py install for dcsdkgen ... done
Successfully installed MarkupSafe-1.1.1 dcsdkgen-0.1a0 jinja2-2.10 protobuf-3.7.1 setuptools-40.8.0 six-1.12.0
You are using pip version 8.1.1, however version 19.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for gimbal
Traceback (most recent call last):
  File "/home/eve/.local/bin/dcsdkgen", line 11, in <module>
    load_entry_point('dcsdkgen==0.1a0', 'console_scripts', 'dcsdkgen')()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/eve/.local/lib/python3.5/site-packages/dcsdkgen/__init__.py", line 5, in <module>
    from .autogen import AutoGen
  File "/home/eve/.local/lib/python3.5/site-packages/dcsdkgen/autogen.py", line 64
    f.name = f"{plugin_name}.{params['file_ext']}"
                                                 ^
SyntaxError: invalid syntax
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: Aufruf von stat für '/home/eve/Downloads/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Gimbal.py' nicht möglich: Datei oder Verzeichnis nicht gefunden
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for core
Traceback (most recent call last):
  File "/home/eve/.local/bin/dcsdkgen", line 11, in <module>
    load_entry_point('dcsdkgen==0.1a0', 'console_scripts', 'dcsdkgen')()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/eve/.local/lib/python3.5/site-packages/dcsdkgen/__init__.py", line 5, in <module>
    from .autogen import AutoGen
  File "/home/eve/.local/lib/python3.5/site-packages/dcsdkgen/autogen.py", line 64
    f.name = f"{plugin_name}.{params['file_ext']}"
                                                 ^
SyntaxError: invalid syntax
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: Aufruf von stat für '/home/eve/Downloads/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Core.py' nicht möglich: Datei oder Verzeichnis nicht gefunden
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for info
[... 162 lines cut ...]
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for telemetry
Traceback (most recent call last):
  File "/home/eve/.local/bin/dcsdkgen", line 11, in <module>
    load_entry_point('dcsdkgen==0.1a0', 'console_scripts', 'dcsdkgen')()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2411, in load
    return self.resolve()
  File "/home/eve/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/eve/.local/lib/python3.5/site-packages/dcsdkgen/__init__.py", line 5, in <module>
    from .autogen import AutoGen
  File "/home/eve/.local/lib/python3.5/site-packages/dcsdkgen/autogen.py", line 64
    f.name = f"{plugin_name}.{params['file_ext']}"
                                                 ^
SyntaxError: invalid syntax
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: Aufruf von stat für '/home/eve/Downloads/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Telemetry.py' nicht möglich: Datei oder Verzeichnis nicht gefunden
 -> [+] Generated plugin for telemetry
[+] Done
```
I guess it's an error to protoc-gen-custom, but I'm quite new to ubuntu and not sure how to fix this. Can anyone maybe assist?

JonasVautherin, 2019-04-02:
> /home/eve/.local/lib/python3.5

You are on Ubuntu 16.04, I presume? You will need Python3.6+, and it seems to me that you have Python3.5.

efischer91, 2019-04-02:
> You are on Ubuntu 16.04, I presume? You will need Python3.6+, and it seems to me that you have Python3.5.

I'm on Ubuntu 16.04 and have both Python 3.5.2 since it's the pre-installed version and 3.6.7

JonasVautherin, 2019-04-02:
`./other/tools/run_protoc.sh` is running `python3`, which in your case defaults to 3.5. Can you try to edit all occurences of `python3` and `pip3` in `proto.sh`?

```
python3 -> python3.6
pip3 -> pip3.6
```

efischer91, 2019-04-02:
This helped with running the shell script, thanks! Now, the next problem is that none of the examples seem to work:

```
eve@eve-ThinkPad-X1-Carbon-3rd:~/Downloads/DronecodeSDK-Python$ python3.6 examples/takeoff_and_land.py 
-- Arming
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 25, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 473, in run_until_complete
    return future.result()
  File "examples/takeoff_and_land.py", line 13, in run
    await drone.action.arm()
  File "/home/eve/Downloads/DronecodeSDK-Python/dronecode_sdk/plugins/action.py", line 167, in arm
    raise ActionError(result, "arm()")
dronecode_sdk.plugins.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 30, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 23, in <module>
    from apport.packaging_impl import impl as packaging
  File "/usr/lib/python3/dist-packages/apport/packaging_impl.py", line 23, in <module>
    import apt
  File "/usr/lib/python3/dist-packages/apt/__init__.py", line 23, in <module>
    import apt_pkg
ModuleNotFoundError: No module named 'apt_pkg'

Original exception was:
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 25, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 473, in run_until_complete
    return future.result()
  File "examples/takeoff_and_land.py", line 13, in run
    await drone.action.arm()
  File "/home/eve/Downloads/DronecodeSDK-Python/dronecode_sdk/plugins/action.py", line 167, in arm
    raise ActionError(result, "arm()")
dronecode_sdk.plugins.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()

```

Gazebo simulation is running, backend is running as well. Examples unfortunately don't, even though I did install python3-apt.

JonasVautherin, 2019-04-02:
> dronecode_sdk.plugins.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()

That is actually an error raised by the SDK: `COMMAND_DENIED` when you called `arm()`. You could `try... catch` around `arm()` and do something when this error is raised.

So it seems like it is working: the drone (here Gazebo) is just refusing to arm for some reason (not ready?) :-). What does the `telemetry.py` example say?

julianoes, 2019-04-03:
@JonasVautherin what's this about?
```
ModuleNotFoundError: No module named 'apt_pkg'
```

JonasVautherin, 2019-04-03:
No clue :sweat_smile:. 

> Examples unfortunately don't, even though I did install python3-apt.

@efischer91: Are you sure you installed it with `pip3.6`, by the way?

efischer91, 2019-04-03:
> That is actually an error raised by the SDK: `COMMAND_DENIED` when you called `arm()`. You could `try... catch` around `arm()` and do something when this error is raised.
>
> So it seems like it is [...]

I actually don't know why, but it works fine today :)

> @efischer91: Are you sure you installed it with `pip3.6`, by the way?

Definitely. Everything's working fine now. Thank you!

closed 2019-04-04 by JonasVautherin