mavlink/MAVSDK-Python #62: encoding into SDK and sending mavlink messages to the drone.

lioneleeedadad, 2019-04-26:
I have already imported a teleoperation package from github which uses keyboard and written in python also I think that I should use the offboard mode for my task but I don't know how I can encode this teleoperation package into SDK and how to send this mavlink message to the drone.

JonasVautherin, 2019-04-28:
That's not clear to me. What is this package, and what do you mean by "encode this package into SDK"?

julianoes, 2019-04-29:
You need to provide more details for us to help. Provide the code that you already have. Link to this "teleoperation package". Explain what you're trying to achieve.

Also, it might be a good time to learn about what a C/C++ library is, what header files are, what a binary is and how it all fits together.

JonasVautherin, 2019-06-25:
Closing due to inactivity.

closed 2019-06-25 by JonasVautherin