mavlink/MAVSDK-Python #67: Failed to build and run.

SalimTerryLi, 2019-05-24:
`******@******:/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/examples$ python3 takeoff_and_land.py 
Traceback (most recent call last):
  File "takeoff_and_land.py", line 5, in <module>
    from dronecode_sdk import connect as dronecode_sdk_connect
  File "/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/dronecode_sdk/__init__.py", line 48, in <module>
    from .plugins import * # NOQA
  File "/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/dronecode_sdk/plugins/__init__.py", line 3, in <module>
    from .core import *
ModuleNotFoundError: No module named 'dronecode_sdk.plugins.core'`
It seems not be installed correctly. Actually, when I execute `./other/tools/run_protoc.sh` it complains about wrong path.
`SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
WORK_DIR="${SCRIPT_DIR}/../../"
PROTO_DIR="${WORK_DIR}proto"
GENERATED_DIR="${WORK_DIR}dronecode_sdk/generated"
PLUGIN_DIR="${WORK_DIR}dronecode_sdk/plugins"
PLUGIN_INIT="${PLUGIN_DIR}/__init__.py"
export TEMPLATE_PATH="${WORK_DIR}/other/templates/"`
This modification works but there is nothing in the dir at all.

JonasVautherin, 2019-05-24:
What does it say when you run `other/tools/run_protoc.sh`?

Could it be related to the just-merged #68?

SalimTerryLi, 2019-05-24:
> What does it say when you run `other/tools/run_protoc.sh`?
> 
> Could it be related to the just-merged #68?

RAW output below. Some Chinese characters due to my region.

`salimterryli@STL-Laptop:/media/salimterryli/dev-linux/px4/DronecodeSDK-Python$ ./other/tools/run_protoc.sh
[+] Installing the DronecodeSDK autogenerator
Processing /media/salimterryli/dev-linux/px4/DronecodeSDK-Python/proto/pb_plugins
Collecting protobuf (from protoc-gen-dcsdk==0.1)
  Using cached https://files.pythonhosted.org/packages/5a/aa/a858df367b464f5e9452e1c538aa47754d467023850c00b000287750fa77/protobuf-3.7.1-cp36-cp36m-manylinux1_x86_64.whl
Collecting jinja2 (from protoc-gen-dcsdk==0.1)
  Using cached https://files.pythonhosted.org/packages/1d/e7/fd8b501e7a6dfe492a433deb7b9d833d39ca74916fa8bc63dd1a4947a671/Jinja2-2.10.1-py2.py3-none-any.whl
Collecting six>=1.9 (from protobuf->protoc-gen-dcsdk==0.1)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting setuptools (from protobuf->protoc-gen-dcsdk==0.1)
  Using cached https://files.pythonhosted.org/packages/ec/51/f45cea425fd5cb0b0380f5b0f048ebc1da5b417e48d304838c02d6288a1e/setuptools-41.0.1-py2.py3-none-any.whl
Collecting MarkupSafe>=0.23 (from jinja2->protoc-gen-dcsdk==0.1)
  Using cached https://files.pythonhosted.org/packages/b2/5f/23e0023be6bb885d00ffbefad2942bc51a620328ee910f64abe5a8d18dd1/MarkupSafe-1.1.1-cp36-cp36m-manylinux1_x86_64.whl
Installing collected packages: six, setuptools, protobuf, MarkupSafe, jinja2, protoc-gen-dcsdk
  Running setup.py install for protoc-gen-dcsdk ... done
Successfully installed MarkupSafe-1.1.1 jinja2-2.10.1 protobuf-3.7.1 protoc-gen-dcsdk-0.1 setuptools-41.0.1 six-1.12.0
[+] Done
[+] Generating plugins from 
 -> [+] Generated protobuf and gRPC bindings for core
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Core.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for core
 -> [+] Generated protobuf and gRPC bindings for telemetry
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Telemetry.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for telemetry
 -> [+] Generated protobuf and gRPC bindings for info
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Info.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for info
 -> [+] Generated protobuf and gRPC bindings for offboard
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Offboard.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for offboard
 -> [+] Generated protobuf and gRPC bindings for param
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Param.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for param
 -> [+] Generated protobuf and gRPC bindings for camera
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Camera.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for camera
 -> [+] Generated protobuf and gRPC bindings for gimbal
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Gimbal.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for gimbal
 -> [+] Generated protobuf and gRPC bindings for action
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Action.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for action
 -> [+] Generated protobuf and gRPC bindings for mission
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Mission.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for mission
 -> [+] Generated protobuf and gRPC bindings for calibration
: program not found or is not executable
--custom_out: protoc-gen-custom: Plugin failed with status code 1.
mv: 无法获取'/media/salimterryli/dev-linux/px4/DronecodeSDK-Python/other/tools/../..//dronecode_sdk/plugins/Calibration.py' 的文件状态(stat): 没有那个文件或目录
 -> [+] Generated plugin for calibration
[+] Done`

But after I fixed `/../..//dronecode_sdk` it still can' t work.

JonasVautherin, 2019-05-24:
> But after I fixed /../..//dronecode_sdk it still can' t work.

That seems like a wrong way to fix it. 

It seems to me like the error you see is because of #68, or because you haven't installed the protoc plugin (from `proto/pb_plugins`).

Can you try with the steps described in #68? Please update your git clone before doing it, as I merged #68.

SalimTerryLi, 2019-05-25:
> That seems like a wrong way to fix it.
>
> It seems to me like the error you see is because of #68, or because you haven't installed the protoc plugin (from `proto/pb_plugins`).
>
> Can you try with the [...]

It works, thank you.

closed 2019-05-25 by SalimTerryLi