mavlink/MAVSDK-Python #69: I'm now getting this error after running `example/takeoff_and_land.py`

JonasVautherin, 2019-05-26:
I'm now getting this error after running `example/takeoff_and_land.py`
```
Traceback (most recent call last):
  File "examples/takeoff_and_land.py", line 5, in <module>
    from dronecode_sdk import connect as dronecode_sdk_connect
  File "/Users/myuser/Desktop/dronecodesdk-python/dronecode_sdk/__init__.py", line 48, in <module>
    from .plugins import * # NOQA
  File "/Users/myuser/Desktop/dronecodesdk-python/dronecode_sdk/plugins/__init__.py", line 1, in <module>
    -e # -*- coding: utf-8 -*-
NameError: name 'e' is not defined
```

[Here](https://gist.github.com/unipheas/4ea0d141fd8ea0341806470b014bc68d) is the entire log for everything I'm doing, hope this helps. I want to add that this is on a fresh install of MacOS on a MacBook Air.

_Originally posted by @unipheas in https://github.com/Dronecode/DronecodeSDK-Python/pull/68#issuecomment-495987904_

JonasVautherin, 2019-05-26:
The installation seems all good to me. When running the example, it says:

```
File "/Users/myuser/Desktop/dronecodesdk-python/dronecode_sdk/plugins/__init__.py", line 1, in <module>
    -e # -*- coding: utf-8 -*-
```

However, my `dronecode_sdk/plugins/__init__.py` starts with:

```
# -*- coding: utf-8 -*-
```

i.e. there is no `-e` at the beginning. It is generated from [this line](https://github.com/Dronecode/DronecodeSDK-Python/blob/master/other/tools/run_protoc.sh#L11) in `run_protoc.sh`:

```
echo -e "# -*- coding: utf-8 -*-\n" > $PLUGIN_INIT
```

I don't get why `-e` gets into the file, though, as it is an argument to `echo`.

unipheas, 2019-05-26:
Here is some more information regarding the system I'm using for this
```
MacOS Mojave 10.14.5
Python 3.7.3
pip 19.1.1 from /usr/local/lib/python2.7/site-packages/pip (python 2.7)
go version go1.12.5 Darwin/amd64
java 12.0.1 2019-04-16
Java(TM) SE Runtime Environment (build 12.0.1+12)
Java HotSpot(TM) 64-Bit Server VM (build 12.0.1+12, mixed mode, sharing)
```

let me know if any other information is required.

JonasVautherin, 2019-05-26:
Oh, it seems like `run_protoc.sh` is not compatible with `sh` on macos. So you should run:

```
bash other/tools/run_protoc.sh
```

or just

```
./other/tools/run_protoc.sh
```

To fix this issue, I will update `run_protoc.sh` so that it is compatible with `sh` on macos :+1:. 

Let me know if that fixes the problem for you!

unipheas, 2019-05-26:
IT WORKED!!!!

```
examples/takeoff_and_land.py
-- Arming
-- Taking off
-- Landing
```

unipheas, 2019-05-26:
I'd say to make sure to add `git clone https://github.com/dronecode/dronecodesdk-python --recursive` so the user can find photo/pb_plugins.

PR #70 (2019-05-26): run_protoc.sh: use printf instead of echo 
That should bring more consistent behavior between sh and bash on different platforms.

Resolves #69.

@unipheas: FYI
closed 2019-05-26 by xvzf