mavlink/MAVSDK-Python #77: Error running python exanples

giovanidiniz, 2019-06-21:
Hey guys,
I have just tried building the back end and wrappers as it is described in the instructions here.
I don't get compilation errors, but when I run any of the python examples I get gRPC status code `UNIMPLEMENTED`.

I'm not knowledgeable in gRPC servers at all but as far I understand the problem is in the server? Or at least it seems the messages in the server and wrapper and not the same?

Any one has any insight into this? I'm trying to write a ROS2 node using the MAVSDK as comm library between OBC and Autopilot.

Appreciate any insights

JonasVautherin, 2019-06-23:
Hello!

Sorry about that, we recently renamed the project to `MAVSDK`, and therefore there are some incompatibilities between the main branches of both project.

To use MAVSDK-Python, just run `pip install mavsdk` (make sure you have python3.6+, it may mean `pip3 install mavsdk`).

Let us know how it goes!

giovanidiniz, 2019-06-24:
Hi Jonas,
Is the backend going to be installed pip? So far, I can install the sdk without errors. The issue is I keep getting these UNIMPLEMENTED return codes from the gRPC server:

> grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "{"created":"@1561335925.969796887","description":"Error received from peer ipv4:127.0.0.1:50051","file":"src/core/lib/surface/call.cc","file_line":1041,"grpc_message":"","grpc_status":12}"

I followed the guide in README to install the backend. I'm connecting it to px4 sitl and this is stdout:
>[08:25:10|Info ] MAVSDK version: 0.18.0-dirty (mavsdk_impl.cpp:25)
[08:25:10|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:333)
[08:25:10|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:44)
[08:25:10|Info ] Server started (grpc_server.cpp:28)
[08:25:10|Info ] Waiting to discover system... (connection_initiator.h:58)
[08:25:19|Info ] New system on: 127.0.0.1:43329 (udp_connection.cpp:228)
[08:25:19|Debug] Component Autopilot (1) added. (system_impl.cpp:400)
[08:25:19|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:563)
[08:25:19|Info ] System discovered [UUID: 5283920058631409231] (connection_initiator.h:62)

I think the server is connecting with the sim but there seems to be a disconnect between the python messages and the server's?

If you have any idea how to go about this, troubleshoot, it will be greatly appreciated.
Best

JonasVautherin, 2019-06-24:
> Is the backend going to be installed pip?

Yes, it will, and those two versions will match :-).

> I think the server is connecting with the sim but there seems to be a disconnect between the python messages and the server's?

Exactly, because of the renaming last week. I'll try to fix that during this week, but in the meantime you should try with `pip3 install mavdsk` ;-).

One thing to note, though (and I will update the examples as well): with the pip-installed MAVSDK, you need to start the backend manually, with `start_mavlink()`. So for instance, add the following just after [this line](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/takeoff_and_land.py#L5):

```python
from mavsdk import start_mavlink

start_mavlink()
```

After running that command, you can see on the list of processes (e.g. running `top` or `htop`) that `mavsdk_server` is running. And it stops when the python example stops.

Does that help?

giovanidiniz, 2019-06-24:
SWEEEEEET!!!

This is awesome! Really. Before the renaming the process was a little bit involved but not too hard, but now is super easy.
Thanks so much! 
I tested the takeoff and land example in different udp port and it also works. Assume serial port should work fine as well?

JonasVautherin, 2019-06-24:
Great! Serial port should work as well, indeed!

giovanidiniz, 2019-06-24:
Sweet!
Thanks so much. I'm closing this

closed 2019-06-24 by giovanidiniz