mavlink/MAVSDK-Python #78: backend_bin is not built and error while using MAVSKD-Python

Doodle1106, 2019-06-24:
Hi guys, 

I was trying to give it a test and I followed the instructions detailed in README.md. I can build MAVSDK without any error but I cannot find this particular binary file "./src/backend/src/backend_bin", and it seems that this file has been renamed as 

    grep -r "add_executable" | grep "backend_bin"

gives no result.

Any suggestion would be great, thanks!
Joey

JonasVautherin, 2019-06-25:
Hello! I am currently updating the documentation, we indeed made some renaming last week. `backend_bin` is now called `mavsdk_server`.

Note that for running MAVSDK-Python, you can now just install through pip: `pip3 install mavsdk` :+1:

JonasVautherin, 2019-06-25:
I updated the [README](https://github.com/mavlink/MAVSDK-Python). Closing this, assuming that `pip3 install mavsdk` solves your problem.

Please reopen if you actually want to build from sources.

closed 2019-06-25 by JonasVautherin