mavlink/MAVSDK-Python #81: Offboard Mode Problem (Error received from peer)

bozkurthan, 2019-06-30:
Hello I' tried to switch offboard mode from examples/offboard.py by running `examples/offboard.py`

But I got following error:

```
bozkurthan@bozkurthan:~/Desktop/MAVSDK-Python-master$ python3 examples/offboard.py 
-- Arming
-- Setting initial setpoint
Traceback (most recent call last):
  File "examples/offboard.py", line 217, in <module>
    loop.run_until_complete(run_offb_ctrl_velocity_ned())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "examples/offboard.py", line 38, in run_offb_ctrl_velocity_ned
    await drone.offboard.set_velocity_ned(VelocityNEDYaw(0.0, 0.0, 0.0, 0.0))
  File "/home/bozkurthan/.local/lib/python3.6/site-packages/mavsdk/plugins/offboard.py", line 704, in set_velocity_ned
    response = await self._stub.SetVelocityNed(request)
  File "/home/bozkurthan/.local/lib/python3.6/site-packages/aiogrpc/channel.py", line 40, in __call__
    return await fut
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "{"created":"@1561876519.775117151","description":"Error received from peer ipv4:127.0.0.1:50051","file":"src/core/lib/surface/call.cc","file_line":1046,"grpc_message":"","grpc_status":12}"
>
```

I tried to run with  `python3 examples/offboard.py ` as well but same error appeared. examples/takeoff_and_land.py worked without error.
By the way, I installed MAVSDK with pip3.

Thanks for helping

julianoes, 2019-07-01:
@JonasVautherin do you know what's happening here?

bozkurthan, 2019-07-18:
The error still exists.. FYI @irsdkv

bozkurthan, 2019-07-18:
```
-- Arming
-- Setting initial setpoint
Traceback (most recent call last):
  File "examples/offboard.py", line 217, in <module>
    loop.run_until_complete(run_offb_ctrl_velocity_ned())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "examples/offboard.py", line 38, in run_offb_ctrl_velocity_ned
    await drone.offboard.set_velocity_ned(VelocityNEDYaw(0.0, 0.0, 0.0, 0.0))
  File "/usr/local/lib/python3.6/dist-packages/mavsdk/plugins/offboard.py", line 704, in set_velocity_ned
    response = await self._stub.SetVelocityNed(request)
  File "/usr/local/lib/python3.6/dist-packages/aiogrpc/channel.py", line 40, in __call__
    return await fut
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "{"created":"@1563434020.574566429","description":"Error received from peer ipv4:127.0.0.1:50051","file":"src/core/lib/surface/call.cc","file_line":1052,"grpc_message":"","grpc_status":12}"
>

```

JonasVautherin, 2019-07-18:
> Can it be related to this comment?

It is not.

What are you trying to run? Did you install using `pip3 install mavsdk`? Do you run `mavsdk_server` through `start_mavlink()`, or do you start another binary manually?

bozkurthan, 2019-07-18:
I installed with pip3 install mavsdk on ubuntu 16.04. This is log:
```
hanco@hanco-WS:~$ sudo pip3 install mavsdk
[sudo] password for hanco: 
WARNING: The directory '/home/hanco/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
WARNING: The directory '/home/hanco/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: mavsdk in /usr/local/lib/python3.6/dist-packages (0.1.0)
Requirement already satisfied: aiogrpc>=1.5 in /usr/local/lib/python3.6/dist-packages (from mavsdk) (1.6)
Requirement already satisfied: grpcio>=1.11.0 in /usr/local/lib/python3.6/dist-packages (from mavsdk) (1.22.0)
Requirement already satisfied: importlib-resources>=1.0.2; python_version < "3.7" in /usr/local/lib/python3.6/dist-packages (from mavsdk) (1.0.2)
Requirement already satisfied: protobuf>=3.5.1 in /usr/local/lib/python3.6/dist-packages (from mavsdk) (3.8.0)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio>=1.11.0->mavsdk) (1.10.0)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from protobuf>=3.5.1->mavsdk) (20.7.0)

```

After it I just follow these steps:
1. `sudo make posix gazebo `
2. `python3 examples/offboard.py`

And getting error. Sorry for my misunderstanding but Does start_mavlink() already implemented in offboard example?

https://github.com/mavlink/MAVSDK-Python/blob/dff711afb9e2fa3407b9b5602f582d72670550c5/examples/offboard.py#L27

JonasVautherin, 2019-07-18:
No you're actually right, it's not working with offboard. Let me make a new release today :+1:.

irsdkv, 2019-07-18:
@bozkurthan 
Hm in my case this was fix same issue.

Did you try 
`git submodule update --init --recursive`

?

JonasVautherin, 2019-07-18:
Yes but the one installed from `pip3 install mavsdk` is still wrong :-). Let me release 0.2.0 ASAP!

@bozkurthan in the meantime, you can build mavsdk-python from `master`, as suggested by @irsdkv.

bozkurthan, 2019-07-18:
@irsdkv and @JonasVautherin thanks for kindly helping. I was feeling fool :)

Build finished with warning even I used `git submodule update --init --recursive`

```
hanco@hanco-WS:~/MAVSDK/MAVSDK-Python$ sudo python3.6 setup.py build

/usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'project_urls'
  warnings.warn(msg)
running build
running build_py
running egg_info
writing mavsdk.egg-info/PKG-INFO
writing dependency_links to mavsdk.egg-info/dependency_links.txt
writing requirements to mavsdk.egg-info/requires.txt
writing top-level names to mavsdk.egg-info/top_level.txt
reading manifest file 'mavsdk.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'mavsdk/bin'
warning: no files found matching 'mavsdk/bin/mavsdk_server'
writing manifest file 'mavsdk.egg-info/SOURCES.txt'
```
After that used `sudo python3.6 setup.py build `command.
Here is the new error, cannot find mavsdk_server file:

```
hanco@hanco-WS:~/MAVSDK/MAVSDK-Python$ sudo python3.6 examples/offboard.py 

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/importlib_resources/_py3.py", line 85, in open_binary
    return open(full_path, mode='rb')
NotADirectoryError: [Errno 20] Not a directory: '/usr/local/lib/python3.6/dist-packages/mavsdk-0.1.0-py3.6.egg/mavsdk/bin/mavsdk_server'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/offboard.py", line 27, in <module>
    start_mavlink()
  File "/usr/local/lib/python3.6/dist-packages/mavsdk-0.1.0-py3.6.egg/mavsdk/__init__.py", line 75, in start_mavlink
  File "/usr/lib/python3.6/contextlib.py", line 81, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.6/dist-packages/importlib_resources/_py3.py", line 188, in path
    with open_binary(package, resource) as fp:
  File "/usr/local/lib/python3.6/dist-packages/importlib_resources/_py3.py", line 99, in open_binary
    raise FileNotFoundError(message)
FileNotFoundError: 'mavsdk_server' resource not found in 'mavsdk.bin'

```

I manually started server by using `sudo ./usr/local/lib/python3.6/dist-packages/mavsdk` by it gives error for all example.

JonasVautherin, 2019-07-18:
By calling `setup.py`, you're trying to package it. But I guess you don't want to release a new package, you just want to build `master` :sweat_smile: . Just try to follow ["build and run from sources"](https://github.com/mavlink/MAVSDK-Python#build-and-run-from-sources).

bozkurthan, 2019-07-18:
> But I guess you don't want to release a new package

Oh god, absolutely no :sweat_smile: 
 Thanks for kindly responding. I didn't look at the instruction. I think I need some rest :disappointed_relieved:

JonasVautherin, 2019-07-19:
@bozkurthan: It seems to me that you are on Linux, right? Can you try to update `mavsdk` now? Something like:

```sh
pip install --upgrade mavsdk
```

should give you mavsdk-python 0.2.0, where offboard works (thanks to @irsdkv :blush:)!

bozkurthan, 2019-07-20:
Thanks to both of you my little iris' running sweetly. I'm closing issue thanks for fixing. @irsdkv @JonasVautherin

closed 2019-07-20 by bozkurthan

raul7, 2019-08-09:
Hi,
I'm having the same issue, but with all python examples. I'm on Windows 10 using Anaconda, installed mavsdk using pip. I tried to upgrade to ver. 0.2.0 using "pip install --upgrade mavsdk" but it says "already up to date" and shows that I'm still with ver. 0.1.0.

If I uninstall, and then try to re-install using "pip install mavsdk==0.2.0", I get: 
```
  ERROR: Could not find a version that satisfies the requirement mavsdk==0.2.0 (from versions: 0.1.0)
ERROR: No matching distribution found for mavsdk==0.2.0
```
Any hints about why I can't upgrade to the newer version?

JonasVautherin, 2019-08-09:
Windows is still on version 0.1.0, and therefore only works with `mavsdk_server` coming from pip. If you built `mavsdk_server` from sources, that's most likely the reason why it doesn't work.

However, that's not related to this issue. Would you mind opening a new issue, explaining what you run to get the UNIMPLEMENTED error? :blush: