mavlink/MAVSDK-Python #82: Generate reference docs for Python frontend

julianoes, 2019-07-01:
@JonasVautherin we should generate reference docs for the Python bindings.

JonasVautherin, 2019-07-03:
Agreed. The idea is to generate docs (doxygen-like) from the proto files. I am quite sure it is possible, and I had seen somebody doing that specifically for C#. I'd like to do something similar, but for all our bindings at once.

If somebody wants to look into how to extend our protoc plugin to do that, I'm happy to have a discussion :blush:.

hamishwillee, 2019-08-13:
I'd love to work on this. Just time poor. Python docs  are petty standardised, and most use Sphinx to auto extract comments.

JonasVautherin, 2019-08-13:
Note that we should first get the comments there :sweat_smile:. That's the issue [here](https://github.com/mavlink/MAVSDK-Proto/issues/91).

closed 2019-09-04 by JonasVautherin

julianoes, 2019-09-04:
Awesome, we should also publish them somehow on s3 and link to them.

PR #104 (2019-09-04): Documentation templates
Update the templates to generate documentation from the proto files.

Note: the second commit is the interesting one.

Fixes #82.
PR #195 (2020-05-30): Generate API docs
I had some fun (or not) with sphinx.

@hamishwillee I'm open to generate markdown or xml instead and integrate it into the existing docs but this is better than nothing for now.
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/

Fixes #82, completely.

Next step is to do it in CI.