mavlink/MAVSDK-Python #84: Not possible to connect using serial - ttyUSB with python?

RaulGH25, 2019-07-04:
From the MAVSDK (C++) it is possible to connect an offboard computer (i.e.Odroid XU4) using a serial connection, specifying the correct path (i.e. "/dev/ttyUSB0:57600"). In the Python examples, the connection is always done as "drone = mavsdk_connect(host="127.0.0.1")", which works great with jmavsim or other simulators. However, it does not work with a serial connection to the pixhawk (using a USB to FTDI adapter). Is it possible to entablish such kind of connection from the Python MAVSDK API? How?

Thank you in advance

JonasVautherin, 2019-07-04:
It is possible, actually :blush:. When you call [`start_mavlink()`](https://github.com/mavlink/MAVSDK-Python/blob/master/mavsdk/__init__.py#L71) (I assume you installed mavsdk using `pip3 install mavsdk`, let me know if that's not true).

So `start_mavlink("/dev/ttyUSB0:57600")` should do it!

hamishwillee, 2019-07-05:
@JonasVautherin Since we don't yet autogenerate docs, setting up the link should be documented in the readme https://github.com/mavlink/MAVSDK-Python - hint!

JonasVautherin, 2019-07-22:
@RaulGH25: is that working?

RaulGH25, 2019-07-22:
Yes, it is working very well. Thank you!

closed 2019-07-22 by JonasVautherin