mavlink/MAVSDK-Python #88: pip3 install error

SelimOzel, 2019-07-17:
Am I doing this right? Please see the attached image.

[image]

JonasVautherin, 2019-07-17:
Nope, you're apparently doing it right, but pip decided that your system was not compatible. What system are you running on? Windows 10, 64 bits?

SelimOzel, 2019-07-17:
Yes, that's the system! Thanks.

JonasVautherin, 2019-07-17:
Ok, I need to have a look at that. I'll keep you posted ASAP!

SelimOzel, 2019-07-17:
No worries! And thanks once more!

julianoes, 2019-07-18:
I tried to reproduce this on Windows 10 64 bit with Python 3.7.4 and it worked successfully:

```
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.0.3
** Copyright (c) 2019 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>python
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> quit()

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>pip3 install mavsdk
Collecting mavsdk
  Downloading https://files.pythonhosted.org/packages/54/e1/7a982a0728b54231bb5ce8ceb89dec92d22829d9b047f58e85f34509e72c/mavsdk-0.1.0-py3-none-win_amd64.whl (3.9MB)
    100% |████████████████████████████████| 3.9MB 723kB/s
Collecting aiogrpc>=1.5 (from mavsdk)
  Downloading https://files.pythonhosted.org/packages/06/73/aa687dd6f103e51c50703c43936f26bc13e563ac0489e74b7a3398eaf391/aiogrpc-1.6-py3-none-any.whl
Collecting protobuf>=3.5.1 (from mavsdk)
  Downloading https://files.pythonhosted.org/packages/5a/5b/2ffcea713a542179878d0f03e7c6e50d09ca470b36e730f94ed51e51b35d/protobuf-3.9.0-cp37-cp37m-win_amd64.whl (1.0MB)
    100% |████████████████████████████████| 1.0MB 66kB/s
Collecting grpcio>=1.11.0 (from mavsdk)
  Downloading https://files.pythonhosted.org/packages/7a/f5/fe046577387a3589ab3092096ca423fcf9a8c7ac876f56c6f3b4c9b9e533/grpcio-1.22.0-cp37-cp37m-win_amd64.whl (1.6MB)
    100% |████████████████████████████████| 1.6MB 1.6MB/s
Collecting six>=1.9 (from protobuf>=3.5.1->mavsdk)
  Downloading https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in c:\users\julian oes\appdata\local\programs\python\python37\lib\site-packages (from protobuf>=3.5.1->mavsdk) (40.8.0)
Installing collected packages: six, grpcio, aiogrpc, protobuf, mavsdk
Successfully installed aiogrpc-1.6 grpcio-1.22.0 mavsdk-0.1.0 protobuf-3.9.0 six-1.12.0
You are using pip version 19.0.3, however version 19.1.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>python
Python 3.7.4 (tags/v3.7.4:e09359112e, Jul  8 2019, 20:34:20) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import mavsdk
>>> help(mavsdk)
Help on package mavsdk:

NAME
    mavsdk - # -*- coding: utf-8 -*-

PACKAGE CONTENTS
    _base
    async_plugin_manager
    bin (package)
    generated (package)
    plugins (package)

SUBMODULES
    action
    action_pb2
    action_pb2_grpc
    ...
```

JonasVautherin, 2019-07-18:
@SelimOzel: what's the output of `pip3 --version` and `python3 --version`?

SelimOzel, 2019-07-20:
[image]

It should be `pip 19.1.1` and `python 3.7.2`. I am not certain if all path variables are correct on the particular laptop I'm using. I haven't used python in a while with this laptop. I'll give this command another shot with the setup I got home.

JonasVautherin, 2019-07-20:
But Python 3.6+ should be fine. Here somehow pip decides that your machine is not compatible, and I don't get why :thinking:...

SelimOzel, 2019-07-21:
Just to be clear. This is not a critical problem for me. I just wanted to give more data points to work with to whoever ends up solving it.

I tested following commands in the image on my home setup with a Windows 10, x64 PC. Still getting the same error. 

[image]

bozkurthan, 2019-07-21:
I'm here for same Issue for raspberry pi 4. I tried to download and install manually but no chance.

[image]

I tried to follow this instruction but it gives me error too:

> Build and run from sources
> Note: this is more involved and targetted to contributors.
> 
> Get the Python wrapper
> Clone this repo and recursively update submodules:
> 
> git clone https://github.com/mavlink/MAVSDK-Python --recursive
> cd MAVSDK-Python
> Install prerequisites
> First install the protoc plugin (protoc-gen-dcsdk):
> 
> cd proto/pb_plugins
> pip install -r requirements.txt
> pip install -e .
> You can check that the plugin was installed with $ which protoc-gen-dcsdk, as it should now be in the PATH.
> 
> Then go back to the root of the repo and install the dependencies of the SDK:
> 
> cd ../..
> pip install -r requirements.txt -r requirements-dev.txt
> Generate wrapper
> Run the following helper script. It will generate the Python wrappers for each plugin. If the submodules are not initialized already, the script will take care of it.
> 
> ./other/tools/run_protoc.sh
> Install the package locally
> After generating the wrapper you can install a development (editable) version of the package using:
> 
> pip install -e .
> 

On ubuntu bionic it was running well.

JonasVautherin, 2019-07-21:
> I'm here for same Issue for raspberry pi 4(raspbian buster). I tried to download and install manually but no chance.

That's normal, we don't distribute for arm (yet). We have linux_x86_64, windows 64 bits and macOS. Now I don't get why the pip of @SelimOzel doesn't detect it as windows-compatible.

@SelimOzel: Could you try to download the wheel [here](https://pypi.org/project/mavsdk/0.1.0/#files) and install it manually?

```sh
pip install mavsdk-0.1.0-py3-none-win_amd64.whl
```

SelimOzel, 2019-07-22:
My system type is x64-based PC and CPU is i7-7700 as written in System Information panel of Windows 10. I downloaded the .whl file in the link you provided into C:\>Wheel and run the install command from there. Unfortunately it still is complaining about platform and support. :(

[image]

iemre98, 2019-07-22:
Can't we use with raspberry pi?

JonasVautherin, 2019-07-22:
> Can't we use with raspberry pi?

You can, but you have to compile `mavsdk_server` manually, as we don't provide it with the pip package yet.

JonasVautherin, 2019-07-22:
@SelimOzel can you try to see if a verbose call of pip gives useful info?

```sh
pip3 -vvv install mavsdk
```

julianoes, 2019-07-24:
[image]

v2h, 2019-08-01:
I'm having the same problem on both my personal laptop and work laptop (Windows 10 64 bit, python version 3.7.4). Here is the output for `pip3 -vvv install mavsdk` :

[image]

JonasVautherin, 2019-08-02:
Thanks @v2h for the input!

It seems like it tries to install `mavsdk-0.1.0-py3-none-macosx_10_9_x86_64.whl`, which obviously doesn't match. Interestingly, it does not mention the other variants.

Would you mind trying to [download the wheel](https://files.pythonhosted.org/packages/54/e1/7a982a0728b54231bb5ce8ceb89dec92d22829d9b047f58e85f34509e72c/mavsdk-0.1.0-py3-none-win_amd64.whl) directly and install it with the following command?

```sh
$ pip3 -vvv install mavsdk-0.1.0-py3-none-win_amd64.whl
```

v2h, 2019-08-02:
@JonasVautherin here are the outputs from `$ pip3 -vvv install mavsdk-0.1.0-py3-none-win_amd64.whl`:

[image]

JonasVautherin, 2019-08-02:
Should it have another name? Like `win_x86-64` or `win-arm64` or something like that?

Would you mind trying different solutions? Or maybe there is a list somewhere online (I'm on holiday, I don't have access to a computer right now :blush:)

v2h, 2019-08-02:
@JonasVautherin `win_x86-64` didn't work either. Enjoy your holiday (:  I'll try to figure it out and let you know.

julianoes, 2019-08-05:
@v2h why are you using `python37-32`? This looks like it's the 32 bit version of Python.

julianoes, 2019-08-05:
I can reproduce this with Python 3.7 32 bit:

[image]

julianoes, 2019-08-05:
@v2h I think in the longer term we would like to support 32 bit as well but for now I suggest to install the Python 64 bit version to workaround this.

v2h, 2019-08-05:
@julianoes @JonasVautherin ahhh silly me, I was messing around with Python 2 and 3, and then mistakenly installed the 32 bit version. Reinstalled Python 64 bit and everything works now. Thanks alot!!

julianoes, 2019-08-05:
Yay, glad! I'll close this then.

closed 2019-08-05 by julianoes

MaEtUgR, 2019-08-07:
I had the same problem, thanks to @JonasVautherin for hinting me towards the 32/64-Bit issue. I think the root cause of the problem is the python website which offers a huge download Python for windows button that when pressed installs you the 32-Bit version (in 2019!). It also doesn't mention that it's 32-Bit neither on the button nor in the installer file name. It should offer also a 64-Bit button otherwise people including me are tempted to just press whatever seems to be default.
https://www.python.org/downloads/

JonasVautherin, 2019-08-07:
Maybe we should also publish a 32 bits artifact for Windows? Does that make sense?

julianoes, 2019-08-08:
> Maybe we should also publish a 32 bits artifact for Windows? Does that make sense?

Yes, I think that will spare some troubles.

rahulswimmer, 2019-11-05:
Apparently the package I was looking for was getting installed for a different pip url. Hence I pointed my pip command to that url inside pip.conf file and it solved my problem.

qsqqwqeqe, 2020-01-19:
ERROR: Could not find a version that satisfies the requirement avbin (from versions: none)
ERROR: No matching distribution found for avbin.Can u please tell me the issue anyone????

julianoes, 2020-01-19:
@qsqqwqeqe please don't necrobump but create a new issue with proper log information.

jagadeeshdevula, 2020-09-06:
Hi,
While installing the GitHub-dorks.py package with ( pip3 install github-dorks.py ), got some error like
python version : Python 3.8.5

ERROR: Could not find a version that satisfies the requirement github-dorks.py (from versions: none)
ERROR: No matching distribution found for github-dorks.py
Please help me to solve this issue.

Thank you.

julianoes, 2020-09-07:
Please don't necrobump.