mavlink/MAVSDK-Python #92: Get single latest telemetry item

jarednm, 2019-07-19:
Is there a way to make a call to just get the latest telemetry (e.g. position)?  I'd rather not create an entire co-routine using `async for position in drone.telemetry.position():` syntax.

It seems that currently, drone.telemetry.position() uses `yield` rather than `return` so the`next()` data point will not be the latest, but will be the first one since the last time it was called.

JonasVautherin, 2019-07-20:
I don't think you can use `position()` as a polling interface unless you do something like:

```python
async for position in drone.telemetry.position():
    return position
```

It is probably fine if you call that only once, but not ideal if you call it in a loop. The point is that we provide telemetry as a stream, and you have to consume it.

But you could totally update your state from that stream and write your own polling interface allowing you to get the latest value whenever you want, right?

JonasVautherin, 2019-08-30:
Closing for inactivity. Feel free to re-open if needed.

closed 2019-08-30 by JonasVautherin

vcrann, 2021-11-01:
Hi @JonasVautherin 
you reccomended writing a polling interface to retreive the latest value, would you be able to give me an idea of what that would look like? We are trying to write a program that will need to call a function that returns the most recent position value and we can't quite get our heads around how to do this from the telemetry stream.
Thank you in advance.
Verdon

JonasVautherin, 2021-11-01:
Something like:

```python
async for position in drone.telemetry.position():
    state.position = position
```

in a coroutine, and then when you want to access it, you do something like:

```python
print(state.position)
```

Does that make sense?

vcrann, 2021-11-01:
That's brilliant thanks!