mavlink/MAVSDK-Python #93: aiogrpc pip error

crisdeodates, 2019-07-20:
While compiling from source, I found an error:

```
Collecting aiogrpc>=1.5 (from -r requirements.txt (line 2))
  ERROR: Could not find a version that satisfies the requirement aiogrpc>=1.5 (from -r requirements.txt (line 2)) (from versions: none)
ERROR: No matching distribution found for aiogrpc>=1.5 (from -r requirements.txt (line 2))
```
I  manually install aiogrpc using "pip3". Seems like there is no aiogrpc in the pip verion.

JonasVautherin, 2019-07-20:
Works for me... :thinking:

Just now, in a venv:

```
jones@gemini mavsdk-python % python -m venv venv                                                                                                                      [0]
jones@gemini mavsdk-python % source venv/bin/activate                                                                                                                 [0]
(venv) jones@gemini mavsdk-python % pip install -r requirements.txt                                                                                                   [0]
Ignoring importlib-resources: markers 'python_version < "3.7"' don't match your environment
Collecting protobuf>=3.5.1 (from -r requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/95/ad/682b9dd1bc94cc7d1ecab8bee83ae8e6ae16161ff7a4131207ae96e256bb/protobuf-3.9.0-cp37-cp37m-manylinux1_x86_64.whl
Collecting aiogrpc>=1.5 (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/06/73/aa687dd6f103e51c50703c43936f26bc13e563ac0489e74b7a3398eaf391/aiogrpc-1.6-py3-none-any.whl
Collecting grpcio>=1.11.0 (from -r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/9d/b1/b80dea9e0bbbdd07bf7ba69c6df1aeb3e88b90b85ca326c40be9e29bc37c/grpcio-1.22.0-cp37-cp37m-manylinux1_x86_64.whl
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from protobuf>=3.5.1->-r requirements.txt (line 1)) (40.8.0)
Collecting six>=1.9 (from protobuf>=3.5.1->-r requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Installing collected packages: six, protobuf, grpcio, aiogrpc
Successfully installed aiogrpc-1.6 grpcio-1.22.0 protobuf-3.9.0 six-1.12.0
You are using pip version 19.0.3, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command
```

xdwgood, 2019-07-23:
I also encountered the same problem.

julianoes, 2019-07-23:
@crisdeodates and @xdwgood on what system is that?

Can you paste the whole output, as well as `python3 --version` and `pip3 --version`, and make sure you run everything with `python3` and `pip3`.

crisdeodates, 2019-07-23:
@julianoes , was about to post.. Solved the issue by
```
sudo apt install python3-pip
sudo pip3 install aiogrpc
```
and then running the python examples using "python3" instead of python

Feels like MAVSDK has issues in Python 2.X, but works perfectly in Python 3.X

julianoes, 2019-07-23:
Yes, always Python3.

@JonasVautherin we should change all instructions to always show python3, that might get rid of some of the problems!

closed 2019-07-23 by julianoes

JonasVautherin, 2019-07-23:
Well, the README starts by saying that in "Important Notes", and this issue started with "I manually installed aiogrpc using "pip3". Seems like there is no aiogrpc in the pip verion.", which is why I did not look into the python version.

If we put `python3` everywhere in the instructions, people on Ubuntu 16.04 will be running `python3.5` and that will fail again So should we then write `[python3.6|python3.7]` everywhere? 😅

That's why we release it on pip, so that people can go for the simpler `pip install mavsdk`.