mavlink/MAVSDK-Python #97: Performance benchmark

giovanidiniz, 2019-07-23:
Hi guys, 
I'm playing around with a ROS2 node using the MAVSDK-Python to take care of the mavlink side of things. 
I was wondering if any of you have done a performance benchmark with telemetry broadcasting (global position, battery, system health, etc).
The highest rate I can get so far is about 0.5 Hz. 
I don't think the intent behind the SDK is to provide high-freq control (at least not for now) but those rates super slow even for high-level mission control

Any ideas/suggestions on that would be really appreciated! 

thanks

julianoes, 2019-07-24:
The rate depends on the rate that the MAVLink messages are sent over the link. Therefore, I'd do the measurements on something that is sent at a higher rate such as attitude.
I would expect (or hope) to get rates at least around 20 Hz if not than something is presumably wrong and needs fixing.

I would however not expect the language wrappers to do high(er) performance things at e.g. 250 Hz, for that I would suggest to directly use the C++ API.

JonasVautherin, 2019-08-30:
Closing this, feel free to re-open if needed.

closed 2019-08-30 by JonasVautherin