mavlink/MAVSDK-Python #100: UDP, serial port connection

HanbumKo, 2019-07-29:
I installed mavsdk(0.2.0) at Python3.6(/usr/bin/python3.6) using pip.
And I start 2 SITL(jmavsim), one drone has udp:127.0.0.1:14540, the other drone has udp:127.0.0.1:14541.
The problem is that when I do
```
start_mavlink()
drone = mavsdk_connect(host="127.0.0.1")
```
It works well with the drone using 14540 port. (When I put "127.0.0.1:14540", it shows error)

But I also wanted to connect with 14541, but it shows error. (host="127.0.0.1:14541")
And
I connected actual drone (Yuneec H520, pixhawk4 1.8.0) with command
`drone = mavsdk_connect(host="serial:///dev/ttyACM0")`
it also didn't work. I checked the actual drone is using /dev/ttyACM0.

How can I solve this??

HanbumKo, 2019-07-29:
I also tried
`"udp://127.0.0.1:14541"`
but same result

julianoes, 2019-07-29:
Sorry this is indeed confusing, see #86.

But basically:
```
start_mavlink(connection_url="serial:///dev/ttyACM0")
drone = mavsdk_connect(host="127.0.0.1")
```

@JonasVautherin do you agree now that it's not optimal :smile:.

HanbumKo, 2019-07-29:
What a fast response!!

Oh, I misunderstood the codes. I've been trying like
```
start_mavlink()
drone = mavsdk_connect(host="blahblah")
```
After changing codes like
```
start_mavlink(connection_url="udp://127.0.0.1:14541")
drone = mavsdk_connect(host="127.0.0.1")
```
It works well.

But one more question, then, how can I deal with multiple drones at the same time? Should I run python codes separately?

julianoes, 2019-07-30:
That's currently not supported in Python yet because it means you need to run the `mavsdk_server` multiple times on different gRPC ports.

I've created an issue for this request #102, and I'll close this one.

closed 2019-07-30 by julianoes