mavlink/MAVSDK-Python #98: Error - No module named SDK

eanmikale, 2019-07-29:
Hi, 
We are having issues with the Python-wrapper. We have the wrapper installed in the MAVSDK src folder, upgraded to python3.7 from 3.5 to run it, and we keep getting a "No module named MAVSDK" error message. Please see the output below, and any assistance is greatly appreciated. 

Python 3.7.2 (default, Jul 28 2019, 17:34:51) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
---
This console is running in an asyncio event loop.
It allows you to wait for coroutines using the 'await' syntax.
Try: await asyncio.sleep(1, result=3)
---
>>> from mavsdk import connect
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aioconsole-0.1.15.dev0-py3.7.egg/aioconsole/execute.py", line 95, in aexec
    result, new_local = yield from coro
  File "<aexec>", line 2, in __corofn
ModuleNotFoundError: No module named 'mavsdk'
>>> import mavsdk
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aioconsole-0.1.15.dev0-py3.7.egg/aioconsole/execute.py", line 95, in aexec
    result, new_local = yield from coro
  File "<aexec>", line 2, in __corofn
ModuleNotFoundError: No module named 'mavsdk'

JonasVautherin, 2019-07-29:
Did you run `pip3.7 install mavsdk`?

>  ModuleNotFoundError: No module named 'mavsdk'

This usually means that `mavsdk` is not installed.

julianoes, 2019-07-29:
> upgraded to python3.7 from 3.5 to run it

In that case make sure to re-install it with python3.7. The installation might only be for 3.5.
Try:
`pip3.7 install mavsdk` or check the version of your pip: `pip3 --version`

JonasVautherin, 2019-08-29:
@eanmikale: Any news about this issue?

JonasVautherin, 2019-08-30:
Well, that issue seems inactive, so I'll close it. Feel free to reopen if the problem comes back!

closed 2019-08-30 by JonasVautherin

julianoes, 2019-09-03:
Closing, please re-open if the problem remains.

chris-aeviator, 2019-11-10:
@eanmikale I experienced the same issue after trying to re-install the package manually after I was experiencing issues with the pip version (I only believed that they where not issues with the pip version itself). To sucessfully get rid of the `no module` message I had to run `pip3 install --force mavsdk` to force a re-download and installation