mavlink/MAVSDK-Python #103: Missing dll on windows

Benjdu76, 2019-07-30:
When I run an example, I get errors which said that :
- libgcc_s_seh-1.dll is missing
- libwinpthread-1.dll is missing
- libstdc++-6.dll is missing

I also have the following error in the python shell :
`grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "failed to connect to all addresses"
        debug_error_string = "{"created":"@1564499126.880000000","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3528,"referenced_errors":[{"created":"@1564499126.879000000","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":399,"grpc_status":14}]}"`

I run the script on python 3.7 with mavsdk 0.1.0 installed with pip3 install.
I tried on windows 10 and windows 7 with the same issue.

JonasVautherin, 2019-07-30:
Building with Visual Studio? Would you have an idea what is missing in our CMakeLists for this to work correctly?

Benjdu76, 2019-07-31:
I installed the package from pip. But there is no need to build the package from source ? 
Anyway I will try to see if it changes something

JonasVautherin, 2019-07-31:
> I installed the package from pip.

Right, sorry I missed that part. It's an issue at the `mavsdk_server` level: we are apparently not building it such that it runs standalone on Windows (because it relies on those DLLs you mention).

Let me transfer that to the MAVSDK repo.