mavlink/MAVSDK-Python #106: Support for import_qgroundcontrol_mission

petergerten, 2019-09-04:
Hi,

it seems there is currently no support for the import_qgroundcontrol_mission() method.
I am a bit confused on how this works. If the wrapper is auto-generated, wouldn't it have the same functionality as the "main" MAVSDK?

Peter

JonasVautherin, 2019-09-04:
Hello!

The support exists in C++, but not yet in the other languages. Maybe a part of [this presentation](https://www.youtube.com/watch?v=GXt-eXJ9vfg&feature=youtu.be&t=299) can help on the wording there.

To summarize, it is planned to autogenerate everything down to the C++ API, meaning that everything we have in C++ will always exist in all other languages. Currently, the "glue code" (or what I call "black box" in the presentation) is not auto-generated, so it has to be written manually. And QGC mission import was not added there (yet).

We would be happy if you could contribute that! The first step would be to add the definition in the corresponding [proto file](https://github.com/mavlink/MAVSDK-Proto/tree/master/protos), and then write the "glue code" (gRPC) :+1:.

Does that make sense?

JonasVautherin, 2020-05-13:
It is now in 0.7.0, though I realize now that it works in a weird way: `await drone.mission.import_qgroundcontrol_mission(qgc_plan_path)` takes the path to the qgc plan, which will be opened by `mavsdk_server`. I'm not sure what's the best way to improve that. But assuming that you run MAVSDK-Python and `mavsdk-server` on the same machine, it should be fine.

We would be glad if you could try it and let us know if it works!

closed 2020-05-13 by JonasVautherin