mavlink/MAVSDK-Python #107: Is there set_configuration() function in MAVSDK-Python ?

minimal1, 2019-09-13:
I wanna use MAVSDK as CompanionComputer, not as GroundStation.

I just got to know set_configuration function in MAVSDK-C++.

So, I thought that would be in MAVSDK-Python, too.
But I couldn't find that. Is there set_configuration() function in MAVSDK-Python ?

JonasVautherin, 2019-09-14:
You're right, this is not available from the language bindings (including Python). I guess that could be added as part of the ["core" plugin](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/core/core.proto)...

@julianoes: any opinion on that?

minimal1, 2019-09-16:
I see.
I just did hardcoding in the C++ code to set  the MAVSDK-Python to 'CompanionComputer', not 'GroundStation'.

Thank for replaying.

JonasVautherin, 2019-09-16:
>  just did hardcoding in the C++ code

That's a reasonable workaround.

I'm now wondering if that should be added as a feature of the core plugin, or as a configuration option when starting `mavsdk_server` :thinking:...

julianoes, 2019-09-17:
@JonasVautherin what is the core plugin? Is that an actual thing? Otherwise adding it as a config for the `mavsdk_server` seems good.

JonasVautherin, 2019-09-17:
It is some kind of [weird plugin](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/core/core.proto) :sweat_smile:. That's the one you use to get the `ConnectionState` (<`uuid`, `is_connected`> pair). It doesn't really map one-to-one to the C++ core impl, not sure yet how it will evolve.

Adding it as a config to `mavsdk_server` makes it even more manual to maintain. We may have to manually write the gRPC part of the core plugin (even when the others are generated), but at least that's only on the gRPC side. Adding a config to `mavsdk_server` adds manual code on all the bindings.