mavlink/MAVSDK-Python #110: More than one nested enum cause error.

irsdkv, 2019-09-28:
Putting more than one `enum` in proto `message` cause error.
Example:
```proto
message TestMessage {
     enum FirstEnum {
          STUB_FIRST = 0;
    }

    enum OneMoreEnum {
         STUB_ONE_MORE = 0;
    }
    FirstEnum first_enum_val = 1;
    OneMoreEnum second_enum_val = 2;
}
```
 Run protoc:
```bash
$ ./other/tools/run_protoc.sh
```
Error:
```python
Traceback (most recent call last):
  File "mavsdk-python/.venv/bin/protoc-gen-dcsdk", line 11, in <module>
    load_entry_point('protoc-gen-dcsdk', 'console_scripts', 'protoc-gen-dcsdk')()
  File "mavsdk-python/proto/pb_plugins/protoc_gen_dcsdk/__main__.py", line 17, in main
    AutoGen.generate_reactive(request).SerializeToString())
  File "mavsdk-python/proto/pb_plugins/protoc_gen_dcsdk/autogen.py", line 34, in generate_reactive
    docs = Docs.collect_docs(proto_file.source_code_info)
  File "mavsdk-python/proto/pb_plugins/protoc_gen_dcsdk/docs.py", line 60, in collect_docs
    docs['structs'][struct_id]['enums'][nested_enum_id]['params'].append(
IndexError: list index out of range
```

JonasVautherin, 2019-09-28:
Oh, that's good to know. I'll have a look at the protoc plugin.

julianoes, 2021-03-26:
I would assume this has been resolved in the meantime?

JonasVautherin, 2021-03-29:
Yes it is! With `protoc-gen-mavsdk 1.0.1`

closed 2021-03-29 by JonasVautherin