mavlink/MAVSDK-Python #112: PyPI package needs updating?

Maef, 2019-09-29:
The package in the Python Package Index still contains references to the old names (Drone vs now System). Running those examples will give errors on 'System'. Updating the PyPI package will bring the 'pip install mavsdk' back in sync with the current examples.

EkpoEsua, 2019-10-01:
I did some findings and discovered that System.py module is not available in the downloaded mavsdk package and there was no System class in the init.py module either.

JonasVautherin, 2019-10-03:
Version 0.3.0 got released (for Linux, for now)! Please try it and give your feedback here!

* `pip install mavsdk` to install
* `pip install --upgrade mavsdk` to update

The examples (with `System`) will work with 0.3.0.

closed 2019-10-03 by JonasVautherin

EkpoEsua, 2019-10-03:
Alright, great!, right away, Sire.

JonasVautherin, 2019-10-03:
> I think it only works with pip3 command

Right, it actually depends on the default version on your system. But you're right, MAVSDK requires python 3.

alexander-wu-aw, 2019-10-11:
For some reason, I can't upgrade to mavsdk 0.3.0.

Whenever I try `pip install mavsdk` or `pip install --upgrade mavsdk`, it only gives me version 0.2.0.

When I run `pip install mavsdk==0.3.0` it says:

"RROR: Could not find a version that satisfies the requirement mavsdk==0.3.0 (from versions: 0.1.0, 0.2.0)
ERROR: No matching distribution found for mavsdk==0.3.0"

Traveller30, 2019-10-27:
what about windows?
the same problem exist there , how can we fix it?