mavlink/MAVSDK-Python #115: Generate wrapper again at raspberry pi 3b+ will get weird thing

yingshaoxo, 2019-10-06:
The system image is `2019-07-10-raspbian-buster-lite.iso`

The python version is  `Python 3.7.3 (default, Apr  3 2019, 05:39:12)`

The generate command I was using is `./other/tools/run_protoc.sh`

The result I got is something like this: `from . from . import geofence_pb2 as geofence__pb2`, which cause a syntax error.

If I do the generation process again, I'll get something like this: `from . from . from . import geofence_pb2 as geofence__pb2`, which also cause the error of cause.

JonasVautherin, 2019-10-07:
Hmm that's interesting... I don't think it happens to me.

This said, I just merged #114, so you should be able to use it without having to generate the code yourself: [examples/geofence.py](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/geofence.py).

Note that there is a [bug in `mavsdk_server`](https://github.com/mavlink/MAVSDK/issues/871) regarding the geofence (it does upload the geofence, but crashes at the end).

Would be awesome if you could have a look at it (maybe @coderkalyan has some insights, too).

yingshaoxo, 2019-10-08:
Man, I don't know what you are talking about.

But to be honest, the best way to solve this problem is don't run this command: `./other/tools/run_protoc.sh`

It's OK for just using the default py files at `https://github.com/mavlink/MAVSDK-Python/tree/master/mavsdk/generated`

JonasVautherin, 2019-10-08:
> It's OK for just using the default py files

Yes, that's mostly what I meant. Sorry for the confusion :sweat_smile:

closed 2019-10-08 by JonasVautherin