mavlink/MAVSDK-Python #116: How can I control the drone in real time instead of sending commands asynchronously?

yingshaoxo, 2019-10-11:
Say, I have a camera(OpenMV), I can get some data from that camera.

I need to fly the drone according to the real-time data transferred from the camera sensor.

The problem is MAVSDK-Python is asynchronous, but the camera program is synchronous.

Since MAVSDK-Python doesn't have any documentation, meanwhile the example of this use case is not covered, I really don't know how to make it happen.

julianoes, 2019-10-11:
I suppose if you do `await` before each async call it basically makes it a blocking/sync call, right @JonasVautherin?

JonasVautherin, 2019-10-11:
> Since MAVSDK-Python doesn't have any documentation, meanwhile the example of this use case is not covered, I really don't know how to make it happen.

I think that all the functions are documented, e.g. see [here](https://github.com/mavlink/MAVSDK-Python/blob/master/mavsdk/generated/action.py#L210-L221).

The use of asyncio is not documented in MAVSDK because that's part of the [Python standard library](https://docs.python.org/3/library/asyncio.html) and has [numerous](https://hackernoon.com/a-simple-introduction-to-pythons-asyncio-595d9c9ecf8c) [tutorials](https://realpython.com/async-io-python/) [available](https://pythonprogramming.net/asyncio-basics-intermediate-python-tutorial/) already, so we probably wouldn't make a better job here :innocent:.

This said, it seems to me like your issue is that you need to mix non-asyncio modules with MAVSDK, in which case I would advise you to have a look at [this stackoverflow question](https://stackoverflow.com/questions/41063331/how-to-use-asyncio-with-existing-blocking-library).

To paraphrase the accepted answer there, you'll need an executor, e.g.:

```python
executor = concurrent.futures.ThreadPoolExecutor(max_workers=5)
```

And you'll need to run your blocking functions inside of it, e.g.:

```python
loop.run_in_executor(executor, <your_blocking_function>, 12)
```

An alternative would be to wrap asyncio behind blocking functions (but I believe that's less elegant). For instance by having a class that defines `drone` and the `loop`, and then wrapping e.g. `arm()` into something like:

```python
def arm(self):
    async def arm_aio():
        await self.drone.action.arm()

    self.loop.run_until_complete(arm_aio())
```

yingshaoxo, 2019-10-12:
I see, thanks, I'll try it out.

___

But there is still have something that I can't fully understand:

In this example: https://github.com/mavlink/MAVSDK-Python/blob/master/examples/telemetry_takeoff_and_land.py

It created a parallel task by `asyncio.ensure_future(print_altitude(drone))`

Then, it will call the following function whenever the altitude gets change:
```
async def print_altitude(drone):
    """ Prints the altitude when it changes """

    previous_altitude = None

    async for position in drone.telemetry.position():
        altitude = round(position.relative_altitude_m)
        if altitude != previous_altitude:
            previous_altitude = altitude
            print(f"Altitude: {altitude}")
```

The problem is how does the program know the position has changed? Since it's only a `for item in a_list`? The list should be a fixed-length object. And the function should be called once and return immediately.

It seems like the `drone.telemetry.position()` is infinite in an async architecture. If that was true, how should I get the `drone.telemetry.position()` just for once in a blocking function?

JonasVautherin, 2019-10-12:
> It seems like the drone.telemetry.position() is infinite in an async architecture.

It is an "async generator", and effectively behave like an infinite stream :+1:.

> How should I get the drone.telemetry.position() just for once in a blocking function?

[This example](https://github.com/mavlink/MAVSDK-Python/blob/master/examples/takeoff_and_land.py#L12-L16) does something similar in the beginning:

```python
print("Waiting for drone...")
async for state in drone.core.connection_state():
    if state.is_connected:
        print(f"Drone discovered with UUID: {state.uuid}")
        break
```

So you could just `break` after you get the first value. Or if you want to regularly poll it, you could have a coroutine populating your state somewhere, and poll the value from that, I would say.

yingshaoxo, 2019-10-12:
You are right.

If we want to get the value once even if it blocks the main process, we could use something like this:
```python3
previous_altitude = None
async def print_altitude(drone):
    """ Prints the altitude when it changes """
    global previous_altitude
    async for position in drone.telemetry.position():
        print("reading...")
        altitude = round(position.relative_altitude_m)
        if altitude != previous_altitude:
            previous_altitude = altitude
            print(f"Altitude: {altitude}")
            break # use return if you want to return a value
```

or if we want to get the value `new_altitude` constantly, we could do this:
```python3
asyncio.ensure_future(print_altitude(drone)) # create a parallel task

new_altitude = None
async def print_altitude(drone):
    """ Prints the altitude when it changes """
    global new_altitude

    previous_altitude = None

    async for position in drone.telemetry.position():
        print("reading...")
        altitude = round(position.relative_altitude_m)
        if altitude != previous_altitude:
            previous_altitude = altitude
            print(f"Altitude: {altitude}")
            new_altitude = altitude
```

___

Now everything is clear to me, thank you again for helping me.

___

For those who don't know asyncio in Python3, I would recommend you take a look at:
https://python.readthedocs.io/en/stable/library/asyncio-task.html#example-chain-coroutines
https://docs.python.org/3/library/asyncio-task.html#coroutines

closed 2019-10-12 by yingshaoxo

JonasVautherin, 2019-10-12:
Awesome! :tada: