mavlink/MAVSDK-Python #117: Provide an arm64 version on PyPI

jbeyerstedt, 2019-10-11:
Currently there are just versions of the mavsdk package in PyPI for linux, windows and macOS using the x86_64/ amd64 architecture. Is it possible to also provide a linux version for arm64 and armhf (older raspberry pi).

And maybe you should add to the pip readme section, what operating systems and processor architectures are supported, so you don't have to dig through the pip verbose output to find, what the issue really is.

Thanks

julianoes, 2019-10-11:
This would indeed be nice.

The instructions to build for arm64 would be something along the lines of:

```
docker run --rm dockcross/linux-arm64 > ./dockcross-linux-arm64
./dockcross-linux-arm64 cmake -Bbuild -S. -DBUILD_BACKEND=ON
./dockcross-linux-arm64 cmake --build build
```

And then we just need to figure out how to get the artefacts released and into PyPi.

JonasVautherin, 2019-10-11:
Is there an equivalent to `manylinux-x86_64` on arm? At least I found [some discussions](https://github.com/pypa/manylinux/issues/84) about it... Not sure :thinking:.

Also I'm curious about whether we could have `pip` build `mavsdk_server` from sources, but I've never done that so I don't know if we can tell pip to `git clone` and then run `cmake` and `cmake --build`. Ideas?

hrw, 2019-10-25:
@JonasVautherin manylinux2014 will solve that problem. WIP.

julianoes, 2019-11-04:
@hrw thanks for the note.
https://www.python.org/dev/peps/pep-0599/

hrw, 2019-12-17:
manylinux2014 is published already. Any plans to use it?

JonasVautherin, 2019-12-17:
Yes, through dockcross. I started looking into that last week, hopefully I can spend some more time on it soon!

JonasVautherin, 2019-12-17:
@hrw: working on it, I got `dockcross-manylinux2014-x64` to work on my machine (was pretty easy actually). Next is aarch64! See https://github.com/dockcross/dockcross/issues/358!

BTW, do you know if there is a plan in supporting 32 bits arm architectures as well (e.g. armv7)?

hrw, 2019-12-17:
PEP599 lists armv7l but I do not track how it goes.

hrw, 2019-12-17:
Travis CI does not support 32bit arm could be one of reasons.

JonasVautherin, 2019-12-17:
I think it should not be that. The point of the docker image is to cross-compile, so it doesn't have to run on an arm system... Good to know about PEP599, thanks for the information :-)

hrw, 2019-12-17:
Maybe for you it is to cross-compile. I use them natively. Either on available machines or on CI systems.

JonasVautherin, 2019-12-17:
Oh, I see. Thanks for the explanation :blush:. Well, we'll see if that happens at some point. Right now I have issues running the aarch64 docker image :confused:. But at least the x64 seems to be working!

hrw, 2020-02-14:
Any update @JonasVautherin?

JonasVautherin, 2020-02-16:
I haven't had time to move forward on that, but I did some research on how to do it in dockcross. I hope I'll find some time to continue it soon!

JonasVautherin, 2021-01-25:
We now provide it on pypi: https://files.pythonhosted.org/packages/46/24/8a71b6147111d4819750e453321430d93d8b3ba5bacd79dd68f618f6df3d/mavsdk-0.15.0-py3-none-manylinux2014_aarch64.whl

closed 2021-01-25 by JonasVautherin

SasoPavlic, 2021-02-14:
We tested a new wheel on [Navq ](https://nxp.gitbook.io/8mmnavq/) companion computer (aarch64v8) and is still installing forever (stops on grpcio package).
Command used: `pip install wheel_file.whl`

julianoes, 2021-02-15:
@SasoPavlic I assume that's a grpcio problem. See https://github.com/grpc/grpc/issues/21283.

nihanaltaytass, 2023-02-14:
I have pixhawk4 with px4 software and I would like to send MAVLINK signals to my pixhawk from jetson Nano  which has arm64 system. Is it possible or is there any update

cryptik, 2023-02-25:
@nihanaltaytass, MAVSDK and mavsdk_server work fine on Jetson.  I have it running on a Jetson Xavier and Orin (both aarch64 systems).  You can install both with just a `pip install mavsdk`.  The version I am running is `v1.4.0` on `python 3.8.10`.   The the only thing is you may have to create a symlink from where `mavsdk_server` is installed and the `./local/bin` folder.. or you can put it in your python_path.  But in any case, it works great on linux arm.

Pragadesh-45, 2023-12-11:
@nihanaltaytass Hi. I'm currently working on the exact setup. Any updates???

If possible: [https://github.com/thien94/vision_to_mavros/blob/master/scripts/d4xx_to_mavlink.py](url) provide me any alternative to this.