mavlink/MAVSDK-Python #118: Cannot upgrade MavSDK to 0.3.0

alexander-wu-aw, 2019-10-11:
For some reason, I can't upgrade to mavsdk 0.3.0.

Whenever I try `pip install mavsdk` or `pip install --upgrade mavsdk`, it only gives me version 0.2.0.

When I run `pip install mavsdk==0.3.0` it says:

"RROR: Could not find a version that satisfies the requirement mavsdk==0.3.0 (from versions: 0.1.0, 0.2.0)
ERROR: No matching distribution found for mavsdk==0.3.0"

julianoes, 2019-10-11:
@JonasVautherin can you check this?

JonasVautherin, 2019-10-11:
Linux, or macOS?

alexander-wu-aw, 2019-10-11:
On MacOS

JonasVautherin, 2019-10-12:
That's because it is not released on macOS yet. I'll do that ASAP and will communicate here and on Slack (#mavsdk)!

alexander-wu-aw, 2019-10-12:
Thanks a lot!

JonasVautherin, 2019-10-21:
That's finally released.

To install (first time):

```
pip install mavsdk
```

To upgrade:

```
pip install --upgrade mavsdk
```

It should install mavsdk 0.3.0, as can be seen with the output of `pip show mavsdk`:

```
$ pip show mavsdk
Name: mavsdk
Version: 0.3.0
```

closed 2019-10-21 by JonasVautherin