mavlink/MAVSDK-Python #119: zlib not found in dockcross compilation

julianoes, 2019-10-11:
I tried to build for arm64 but it failed to find zlib:
```
docker run --rm dockcross/linux-arm64 > ./dockcross-linux-arm64
chmod +x dockcross-linux-arm64
./dockcross-linux-arm64 cmake -Bbuild -S. -DBUILD_BACKEND=ON
./dockcross-linux-arm64 cmake --build build
```

The output was:
```
-- The C compiler identification is GNU 4.9.4
-- The CXX compiler identification is GNU 4.9.4
-- Check for working C compiler: /usr/xcc/aarch64-unknown-linux-gnueabi/bin/aarch64-unknown-linux-gnueabi-gcc
-- Check for working C compiler: /usr/xcc/aarch64-unknown-linux-gnueabi/bin/aarch64-unknown-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/xcc/aarch64-unknown-linux-gnueabi/bin/aarch64-unknown-linux-gnueabi-g++
-- Check for working CXX compiler: /usr/xcc/aarch64-unknown-linux-gnueabi/bin/aarch64-unknown-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) at cmake/zlib.cmake:42 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable ZLIB_ROOT is set to:

    /work/build/third_party/install

  For compatibility, CMake is ignoring the variable.
Call Stack (most recent call first):
  CMakeLists.txt:140 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.13/Modules/FindZLIB.cmake:114 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/zlib.cmake:42 (find_package)
  CMakeLists.txt:140 (include)

-- Configuring incomplete, errors occurred!
```

JonasVautherin, 2019-10-11:
[Those two lines](https://github.com/dockcross/dockcross/blob/master/linux-arm64/Toolchain.cmake#L16-L17) should be set to BOTH instead of ONLY, as I did for [android](https://github.com/dockcross/dockcross/blob/master/android-arm64/Toolchain.cmake#L15-L16) and others. I should have propagated that to more of the toolchains there, sorry :confused:.

julianoes, 2019-11-04:
@JonasVautherin has fixed this for all dockcross images that I'm aware of, closing.

closed 2019-11-04 by julianoes