mavlink/MAVSDK-Python #122: Does function set_position_ned work well ?

minimal1, 2019-10-30:
I tried drone control by Offboard. 
2 functions ( set_velocity_ned, set_velocity_body) worked well, but set_position_ned didn't.

I ran offboard example code using function set_position_ned.
there was no error to give command, but the drone was not moving.
Just stayed on ground like there was no command.

According to https://github.com/mavlink/MAVSDK-docs/blob/develop/en/guide/offboard.md, Position or thrust set points is not supported. 

How is it now ?

julianoes, 2019-10-31:
It would also be interesting if you could try the C++ integration test:
https://github.com/mavlink/MAVSDK/blob/develop/src/integration_tests/offboard_position.cpp

minimal1, 2019-10-31:
> @minimal1 what PX4 version were you using, and what MAVSDK version?
 Yes, I am using PX4. If you mean MAVSDK-Python, it is 0.3.0.
 
> It would also be interesting if you could try the C++ integration test:
> https://github.com/mavlink/MAVSDK/blob/develop/src/integration_tests/offboard_position.cpp
I would try this using real drone, right ?

julianoes, 2019-11-01:
I mean the PX4 version. You can get it using `ver all` in `nsh>` or `pxh>`.

I would try it in SITL first so you know what to expect.

julianoes, 2021-03-26:
Stale, closing.

closed 2021-03-26 by julianoes

moshibe, 2021-12-28:
@minimal1
Did you find out why it's not working? 
I'm having the same problem and I'm having trouble. 
Please give me some advice.