mavlink/MAVSDK-Python #124: MAVSDK-Python 0.3.0 not available on pip for Windows

wsolichin-bto, 2019-11-01:
I'm trying to use the MAVSDK-Python for Windows, and when I tried using the following line:

```from mavsdk import System```

it produced the following error: 

```ImportError: cannot import name 'System' from 'mavsdk'```

I checked the Python version in Windows for MAVSDK-Python by typing as follows:

```pip3 show mavsdk```

And it gives me version 0.1.0, and when I try to install version 0.3.0 typing as follows:

```pip3 install mavsdk==0.3.0```

It gives me this error:

```ERROR: Could not find a version that satisfies the requirement mavsdk==0.3.0 (from versions: 0.1.0)```

When is the MAVSDK-Python 0.3.0 planned to be release in Windows?

julianoes, 2019-11-02:
We are working on it:
https://github.com/mavlink/MAVSDK-Python/pull/123
https://github.com/mavlink/MAVSDK/pull/891
https://github.com/mavlink/MAVSDK/pull/893

Now we just need to cut a C++ release and then create a Python release on top of it.

julianoes, 2019-11-04:
`pip install --upgrade mavsdk` should now get you 0.4.0!

closed 2019-11-04 by julianoes

wsolichin-bto, 2019-11-06:
Thank you for the help! Really appreciate it!