mavlink/MAVSDK-Python #130: Cannot connect to mavsdk-server

chris-aeviator, 2019-11-10:
Hi! I'm struggling to get the internally started or externally started mavsdk server connected. I can therefore not reach any further in every example then `drone = System()`

### System Info
- Arch Linux
- conda env with python=3.7
- `pip3 install mavsdk` (0.4.0)

### Expected Behavior 
When I run `python3 takeof_land.py` I expect to see the SDK connecting and running the example code

### Actual Behavior

`Waiting for mavsdk_server to be ready...`

I checked that the mavsdk binary is inside the conda env site packages folder, I confirm an internal version is started, by starting the binary (from github releases page of mavsdk) after the python script has started and it errors out that the port (50051) is already taken.

To test the connection to the externally started mavsdk binary I invoke the init like so `drone = System(mavsdk_server_address="udp://0.0.0.0", port=50051)`

JonasVautherin, 2019-11-10:
> drone = System(mavsdk_server_address="udp://0.0.0.0", port=50051)

I think that's wrong. Can you do `drone = System(mavsdk_server_address="127.0.0.1")` instead? That should connect you to the externally-started instance.

For the "internally-started" instance, could you try to find the binary (somewhere in the installation, probably ending in `mavsdk/bin/mavsdk_server`) and try to start that manually?

chris-aeviator, 2019-12-09:
I can confirm being able to connect everything successfully with 

```python
drone = System(mavsdk_server_address="127.0.0.1")
await drone.connect(system_address="udp://:14540)
```

*with Airsim & PX4 sitl

closed 2019-12-09 by chris-aeviator

21shrinit, 2020-08-26:
> I think that's wrong. Can you do `drone = System(mavsdk_server_address="127.0.0.1")` instead? That should connect you to the externally-started instance.
>
> For the "internally-started" instance, [...]

I tried to find but in bin the mavsdk_server is not there
and I am getting the which says to start server manually but I'm not getting resource for starting the server manually

julianoes, 2020-08-26:
@21shrinit please don't necrobump and create a new issue and provide the information about the steps you tried, etc.