mavlink/MAVSDK-Python #131: Cannot connect to device

kiranshila, 2019-11-12:
I have been struggling with this for a few days now and I can't for the life of me figure out how to actually connect to the hardware. I am running macOS with Python 3.7.

I built everything from source and ran `mavsdk_server serial:///dev/tty.usbserial-AJ02ZC94:115200` to which I get the following response

```
[12:25:56|Info ] MAVSDK version: 0.21.1 (mavsdk_impl.cpp:25)
[12:25:56|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:390)
[12:25:56|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:47)
[12:25:56|Info ] Server started (grpc_server.cpp:31)
[12:25:56|Info ] Waiting to discover system... (connection_initiator.h:58)
[12:25:56|Debug] Component Autopilot (1) added. (system_impl.cpp:401)
[12:25:56|Debug] Component Gimbal (154) added. (system_impl.cpp:401)
[12:25:56|Debug] Discovered 2 component(s) (UUID: 1) (system_impl.cpp:561)
[12:25:56|Info ] System discovered [UUID: 1] (connection_initiator.h:62)
[12:25:56|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:222)
[12:25:56|Error] Error: get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:456)
[12:25:57|Error] Error: UUID changed (system_impl.cpp:257)
[12:25:57|Debug] MAVLink: critical: Data link lost (system_impl.cpp:308)
[12:25:57|Debug] MAVLink: info: Data link regained (system_impl.cpp:308)
```

To me, that looks like it connected to the platform.

Now, when I try to run any of the examples, it always hangs on `Waiting for drone to connect...`

I have tried setting the connection string directly to `serial:///dev/tty.usbserial-AJ02ZC94:115200` without the mavsdk_server running, I have tried setting the `system_address` kwarg to `localhost` and connect to the running instance, nothing.

Is there something fundamentally wrong with what I am doing?

Additionally, I don't know if it ever actually connects to the running mavsdk_server. I can put in garbage into the `system_address` kwarg and it always returns `Connected to mavsdk_server!` so I can't tell what's working, especially with all the asynchronous stuff.

julianoes, 2019-11-13:
No, it has not connected:
```
[12:25:56|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:390)
```

This is a dummy/non-vehicle. It should say something like `New: System ID: 1 Comp ID: 1`. I agree that's a bit confusing!

```
Connected to mavsdk_server!
```
This only refers to the gRPC connection between the Python client and the mavsdk_server running.

Are you sure the baudrate is 115200? Is this for a telemetry radio? Usually these are at 57600.

kiranshila, 2019-11-13:
Yes the baudrate is 115200. Hasn't it connected, though if it says
```
[12:25:56|Debug] Component Autopilot (1) added. (system_impl.cpp:401)
[12:25:56|Debug] Component Gimbal (154) added. (system_impl.cpp:401)
```

It certainly doesn't do that if the cable is unplugged. As in, it found the pixhawk and the gimbal, right?

julianoes, 2019-11-14:
Oh, it has, sorry I did not read that through!

I think I'm getting what is happening. You are running `mavsdk_server` manually and then using Python which also tries to start `mavsdk_server` but can't because your started one already has the port. I think you don't need to start your own `mavsdk_server`, and cnan just do this in Python:

```
# Init the drone
    drone = System()
    await drone.connect(system_address="serial:///dev/tty.usbserial-AJ02ZC94:115200")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break
```

JonasVautherin, 2020-01-08:
@kiranshila: any success on that front?

kiranshila, 2020-01-13:
I can try it again, but I switched over to using dronekit a while ago

JonasVautherin, 2020-01-13:
Ok! Closing for now, please feel free to re-open this issue if you need more help.

Also [this blogpost](https://auterion.com/getting-started-with-mavsdk-python/) might be a useful read to get started with MAVSDK-Python :blush:.

closed 2020-01-13 by JonasVautherin