mavlink/MAVSDK-Python #133: Silent error when checking connection_state() immediately after connection call

mwshafer, 2019-11-15:
I either found a bug, or I am doing something wrong. I have a Pixhawk connected via USB and I am finding that even if I `asyncio.create_task(my_function())` after I `await drone.connect(system_address = connection_string)` the code in `my_func()` is silently erroring out when I call `async for state in drone.core.connection_state():`

To highlight this problem, I have created the following program that has two different tasks that are created after a connection. One works, one doesn't work. The function names show you which is which. If you switch which line is commented, you can see the output of the one that works is
```
Connected to mavsdk_server!
Connected!
Step 1
Step 2
Exception!
Looping. Waiting 2 second
Step 2
Step 3
Step 4
-141.06626892089844
-141.06666564941406
-141.06735229492188
-141.06790161132812
-141.06768798828125
-141.06829833984375
...
```
The output of the one that doesn't work but should is
```
Started connection function...
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Connected!
Step 1
Step 2
Exception:

Looping. Waiting 2 second
Step 2
```
It is clear from the first output that an exception is being raise, as it is caught. The second output shows the exception being caught, but printing nothing. My main question is why are the tasks created on the `asyncio.create_task(...)` line if the initial `await drone.connect(system_address = connection_string)` on the line above that call hasn't finished running? We can tell it isn't connected, because the functions in the task are erroring out when trying to use the `drone.core.connection_state()` generator.
My code is below.

```
#!/user/bin/env python3

import asyncio
from mavsdk import System
import time

async def conn(drone):  #Note this used to be called run()
    #Initialize the drone
    print('Started connection function...')
    #connection_string = "serial:///dev/tty.usbmodem1:57600"
    connection_string = "serial:///COM3"
    #await asyncio.wait_for(drone.connect(system_address = connection_string), timeout=3)
    await drone.connect(system_address = connection_string)
    print('Connected!')

    asyncio.create_task(this_works(drone))
    #asyncio.create_task(this_should_work_but_doesnt(drone))

async def this_works(drone):
    print('Step 1')
    while True:
        print('Step 2')
        try:
            async for state in drone.core.connection_state():
                print('Step 3')
                if state.is_connected:
                    print('Step 4')
                    async for ang in drone.telemetry.attitude_euler():
                        print(ang.yaw_deg)
                else:
                    print('Disconnected. Patiently waiting for reconnection...')
        except Exception as e:
            print('Exception:')
            print(e)
        print('Looping. Waiting 2 second')
        await asyncio.sleep(2)

async def this_should_work_but_doesnt(drone):
    print('Step 1')
    async for state in drone.core.connection_state():
        print('Step 3')
        if state.is_connected:
            print('Step 4')
            async for ang in drone.telemetry.attitude_euler():
                print(ang.yaw_deg)
        else:
            print('Disconnected. Patiently waiting for reconnection...')

async def main():
    drone = System()
    await asyncio.gather(conn(drone))

if __name__=="__main__":
    asyncio.run(main())
    print('Done with program')
```

mwshafer, 2019-11-19:
It seems like an `await` is required with the `asyncio.create_task()` call. It should be `await asyncio.create_task(this_should_work_but_doesnt(drone))`. This isn't in the current mavsdk-python example for telemetry (those examples use the older `ensure_future()' function). I tried using `ensure_future()` with and without the `await` and it too needed the `await` to run the coroutine properly.
The issue has nothing to do with the contents of the coroutine. A simple `await asyncio.sleep(5)` in the coroutine will produce this same issue if the `await` isn't used in conjunction with the `asyncio.ensure_future()` or 'asyncio.create_task()' calls after the 'done.connect()` call.

Should the examples use this `await` prior to calls like `  asyncio.ensure_future(print_battery(drone))`? Have they been tested with a real flight controller connected, or just SITL (`system_address="udp://:14540"`)?

julianoes, 2019-11-19:
@xvzf it would be great to get your input here!

xvzf, 2019-11-19:
@mwshafer The `asyncio.create_task` silently creates a coroutine to be scheduled asap and should not be awaited as it is simply not needed and should not make a difference.

It is strange that the `async for` loop is not working, the `await drone.connect()` should initiate everything so that this does not fail and it also *seems* to be working with the SITL.

I could not identify which type of exception is raised from the provided output - can you try to figure it out and let us know?

@julianoes I don't have a pixhawk available right now. Can you try to replicate the issue? It'd be great If you can provide some feedback, I'll try to debug it then!

julianoes, 2019-11-19:
Thanks for chiming in @xvzf. I'll try to replicate it.

mwshafer, 2019-11-19:
I've built a more simple program that highlights what I believe to be the same issue with fewer lines of code than the original post. I took the `firmware_version.py` example and simply moved the connection state into a coroutine and added an `ensure_future()` to call the coroutine. My understanding is that for this purpose `ensure_future()` and `create_task()` are more or less equivalent. If you don't `await` the 'ensure_future()` call, the `async for` appears to never run. I think it is running, but is throwing this "silent error." You could confirm this by adding your own `try/except` in the `check_conn()` function, but I left it out here to try to make the script very clear. The code is below. Obviously, may need to change the `system_address`, from what is below, as I am connecting on a Mac.

```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():
    drone = System()
    await drone.connect(system_address="serial:///dev/tty.usbmodem1:57600")
    #Uncomment the line below to break this program
    #asyncio.ensure_future(check_conn(drone))
    #Uncomment the line below to fix this program
    await asyncio.ensure_future(check_conn(drone))
async def check_conn(drone):
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break
    info = await drone.info.get_version()
    print(info)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

mwshafer, 2019-11-19:
I want to clarify that this issues seems come up when

```
async for state in drone.core.connection_state():
        if state.is_connected:
        .....
```
is use in a coroutine. I have tested the `telemetry.py` example that does not include the `awaits` and it seems to work fine with a Pixhawk over usb. It seems like the issue may be related to the `core` or `connenction_state()` being used in the coroutine. The original `firmware_version.py` example works just fine, because the `connection_state()` is not used in a coroutine. I hope this helps.

xvzf, 2019-11-21:
I know what your issue is now. You are using `loop.run_until_complete(run())` and create a new coroutine inside of `run()` - though: the method exits and therefore the `loop.run_until_complete` is finished before the execution comes to the created coroutine. This should work:

```python
import asyncio
from mavsdk import System

async def run():
    drone = System()
    await drone.connect(system_address="serial:///dev/tty.usbmodem1:57600")
    # Coroutine is created
    asyncio.create_task(check_conn(drone))
    # run() exits after creating the coroutine which was not scheduled *yet*

async def check_conn(drone):
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break
    info = await drone.info.get_version()
    print(info)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    # The run coroutine just connects to the drone and creates your coroutine from check_conn()
    loop.run_until_complete(run())
    # This continues event loop execution and therefore the created coroutine gets scheduled
    loop.run_forever()
```

What would be easier to understand for such a simple program is this one:

```python
import asyncio
from mavsdk import System

async def run():
    drone = System()
    await drone.connect(system_address="serial:///dev/tty.usbmodem1:57600")
    # Just use await to ensure the coroutine is executed   
    await check_conn(drone)

async def check_conn(drone):
    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break
    info = await drone.info.get_version()
    print(info)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    # The run coroutine just connects to the drone and creates your coroutine from check_conn()
    loop.run_until_complete(run())
    # This continues event loop execution and therefore the created coroutine gets scheduled
    loop.run_forever()
```

@julianoes I'll close this issue as it is not related to the mavsdk internals

closed 2019-11-21 by xvzf

mwshafer, 2019-11-21:
@xvzf Thanks so much! I am still learning about asyncio and it is still hard for me to know when it is my code causing the problem. Thanks again.