mavlink/MAVSDK-Python #134: How to use the gimbal properly?

MicroYY, 2019-11-19:
The sdk provides an implementation of gimbal. 
I tried TYPHOON480 which is equipped with a gimbal and successfully set pitch and yaw angle using the command `vmount test yaw 30`.
And in my python script, the gimbal seems to work when I use the  `drone.action.takeoff()` and then `drone.gimbal.set_pitch_and_yaw`. 
But when it comes to the OFFBOARD mode, given a velocity to takeoff and then `drone.gimbal.set_pitch_and_yaw`, the gimbal will shakely very badly.
So how to use the gimbal properly?  Would you offer an example? I think there is something wrong with my configuration.

julianoes, 2019-11-19:
Thanks for the note. That's odd. @Jaeyoung-Lim have you seen that the Typhoon H480 gimbal would behave differently in offboard?

julianoes, 2019-11-19:
Let's track this in https://github.com/mavlink/MAVSDK/issues/907.

closed 2019-11-19 by julianoes

Jaeyoung-Lim, 2019-11-19:
On my side no, @MicroYY can you also try takeoff and control the gimbal without entering offboars mode?

If you are using a older version of the firmware, the gimbal was overturned and that might explain the oscillations after takeoff, but it is not related to offboard mode

Jaeyoung-Lim, 2019-11-19:
Copied reply to the tracked issue