mavlink/MAVSDK-Python #135: telemetry.landed_state() error "UNIMPLEMENTED"

gianlu33, 2019-11-20:
Hello,

I was trying to implement a blocking version of the `takeoff` action, basically i want to wait until the drone is in air. Unfortunately, when calling the method `drone.telemetry.landed_state()` in order to check the state of the drone, i get the following error:
```
Traceback (most recent call last):
  File "test.py", line 17, in <module>
    loop.run_until_complete(run())
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "test.py", line 7, in run
    await Control.takeoff_blocking()
  File "/home/gianlu33/Desktop/dromt/gitlab/mavsdk/api/Control.py", line 111, in takeoff_blocking
    async for landed in drone.telemetry.landed_state():
  File "/home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages/mavsdk/generated/telemetry.py", line 1931, in landed_state
    async for response in landed_state_stream:
  File "/home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages/aiogrpc/utils.py", line 138, in __anext__
    return await asyncio.shield(self._next_future, loop=self._loop)
  File "/usr/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages/aiogrpc/utils.py", line 126, in _next
    return next(self._iterator)
  File "/home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages/grpc/_channel.py", line 392, in __next__
    return self._next()
  File "/home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages/grpc/_channel.py", line 544, in _next
    raise self
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
	status = StatusCode.UNIMPLEMENTED
	details = ""
	debug_error_string = "{"created":"@1574289162.088409913","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1055,"grpc_message":"","grpc_status":12}"
```

Here is my code:

```python
async def takeoff_blocking():
    global drone

    print("-- Taking off")
    while True:
        try:
            await drone.action.takeoff()
            break
        except ActionError as error:
            # TODO maybe do something else
            print(f"Takeoff failed: {error._result.result}")
            await asyncio.sleep(Conf.REPEAT_TIME)
 
   # the error is triggered here
    async for landed in drone.telemetry.landed_state():
        if landed == LandedState.LANDED_STATE_IN_AIR:
            break

    print("-- In air!")
```
Is there some bug in the implementation of MAVSDK or i'm doing something wrong?

Note: i installed mavsdk through `pip`

julianoes, 2019-11-21:
Have you tried upgrading?

What's the output of this?
```
pip3 install --upgrade mavsdk
```

gianlu33, 2019-11-21:
Yes, nothing changes:

```
Requirement already up-to-date: mavsdk in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (0.4.0)
Requirement already satisfied, skipping upgrade: protobuf>=3.5.1 in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (from mavsdk) (3.10.0)
Requirement already satisfied, skipping upgrade: grpcio>=1.11.0 in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (from mavsdk) (1.25.0)
Requirement already satisfied, skipping upgrade: aiogrpc>=1.5 in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (from mavsdk) (1.6)
Requirement already satisfied, skipping upgrade: importlib-resources>=1.0.2; python_version < "3.7" in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (from mavsdk) (1.0.2)
Requirement already satisfied, skipping upgrade: six>=1.9 in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (from protobuf>=3.5.1->mavsdk) (1.12.0)
Requirement already satisfied, skipping upgrade: setuptools in /home/gianlu33/.virtualenvs/drone/lib/python3.6/site-packages (from protobuf>=3.5.1->mavsdk) (41.6.0)
```

julianoes, 2019-11-22:
Ok thanks. I'll check it.

julianoes, 2019-11-22:
Thanks, I can reproduce it.

julianoes, 2019-11-22:
It looks like we forgot to implement this in https://github.com/mavlink/MAVSDK/blob/develop/src/backend/src/plugins/telemetry/telemetry_service_impl.h.

julianoes, 2019-11-22:
Fix is coming soon: https://github.com/mavlink/MAVSDK/pull/911

closed 2019-11-25 by julianoes