mavlink/MAVSDK-Python #136: Issue on source build

petergerten, 2019-11-26:
Hi,
I am trying to get mavsdk-python running on an embedded system using a non standard build procedure and currently get the following error:

```
Python 3.7.4 (default, Oct 16 2019, 08:13:05) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mavsdk
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.7/site-packages/mavsdk/__init__.py", line 7, in <module>
    from .system import System
  File "/usr/lib/python3.7/site-packages/mavsdk/system.py", line 9, in <module>
    class System:
  File "/usr/lib/python3.7/site-packages/mavsdk/system.py", line 73, in System
    def action(self) -> Action:
NameError: name 'Action' is not defined
>>> 
```
Does anyone have a hint on what might be wrong in my setup?

julianoes, 2019-11-26:
Not sure, can you share the steps that you have done so far?

petergerten, 2019-12-04:
this was an unrelated build system issue

closed 2019-12-04 by petergerten