mavlink/MAVSDK-Python #137: GPS estimate doesn't work via MAVSDK-Python, but it works well from QGC

ramgrandhi, 2019-11-29:
Hello -
I'm trying to run takeoff_and_land.py from examples. Managed to connect pixhawk4 via USB/serial from python. However, it waits forever to obtain global position estimate. However, when I open up QGC - it was able to locate 7 satellites and has a decent position estimate established. 

`
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for drone to connect...
**Drone discovered with UUID: 354667xxxxxxxxx**
Waiting for drone to have a global position estimate...
`

Appreciate your thoughts!

cheers,
ram

ramgrandhi, 2019-11-29:
Ah, now it works (after say 2-3 mins of waiting for GPS -- it says GPS OK :) -- I wasn't patient enough I guess. Sorry for that!

However, the issue is with Arming itself. It throws:

`await drone.action.arm()  File` `"/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/mavsdk/generated/action.py", line 230, in arm`
`raise ActionError(result, "arm()")`
`mavsdk.generated.action.ActionError: COMMAND_DENIED: 'Command denied'; origin: arm(); params: ()`

Appreciate your thoughts!

cheers,
ram

julianoes, 2019-11-29:
The GPS information displayed in QGC is not necessarily what is required for PX4 to decide that the position estimate is good enough to fly. What counts is if PX4 has a global position estimate that is good enough and that's what MAVSDK tries to grab.

@maetugr is working on improving these preflight checks and how they are visualized.

Things that influence GPS signal:
- open sky: how much of the horizon is covered by buildings, etc.
- GPS receiver: some have not just GPS but also GLONASS, etc.
- GPS antenna: usually a big ground plane is better, and there are other factors.
- RTC battery: some receivers have a small batter for warm start which is faster than cold start.
- interference from other sources such as companion computers with wifi, usb3, flow sensors, cameras, etc.

ramgrandhi, 2019-11-29:
Thanks @julianoes for describing what's behind why **is_global_position_ok: False**!

Would like to have more feedback/metadata on preflight checks as they are going on inside. Despite the time it takes before returning is_global_position_ok: True, in my case, I still get COMMAND_DENIED when trying to arm(). And, I looked for all possibilities that I can including disabling safety switch on GPS, disconnecting/reconnecting RC, reconnect with QGC and close QGC etc. 

Interesting thing is, MAVSDK was able to upload mission but just could not arm(). 

Do you see any checklist that I'm missing here before arm() ?

Thanks again!

julianoes, 2019-12-02:
I suggest you open QGC next to it when you try to arm and maybe you'll see status text messages indicating what is wrong? There are other preflight checks which might be triggering.

ramgrandhi, 2019-12-06:
Good tip, @julianoes - Thanks!

It seems that I cannot test MavSDK APIs when my laptop is tethered to pixhawk4 (with GPS mounted). QGC shows up two errors as the main reasons for not arming.

1) USB check failed (as it is tethered) 
2) Very low voltage check

I would like to check if I'm in the right direction to build/test my MavSDK programs on real hardware without disabling USB checks and tinkering voltage numbers.

My goal is to test my programs as close to the real target hardware (pixhawk4 with px4).

Please advise.

cheers,
ram

ramgrandhi, 2019-12-06:
Never mind @julianoes .. i'm going with using headless-Gazebo to test my programs. 

Courtesy: blog post [here](https://www.google.com/search?client=safari&rls=en&q=mavsdk+python&ie=UTF-8&oe=UTF-8).

julianoes, 2019-12-09:
> Courtesy: blog post here.

I don't see what you meant to link there.

For the USB and voltage checks there are circuit breakers:
https://dev.px4.io/master/en/advanced/parameter_reference.html#circuit-breaker

ramgrandhi, 2020-01-25:
Thanks for the inputs! This can be closed. I resolved the issue after fixing the errors which QGC was throwing.

closed 2020-01-25 by ramgrandhi

Pratikw369, 2025-03-19:
prot:~$ /bin/python3 /home/pratik/Documents/test.py
Waiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate... i have lots of problem with my pixhawk arming can anyone reach me out to help me

julianoes, 2025-03-19:
@Pratikw369 please don't necrobump. Create a new issue and properly describe your problem including versions, context, etc.