mavlink/MAVSDK-Python #144: pip requirement.text install error

darkratio, 2019-12-17:
I tried install grpc using pip install gprc it shows its installed but still when I try to install the requirement.txt, it shows this error. please if anyone knows about this issue, it would a great help. 
below are the command and the error it shows:-

sam@sam:~/MAVSDK-Python$ sudo pip3 install -r requirements.txt -r requirements-dev.txt

WARNING: The directory '/home/sam/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
WARNING: The directory '/home/sam/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: protobuf>=3.5.1 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 1)) (3.11.1)
ERROR: Could not find a version that satisfies the requirement aiogrpc>=1.5 (from -r requirements.txt (line 2)) (from versions: none)
ERROR: No matching distribution found for aiogrpc>=1.5 (from -r requirements.txt (line 2))

JonasVautherin, 2019-12-17:
What about those warnings you have? Is it safe to ignore them? Also, which version of pip are you running (`$ pip3 --version`)?

I just tried and it works for me in a venv:

```
% pip install -r requirements.txt -r requirements-dev.txt                                                                           [0]
Ignoring importlib-resources: markers 'python_version < "3.7"' don't match your environment
Collecting protobuf>=3.5.1 (from -r requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/4e/26/1517e42a81de4f28ed0f3cdadb628c1b72f3a28f38323a05e251f5df0a29/protobuf-3.11.1-py2.py3-none-any.whl
Collecting aiogrpc>=1.5 (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/a3/61/00bb522858af1d0333316e3990c42027b2a2da8a14ff592616fc027bb658/aiogrpc-1.7-py3-none-any.whl
Collecting grpcio>=1.11.0 (from -r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/fc/bd/22a9efc527f5e43061b269abea390e402c5b8c1e6a0db0f4159ef2bd1eda/grpcio-1.25.0-cp38-cp38-manylinux2010_x86_64.whl
Collecting grpcio-tools>=1.11.0 (from -r requirements-dev.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/35/3e/76404669fad19ae4af499cad157cf8c432364abe27e6b16b154a5b35c295/grpcio_tools-1.25.0-cp38-cp38-manylinux2010_x86_64.whl
Collecting six>=1.9 (from protobuf>=3.5.1->-r requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/65/26/32b8464df2a97e6dd1b656ed26b2c194606c16fe163c695a992b36c11cdf/six-1.13.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in ./venv/lib/python3.8/site-packages (from protobuf>=3.5.1->-r requirements.txt (line 1)) (41.2.0)
Installing collected packages: six, protobuf, grpcio, aiogrpc, grpcio-tools
Successfully installed aiogrpc-1.7 grpcio-1.25.0 grpcio-tools-1.25.0 protobuf-3.11.1 six-1.13.0
```

with aiogrpc 1.7:

```
 % pip show aiogrpc                                                                                                                  
Name: aiogrpc
Version: 1.7
```

julianoes, 2019-12-17:
Try:
```
sudo pip3 install --upgrade -r requirements.txt -r requirements-dev.txt
```

JonasVautherin, 2020-01-08:
@darkratio: any news on that?

LiangR2017, 2020-03-26:
i meet the question too 
`ERROR:` No matching distribution found for aiogrpc>=1.5 (from -r requirements.txt (line 2))

LiangR2017, 2020-03-26:
tried for aiogrpc1.7

> python setup.py build 

> python setup.py  install 
and i got

> Finished processing dependencies for aiogrpc==1.7
also i successly import aiogrpc
but when i run

> pip3 install mavsdk
or try to `pythonsetup.py build`  for mavsdk
it shows 

> ERROR: Could not find a version that satisfies the requirement aiogrpc>=1.5 (from mavsdk) (from versions: none)

> ERROR:`No matching distribution found for aiogrpc>=1.5 (from mavsdk)

julianoes, 2020-03-26:
I think your Python 3 is too old.
Can you paste the output of:
```
which python
which python3
which pip
which pip3
python --version
python3 --version
pip --version
pip3 --version
```

LiangR2017, 2020-03-27:
> I think your Python 3 is too old.
> Can you paste the output of:
> 
> ```
> which python
> which python3
> which pip
> which pip3
> python --version
> python3 --version
> pip --version
> pip3 --version
> ```
i dont understand. i firstly have python2.7 python3.5.2 in my windows system
but the aiogrpc needs python 3.6 so i install the 3.6
however when i use pip 3  to install the aiogrpc (py2.7 is pip2,py3.5.2 is pip), 
it shows me the  that aiogrpc  needs the python of which the verson should be not up to3.6.
then i uninstall py3.6 install py3.7,but nothing changes.
So i turn to ubuntu system,and it works well.

julianoes, 2020-03-30:
Right, so presumably Python 3.5.2 is too old to work with aiogrpc.

closed 2020-03-30 by julianoes

JonasVautherin, 2020-03-31:
MAVSDK-Python requires python 3.6+, anyway.