mavlink/MAVSDK-Python #149: there is no connection with telemetry

gulyvz, 2019-12-19:
Hi,
i run examples with sitl and hitl(with esp8266) mode successfully but when i try examples with telemetry connection i couldnt reach to drone.  Is there any step which i escape? 

> Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!

julianoes, 2019-12-19:
What version is your esp8266? Does it support mavlink2?

gulyvz, 2019-12-20:
It will be the latest version and also support mavlink2 but i will check again. In hitl mode it works successfully i can read battery, firmware version etc.

JonasVautherin, 2019-12-20:
I'm not sure I understand what works and what doesn't. Would you mind giving more details on your setup?

gulyvz, 2019-12-22:
@JonasVautherin Of course, I have 3 different test modes. the first is sitl. I can run examples with sitl. second test mode hitl, usb connect to pixhawk 4 and esp8266 connect to wifi I set up. (esp8266, when connected to the telemetry 2 port of pixhawk 4). I can also run examples in this mode. the third mode is the mode I connect to pixhawk 4 with telemetry. I can't get data when I try with mavsdk while it's running in this mode.

JonasVautherin, 2019-12-22:
@unipheas wasn't esp8266 the one with which you had issues? :thinking:

@gulyvz: Can you explain how you connect to esp8266? That one creates a wifi network to which you connect, and then you listen on some UDP port, right?

unipheas, 2019-12-23:
@JonasVautherin Yes that's correct. I had a lot of connection issues with that module as well. Sometimes it worked, sometimes it didn't. It had something to do with the baud rate and the way I was connecting it through the telemetry radio. 

I discuss the issue [here](https://discuss.px4.io/t/developing-for-the-holybro-pixhawk4-mini-qav250-kit/11978/5) and [here](https://github.com/PX4/Firmware/issues/12493#event-2786971193)

gulyvz, 2019-12-23:
> @gulyvz: Can you explain how you connect to esp8266? That one creates a wifi network to which you connect, and then you listen on some UDP port, right?
Yes, exactly like that

> I discuss the issue [here](https://discuss.px4.io/t/developing-for-the-holybro-pixhawk4-mini-qav250-kit/11978/5) and [here](https://github.com/PX4/Firmware/issues/12493#event-2786971193)
Thanks, I missed it.

JonasVautherin, 2020-01-10:
I'm assuming that @unipheas' answer solved the issue. Feel free to reopen if necessary :wink:.

closed 2020-01-10 by JonasVautherin