mavlink/MAVSDK-Python #151: `global_from_local` helper?

JonasVautherin, 2020-01-03:
I'm wondering if we should add selected helpers to the SDK, like for instance `global_from_local()`:

```python
def global_from_local(north_m, east_m, ref_lat, ref_lon):
    world_radius_m = 6371000
    ref_lat_rad = math.pi / 180 * ref_lat
    ref_lon_rad = math.pi / 180 * ref_lon
    x_rad = north_m / world_radius_m
    y_rad = east_m / world_radius_m

    c = math.sqrt(x_rad * x_rad + y_rad * y_rad);

    if (c == 0):
        return ref_lat, ref_lon

    sin_c = math.sin(c)
    cos_c = math.cos(c)

    ref_sin_lat = math.sin(ref_lat_rad)
    ref_cos_lat = math.cos(ref_lat_rad)

    lat_rad = math.asin(cos_c * ref_sin_lat + (x_rad * sin_c * ref_cos_lat) / c)
    lon_rad = (ref_lon_rad + math.atan2(y_rad * sin_c, c * ref_cos_lat * cos_c - x_rad * ref_sin_lat * sin_c))

    global_lat = 180 / math.pi * lat_rad
    global_lon = 180 / math.pi * lon_rad

    return global_lat, global_lon
```

Then the question would be whether we want to add this as python code, or if we want to create a helper plugin that would do that in C++ and expose the interface for the bindings?

@julianoes, do you have an opinion on that?

julianoes, 2020-01-10:
I would probably add it to the Python wrappers for now. Once the mission interface supports global and local coordinates we could remove it again.

JonasVautherin, 2020-01-10:
Actually, I'll just close this for now, and we'll see if there is a need in the future. And we have this issue for the record :slightly_smiling_face:. Thanks for the input!

closed 2020-01-10 by JonasVautherin