mavlink/MAVSDK-Python #152: Syntax Error: takeoff.py

gbaweja, 2020-01-13:
```
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"Drone discovered with UUID: {state.uuid}")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok:
            print("Global position estimate ok")
            break

    print("-- Arming")
    await drone.action.arm()

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(5)

    print("-- Landing")
    await drone.action.land()

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
```

 print(f"Drone discovered with UUID: {state.uuid}")

Is there a syntax error in MAVSDK system as well?

Can I get some help to understand the hardware setup as well for Mavsdk?

JonasVautherin, 2020-01-13:
Hello! Could you please give more details?

> Is there a syntax error in MAVSDK system as well?

What makes you think that? Do you get a syntax error when running the code above?

> Can I get some help to understand the hardware setup as well for Mavsdk?

Maybe it's easier if you tell us on which system you want to run MAVSDK-Python :slightly_smiling_face:. We distribute it on pypi for Linux, macOS and Windows, and since it's Python3, it should run almost everywhere, I would say...

tomek-l, 2020-01-15:
Hi @gbaweja ,

You're likely using python version lower than 3.6. 

This way of formatting, called f-strings was introduced in python 3.6 and earlier versions will treat it as a syntax error.
```python
print(f"Drone discovered with UUID: {state.uuid}")
```

You can check your python version by executing:
```shell
python3 --version
```

JonasVautherin, 2020-04-06:
Closing because of inactivity. Feel free to re-open if needed!

closed 2020-04-06 by JonasVautherin