mavlink/MAVSDK-Python #153: set_takeoff_altitude raises ActionError: UNKNOWN

tomek-l, 2020-01-15:
Hi!

There seems to be a problem with the ```set_takeoff_altitude``` function.
It raises an error, even though the value seems to be changed (see snippet below)

It's the same on a physical Pixhawk 4 Mini with PX 1.9.2 and Gazebo

```python
>>> from mavsdk import start_mavlink
>>> from mavsdk import connect as mavsdk_connect
>>> start_mavlink(connection_url="udp://:14552")
>>> drone = mavsdk_connect(host="127.0.0.1")
>>> await drone.action.arm()
>>> await drone.action.set_takeoff_altitude(10)
Traceback (most recent call last):
  File "/home/tomek/.local/lib/python3.6/site-packages/aioconsole/execute.py", line 95, in aexec
    result, new_local = yield from coro
  File "<aexec>", line 2, in __corofn
  File "/home/tomek/.local/lib/python3.6/site-packages/mavsdk/generated/action.py", line 331, in set_takeoff_altitude
    raise ActionError(result, "set_takeoff_altitude()", altitude)
mavsdk.generated.action.ActionError: UNKNOWN: ''; origin: set_takeoff_altitude(); params: (10,)
>>> await drone.action.get_takeoff_altitude()
10.0
>>> await drone.action.set_takeoff_altitude(555)
Traceback (most recent call last):
  File "/home/tomek/.local/lib/python3.6/site-packages/aioconsole/execute.py", line 95, in aexec
    result, new_local = yield from coro
  File "<aexec>", line 2, in __corofn
  File "/home/tomek/.local/lib/python3.6/site-packages/mavsdk/generated/action.py", line 331, in set_takeoff_altitude
    raise ActionError(result, "set_takeoff_altitude()", altitude)
mavsdk.generated.action.ActionError: UNKNOWN: ''; origin: set_takeoff_altitude(); params: (555,)
>>> await drone.action.get_takeoff_altitude()
555.0
>>> 
```

JonasVautherin, 2020-01-15:
Thanks for the report, @tomek-l!

That's a duplicate of https://github.com/mavlink/MAVSDK-Python/issues/85, though. There I identified the bug and explained how to solve it. Would you be willing to contribute that? :blush:

tomek-l, 2020-01-22:
I opened a draft PR in MAVSDK. Please take a look.

closed 2020-01-22 by tomek-l