mavlink/MAVSDK-Python #154: gRPC server dies after uploading a geofence

bbworld1, 2020-01-22:
When you try and upload a geofence with `drone.geofence.upload_geofence()`, the geofence uploads, but subsequent actions (I tested with `drone.mission.upload_mission()`) fails with:
```
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "{"created":"@1579677625.837417436","description":"Error received from peer ipv4:127.0.0.1:50051","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"Socket closed","grpc_status":14}"
>
```
Not sure whether it's a bug with backend or binding, but it definitely fails on the Python example so filing here.

Steps to replicate:
1. add
```python
await drone.mission.upload_mission([MissionItem(lat, lon, 10, 5, True, float('nan'), float('nan'), MissionItem.CameraAction.NONE    , float('nan'), float('nan'))]) 
await drone.mission.start_mission()
```
below the `upload_geofence` function in `examples/geofence.py`.
2. Run `examples/geofence.py`

MAVSDK-Python version: 0.6.1, latest from pip (Python 3.8.0), Arch Linux

JonasVautherin, 2020-01-22:
I believe this is an issue in `mavsdk_server`, that I reported a few months ago in https://github.com/mavlink/MAVSDK/issues/871.

Would you be willing to have a look at that? That would be super helpful :blush:

coderkalyan, 2020-01-23:
@JonasVautherin I completely forgot to fix this, I apologize for this. I work with @bbworld1 on our drone project and he recently brought this to my attention again, so I will try to get this done ASAP.

obbe79, 2020-03-23:
After uploading a geofence (which I can't get rid of - please help!) in SITL, I get a similar error when arming:

`-- Arming
Traceback (most recent call last):
  File "test_main.py", line 87, in <module>
    loop.run_until_complete(run())
  File "C:\Users\user\Development\python_proj\mavsdk\mavsdk_env\lib\asyncio\base_events.py", line 583, in run_until_complete
    return future.result()
  File "test_main.py", line 74, in run
    await drone.action.arm()
  File "C:\Users\user\Development\python_proj\mavsdk\mavsdk_env\lib\site-packages\mavsdk\generated\action.py", line 224, in arm
    response = await self._stub.Arm(request)
  File "C:\Users\user\Development\python_proj\mavsdk\mavsdk_env\lib\site-packages\aiogrpc\channel.py", line 40, in __call__
    return await fut
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.UNKNOWN
        details = "Stream removed"
        debug_error_string = "{"created":"@1584991977.922000000","description":"Error received from peer ipv6:[::1]:50051","file":"src/core/lib/surface/call.cc","file_line":1056,"grpc_message":"Stream removed","grpc_status":2}"`

JonasVautherin, 2020-03-24:
Thanks for reporting. As far as I can see, the [corresponding issue](https://github.com/mavlink/MAVSDK/issues/871) is still open in MAVSDK. Would you be willing to have a look at this?

Otherwise I may get to fix it in the next few weeks as part of the C++ auto-generation (I currently only finished the action plugin, but all of them will eventually be auto-generated).

coderkalyan, 2020-03-24:
@JonasVautherin I took a look at this again but I have no idea where the problem is :(. Do you have any pointers on how to debug?

coderkalyan, 2020-03-27:
@obbe79 If you're interested/able to to fix the bug, feel free to. In the meantime, I'm going to try to look into it again now that @julianoes refactored mission into core.

siddux, 2020-03-31:
Hi guys, 

does anyone know where is the problem? I was trying to fix it but for the moment I get nothing.

julianoes, 2020-04-01:
I think we should get this fixed with the next round of releases/updates.

obbe79, 2020-04-01:
> @obbe79 If you're interested/able to to fix the bug, feel free to. In the meantime, I'm going to try to look into it again now that @julianoes refactored mission into core.

thank you, I will see but I don't think I'm able to fix it

coderkalyan, 2020-04-02:
@julianoes I agree completely! I also have no clue what's going on, and my gRPC experience is limited to reading some online docs and copying existing plugins! Do you have any idea what's wrong, or even any debug techniques? The error logs by themselves are very unhelpful :(

JonasVautherin, 2020-05-13:
We have a new `geofence` implementation since `0.7.0`, and it works for me. I'll therefore close this, but feel free to reopen if necessary!

closed 2020-05-13 by JonasVautherin